sparkfun/SparkFun_RTK_Firmware #407: Connect Facet to ChcNav Landstar 7 App

amlago, 2023-02-18:
I would like to use the Facet with Chcnav's Landstar 7 app for Android.
Of course the Facet is not in the list of devices in the application and I connect as a generic NMEA device and it works.
The satellites shown by the application vary a lot from one message to another.
Is it a problem of misinterpretation by Landstar 7 or should I change the message rate?
I currently use a message rate of 1.

gdt, 2023-02-18:
NMEA processing of messages is tricky, and requires merging messages within a cycle and then detecting when there is a new cycle.   It is possible that because of multiple constellations, the app you are using is just displaying the information in each sky message, rather than unioning them together.   As an experiment, change to GPS only and see what happens.

amlago, 2023-02-20:
I tried only with GPS and the change of indicated satellites continues to occur on some occasions.
This does not affect the fix position, which is maintained without problems.
Thanks for the tips.

tonycanike, 2023-02-20:
I agree with Greg's comments.  I've seen similar behavior in other software, especially older software developed when there was only one or two constellations (GPS & GLONASS).  

My newer software (SurvPC 6) doesn't have this issue, but SurvCE 2 did. I don't recall SWMaps having this issue either. 

This is likely a CHC issue.

amlago, 2023-02-20:
I think Tony's comments are correct. It must be something from the CHC application. I tried with Landstar 8, which is the latest update, very recent, and the same thing happens to me.
The number of satellites fluctuates between arrival and arrival of messages. It seems that the fixed data is maintained but it does not give me confidence.
I'll keep trying.

nseidle, 2023-03-08:
I attempted to test this but the  Landstar 7 app for Android is locked for eval use only and does not allow connections over Bluetooth (it's the thing they do to cripple the software). If anyone knows where to get a working version for eval, I'll write docs against it. Otherwise, I suspect it will work just fine with the above recommendations, I just can't say for sure.

Closing for now. If anyone knows with certainty, or can point me at a working app, please let me know.

closed 2023-03-08 by nseidle