sparkfun/SparkFun_RTK_Firmware #409: Local Ntrip

LarteyLawson, 2023-02-21:
**Can we have a Local Ntrip mode?**

* SparkFun RTK Facet L-Band v3.1 Feb 4 2023

Assuming somewhere in the field I wish to use my SparkFun RTK unit as a Base to output corrections to my RTK drone 
and there is no internet available. Given that the Sparkfun RTK unit can generate its own wifi-hotspot or connect to an external one, can we have the feature where it can output RTCM3 corrections to an RTK drone Remote controller that is already connected to the SparkFun's hotspot or the same external wifi to the Sparkfun is connected to?  That is to say that the feature allows us to configure 
a Local Ntrip credentials such as IP address, port, username, password, and Mount point without having to connect to a caster via the internet. This use case scenario makes perfect sense as the drone pilot seldom changes the take-off point (within the wifi coverage of the SparkFun RTK)

nseidle, 2023-02-21:
I think you're asking for the RTK device to act as a NTRIP Caster. Hard, but not impossible. At that point you would be limited by the WiFi range of the RTK device. Why not use an external radio such as the [100mW Serial Telemetry radios](https://www.sparkfun.com/products/19032)? They work without any further mod and add only 23g to your drone. Docs are [here](https://docs.sparkfun.com/SparkFun_RTK_Firmware/correction_sources/#temporary-base).

Closing since I don't think the code bloat is worth the feature when this is really what an external radio is for. If other folks would like to add their support, I'd consider it.

closed 2023-02-21 by nseidle

LarteyLawson, 2023-02-21:
Thanks, @nseidle. 

Clearly, I have been misunderstood.  I will just add this reference to this Emlid feature I have used for a while now and hope it comes to the Sparkfun RTK. This will mean so much to me and the drone surveyors community.

 [https://docs.emlid.com/reachrs2/tutorials/integration/p4p-rtk](url)
[image]

nseidle, 2023-02-21:
[image]

Ahh, so the DJI remote must be transmitting RTCM data along side telemetry data. I won't say no, but it will be a future feature for now.

jakevis, 2023-06-29:
Really looking for this feature as well - if it helps, adrusimple is using this on their ESP32's to provide a local caster:
https://github.com/nebkat/esp32-xbee

Agyapong12, 2023-06-29:
If this feature is implemented, it would be a game changer for us in the drone surveying community.

tonycanike, 2023-06-29:
In the Emlid solution, is the NTRIP caster code running in the Reach or is it running in the Emlid Flow phone app?  In other words, does the operational configuration include (1) the Reach, (2) a phone running Flow, and (3) the drone controller?

When using a local base for an RTK Drone in an environment with no internet access, how is the WiFi network provided? Is there another device being the WiFi Access Point, or does the Reach do that?

The CPU, IO, and memory constraints in the ESP32 need to be well-managed, and the SparkFun folks have done an amazing job of adding functionality to their RTK product line given the low-cost ESP32 and the very approachable price point of their products. The SparkFun products are 1/2 or even 1/3 the cost of the Emlid products. 

Would a third device (WiFi AP, NTRIP caster, some minimal network services, etc.) be an acceptable solution?  I would imagine this would be very doable with an inexpensive SBC.

gdt, 2023-06-29:
Following up on @tonycanike's comment, while I have don't this, it seems that one can already point two devices at each other for base/rover over IP, without any NTRIP, assuming they are on the same wifi.  So I don't see how NTRIP would give functional benefit.

I can see that someone might want to have one reference station and then say 5 rovers.  However, making NTRIP work for 1 and making a caster that handles 5 connections at once is different, not so much code as memory.   So indeed, having a computer (RPI, laptop, whatever) be a DHCP server and NTRIP makes a lot of sense to me.

LarteyLawson, 2023-06-29:
@tonycanike  If it boils down to resource management on the ESP32 then perhaps you are right. 
In the Emlid solution, the local NTRIP caster is running on the Reach itself. Nonetheless, if there is another wifi source, let's say
a hotspot from a phone, the RTCM3 messages can be transmitted over that to the drone's remote controller as well. One should know that no internet is required in both scenarios.

The question I keep asking myself is can the RTCM3 messages be broadcast over the SparkFuns Wifi AP? I ask because I can't get it to work with ESRI's Survey123  using the WIFI / TCP server but works fine via Bluetooth. 
And we ask for the local Ntrip feature because regardless of the price difference between the Emlid rs2 and Sparkfun, tech consumers live for healthy competition. Otherwise we wont be out there comparing GNSS devices

tonycanike, 2023-06-29:
@LarteyLawson That all makes sense, thanks.  And yes I compare these devices too!

It's my understanding that the SparkFun WiFi Access Point only runs when the device is being configured. When the device is actually running for survey work, it can be a WiFi client to send or receive NTRIP, but not a WiFi AP.  But I could be wrong about that, I usually use the radios and a local base-rover pair.

LarteyLawson, 2023-06-29:
@tonycanike Unfortunately, I could not go back to the radios after trying network RTK, and I do not want to go back to network RTK after trying L-band PPP. I believe the coverage a GNSS device can obtain and still be centimeter-level accurate is the ASCENSION for me but the rest are all necessary to have in one's arsenal. 

I personally feel there are many ways to configure the Sparkfun RTK. 
1. USB Terminal (Tera Term)
2. Bluetooth terminal
3.  Wifi config.
All are excellent in several use cases and "RCTM over Wifi AP or Client" should not be where the ESP32 draws the line 😅.

jakevis, 2023-07-07:
So I am actually playing with some code on an WEMOS D1 MINI ESP32 based on the socket code from https://github.com/nebkat/esp32-xbee/blob/master/main/interface/ntrip_caster.c - intent being I could connect this ESP32 to the radio port and have it either create a wifi AP or join an existing network and establish a very hacked together caster. This will make my drone work WAY WAY WAY easier... that said, just like there is a TCP socket server/client option in the existing firmware, I would love to see an "NTRIP Caster" option as well.

ne14Yah, 2023-07-19:
I am on the hunt for an economical GNSS receiver that can cast NTRIP via wifi to my dji pro receiver/M300.  I would not hesitate to purchase a SparkFun receiver except it lacks this capability.  I have inquired on SparkFun's forum if it is possible, but so far crickets.  Not sure how I found this thread, but thankful to have done so.  I will most likely purchase a Robodot if the SparkFun line of receivers are unable to cast NTRIP.  I am not a coder, so go easy on me and thank you in advance for any help that someone can provide!

johandewaal, 2024-07-25:
Also super keen for this feature - my main GPS I use as RTK base with my M3E need to be shipped off for repairs leaving me stranded without a base to be able to connect and provide correction data via my M3e's controller via wifi or Bluetooth.  I usually pilot from quite close to the base within wifi range, and connect the controller via wifi to the RTK base. 

I have a pair of Sparfun surveyors which i used before my more professional grade GPS that now needs repairs,  so repurposing the Surveyor as  a base with local NTRIP  would be awesome.  I tried setting it up via wifi and Bluetooth without success and in through the process trying to find documentation on how to setup the functions currently available on the RTK Surveyor, I  stumbled on this thread

Has there been any further development on this request? Or is there reference guide on how to setup this possibly via wifi even if it means setting up an AP (not internet connected) to generate a wifi platform that both the M3E controller and Surveyor base can connect to to transmit correction data via the AP?

YaoEscal, 2024-11-06:
> **Can we have a Local Ntrip mode?**
>
> * SparkFun RTK Facet L-Band v3.1 Feb 4 2023
>
> Assuming somewhere in the field I wish to use my SparkFun RTK unit as a Base to output corrections to my RTK drone [...]

Hello,
Any news on that topic ?
It happens that I have  the same need.
Hope something is possible from now
Rgds

that2, 2025-07-14:
Hi, in July 2025. Just asking if this feature is in plan to be updated in a near future or I have to migrate to an ardusimple solution?

nseidle, 2025-07-14:
This feature is called [BaseCast](https://docs.sparkfun.com/SparkFun_RTK_Everywhere_Firmware/menu_base/?h=basecast#base-cast) and has been added to the next version of the RTK Firmware called RTK Everywhere, and is located [here](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware). Unfortunately we cannot add it back to this codebase because of RAM constraints.