sparkfun/SparkFun_RTK_Firmware #411: sdCardSemaphore failed to yield

thealastair, 2023-02-25:
### Subject of the issue
I'm often seeing the error log "sdCardSemaphore failed to yield" and this seems to results in data not being written to the SD card and/or panics

### Your workbench
* SparkFun RTK Express v3.1-Feb  4 2023
* What radios are you using: Bluetooth, WiFi, and/or ESP-Now? What app are you using to connect over Bluetooth? Are you transmitting NTRIP back to the device?
Wifi to receive RTCM corrections, Bluetooth is not connected
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post any log files from serial output that may have been generated.

Operating with a 16 GB SD card. 

```

sdCardSemaphore failed to yield, menuMessages.ino line 580

sdCardSemaphore failed to yield, held by SD Size Check, menuSystem.ino line 965

assert failed: xQueueGenericSend queue.c:832 (pxQueue->pcHead != ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == xTaskGetCurrentTaskHandle())

Backtrace:0x40083dad:0x3ffd0b900x40095369:0x3ffd0bb0 0x4009a9d5:0x3ffd0bd0 0x40095e06:0x3ffd0d00 0x400e4d4e:0x3ffd0d40 0x400e8522:0x3ffd0ed0 0x400ed456:0x3ffd0f10 0x400ed555:0x3ffd0f30 0x400ed58f:0x3ffd0f50 0x4011602d:0x3ffd0f70 

ELF file SHA256: 0000000000000000

```

### Expected behavior
Data should be logged to SD card and device shouldn't crash.

### Actual behavior
Data is often not logged to SD card and device crashes/reboots. After reboot SD card cannot be written to.

nseidle, 2023-03-02:
I'm struggling to replicate this. I've got an Express with a 64GB card (and 16, and 1GB) and v3.1 firmware. Everything is working as expected.

Can you post your system menu output? Here's mine:

[image]

Can you post more of the serial output during a system reset? I'd like to know more about where it's failing.

This is what my boot looks like with v3.1:

    Display started
    ZED-F9P firmware: HPG 1.13
    LittleFS Started
    SparkFun RTK Express v3.1-Feb  4 2023
    microSD: Online
    Profile 'Profile1' loaded
    MAX17048 configuration complete
    Batt (104%): Voltage: 4.22V Discharging: 0.00%/hr Green
    GNSS configuration complete
    Accelerometer configuration complete
    System time set to: Thursday, March 02 2023 20:37:05
    *WIFI_OFF
    Bluetooth broadcasting as: Express Rover-9CEE
    *NTRIP_CLIENT_OFF
    NTRIP Client start
    NTRIP_CLIENT_ON
    State: Rover - No Fix, 2023-03-02 20:37:07.149
    Batt (104%): Voltage: 4.22V Discharging: 0.00%/hr Green
    WIFI_CONNECTING
    NTRIP_CLIENT_WIFI_STARTED
    State: Rover - Fix, 2023-03-02 20:37:07.236
    Connecting WiFi... SD card size: 59.5 GB / Free space: 35.4 GB
    
    WIFI_CONNECTED
    NTRIP_CLIENT_WIFI_CONNECTED
    Rover Accuracy (m): 3.5102
    Log file name: SFE_Express_230302_203710.ubx
    NTRIP Client connecting to rtk2go.com:2101
    NTRIP Client connected
    NTRIP Client sending credentials: test@test.com:
    NTRIP Client serverRequest size: 120 of 515 bytes available
    NTRIP Client sending server request:
    GET /bldr_SparkFun1 HTTP/1.0
    User-Agent: NTRIP SparkFun_RTK_Express_v3.1
    Authorization: Basic dGVzdEB0ZXN0LmNvbTo=
    
    
    NTRIP_CLIENT_CONNECTING
    Log file size: 2510 - Generation rate: 0.5kB/s
    NTRIP_CLIENT_CONNECTED
    Batt (104%): Voltage: 4.22V Discharging: 0.00%/hr Green
    Rover Accuracy (m): 1.0151
    WiFi IP address: 10.8.252.137 RSSI: -55
    State: Rover - RTK Fix, 2023-03-02 20:37:13.739
    Rover Accuracy (m): 0.0141

Is your card formatted for FAT16? I doubt that would be an issue though.

thealastair, 2023-03-02:
My device completely died soon after this, it somewhat looks like it's failing to charge the battery via USB but I haven't had a chance to debug. I was often seeing faults where the GPS would fault to initialise as well, not sure if I've received faulty hardware or if other users are experiencing this as well?

nseidle, 2023-03-02:
What's your order number? I'll get an RMA started.

nseidle, 2023-04-24:
Hi @thealastair - I haven't heard anything. Did you get support from our returns team? 

Additionally, we pushed a fix [here](https://github.com/sparkfun/SparkFun_RTK_Firmware/commit/3755f8b7c21101eb428104d1ad1027b35292cbee) that may help resolve your issue. 

I'm going to close this for now. Please let us know if you have any new issues.

closed 2023-04-24 by nseidle