sparkfun/SparkFun_RTK_Firmware #414: Battery icon at the start of AP Config

amlago, 2023-02-26:
Battery icon at the start of AP Config

I would like to see the battery charge value with a battery icon on the AP config page.
It could be put at the beginning below the Sparkfun logo.

tonycanike, 2023-02-28:
I like this idea. 

Could it include external power connected status?  I have a need to know external power is connected when the battery is fully charged.  

The battery LED on the front panel goes dark when the battery is fully charge whether or not there is external power connected.  

See

https://github.com/sparkfun/SparkFun_RTK_Facet/issues/9

nseidle, 2023-02-28:
These are great ideas. I'm not certain we can detect external power or not but we'll try.

amlago, 2023-02-28:
In the serial configuration, the battery status is shown and if it is charging or discharging.
I think that information would be more than enough if it was placed under the initial Sparkfun icon with an icon next to it.

nseidle, 2023-02-28:
That's what we're leveraging. The problem is that the fuel gauge IC is very slow to respond. It can takes multiple minutes to switch from an indication of discharging to charging at near 100% capacity. We'll get something implemented.

nseidle, 2023-03-03:
Here's the basic implementation:

[image]

There's still some alignment, and the icon may move eventually, but it works. Icon is green if charging. '+97%' is charging, whereas '97%' is discharging.

nseidle, 2023-03-03:
Also, battery level is updated dynamically every second. As noted previously, the system can take a few minutes to recognize the transition from discharging to charging.

amlago, 2023-03-03:
You really do a spectacular job.
Excellent

nseidle, 2023-03-03:
> You really do a spectacular job.

Thank you! That makes my Friday :)

nseidle, 2023-03-08:
This is added in [v3.2](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases).

closed 2023-03-08 by nseidle