sparkfun/SparkFun_RTK_Firmware #417: Added to the Fixed Geodetic base configuration over wifi

amlago, 2023-03-03:
facet v3.2

Added to the Fixed Geodetic base configuration over wifi

1) As of today, if I press the paste button, the coordinates are added as if they were brand coordinates when in reality they are antenna coordinates (phase center)
2) I think we agree that the coordinates we save in the base station file are mark coordinates and from the HAE Mark height we calculate the HAE APC.
3) So what I would like is a check to indicate if the coordinates are of the mark or of the antenna (phase center) and thus the corresponding boxes are completed and the missing height is calculated.
4) If the check indicates that they are brand coordinates, the values of Ll will be loaded where they correspond and the height of the antenna phase center will be calculated as always.
HAE APC = HAE Mark + AH+ARP
5) If the check indicates that they are phase center coordinates, Ll is loaded where it corresponds and the height goes in the HAE APC box.
The mark height is then calculated and loaded into the HAE Mark box.
HAE Mark = HAE APC - AH-ARP

**In the file we will always save the coordinates of the Mark and that could be clarified in an info button.**

Excuse the English translation.
I hope I have been clear, what I am saying would leave the basic configuration almost perfect from my point of view.

Greetings

nseidle, 2023-03-03:
Thank you for explaining this to me *again*! I think I'm understanding.

[image]

[image]

Adjust HAE Mark by Antenna Height and ARP [checkbox]

Tool tip text: If enabled, HAE Mark is calculated by *subtracting* the entered Antenna Height and ARP from the altitude reported by the GNSS receiver. This is handy when measuring a mark with a known pole height and ARP, but unknown mark elevation. Note: The HAE Mark value is always recorded to the Commonly Used Coordinates table.

Do we need to store the value of the checkbox with each common used coordinate?

nseidle, 2023-03-03:
[image]

Above is what we've always had.

[image]

Above, clicking Adjust Mark locks the HAE Mark box. When a user adjusts the AH, ARP, or HAE APC then HAE_Mark is adjusted.

Is this what you are looking for?

nseidle, 2023-03-03:
I've added this to the RC if you want to try.

tonycanike, 2023-03-04:
Does it still work the other way around?

If I type in mark coords and elevation, or load mark coords and elevation from the csv file, the mark elevation, ant height, and ant offset are ADDED to calc the phase center elevation to set the ZED base?

amlago, 2023-03-04:
We are on the right way. I like it and it's what I was looking for. But we must adjust a couple of things

1) When I load coordinates with the Paste current Llh button, the Adjust Mark check should be checked automatically and the h should be loaded into the HAE APC box and calculate the HAE Mark

2) When I load coordinates from the list of coordinates saved in the file, the height h should be loaded in the correct box according to whether the Adjust mark check is checked or not.

I clarify below what is expected when pressing the Load button

2) a) If check is checked, the h coordinate must be loaded in the HAE APC box and the HAE Mark is calculated as we spoke before
2)b) If check is empty, the h coordinate should be loaded in the HAE Mark box and the HAE APC is calculated as usual

3) I have seen that sometimes the values are not updated dynamically. Maybe it is useful to add a button to update data as in Windows-like operating systems. It's an idea and if it's nonsense they tell me.

tonycanike, 2023-03-04:
This UI is getting overloaded and I believe we're trying to build too many capabilities into one screen.  

I'm getting confused about what's what.

Any chance of separating the two different capabilities from each other and put them into different workflows/pages/forms/screens to prevent confusion. 

1.  Setting the coordinates of the ZED in fixed base mode via manual entry into a form or by recalling the coordinates from the coordinate file. 
(adding three heights to calc APC height)

2. Getting the current ZED coordinates and saving them to the coordinate file.
(subtracting heights to get mark height)

The ongoing frequent changes and confusion over the past four months on this topic quite frankly concerns me that I can't rely on the Facet for my surveying work. I spend hours testing each version to make sure it works and then it changes again and again. 

#330

amlago, 2023-03-04:
The idea seems simple to me and it is what you once raised.  I thought that at the beginning I had clarified it.  With the check it is indicated if the coordinates are of the mark or of the phase center of the antenna.  The intent is to get tick heights and save them to the file without having to do the math manually.  The subtractions or additions that you already know.  Greetings

tonycanike, 2023-03-04:
Perhaps the UI text is unclear.  Questions & comments:

1.  Does "Adjust Mark" adjust the LL of the mark, or just the height of the mark?  Should it be labeled "Adjust Mark Height?"
2. If the "Adjust Mark" box is unchecked, does this imply "Adjust APC"?  
3. Can the Tool Tip be clarified as to what happens if it's not checked?
4. The verb "Adjust" has a very specific meaning to US surveyors.  "Adjust" specifically refers to mathematical calculations performed when one has redundant measurements to achieve a single maximally-accurate measurement.  Common adjustment techniques include "Compass Rule", "Transit Rule", and "Least Squares".  What is happening on this UI is not adjusting, it is calculating.
5. Can we replace the term "Adjust" with "Calculate" or "Calc" 
6. I'd really like to see the UI be very clear on the fact that there are two exclusive options, "Calculate APC Height" and "Calculate Mark Height", and that only one of the options can be chosen at a time.
7. Would it be more clear to have two exclusive radio buttons  "Calculate APC Height" and "Calculate Mark Height" ?  Radio Buttons are a common UI control to indicate "choose one but not multiple".
7. I think we want to require one of the options be chosen, but I am not sure.   Is there a use case in which no height calculation would be needed to be done?   Perhaps we really want three radio buttons, "Calculate APC Height", "Calculate Mark Height", and "No height calculations"?  I am not sure.  
11. If a user is unsure, clicks/checks the wrong choice, and changes their mind, will data be lost or overwritten?
12. When is the calculation performed?  Can it be undone?

Thanks for reading all this!

Tony.

amlago, 2023-03-04:
Hi, Tony.
I will try to answer in the order you asked and with the ideas that I have. It is not the revealed truth. Ha ha

1) If Adjust Mark is checked, it means that the coordinates are from the antenna phase center (HAE APC) and the HAE Mark is calculated. HAE Nark's locker is grayed out and cannot be changed manually.
ll be loaded where ever
2) If Adjust Mark is not checked, it means that the coordinates to enter are of the mark (HAE Mark) and the height of the phase center (HAE APC) is calculated. The HAE APC box is grayed out and cannot be manually modified.
3) I suppose that Nathan should be able to put the information that is necessary
4) and 5) I am a professional in Topography and Geodesy and I understand very well what you are saying. Possibly Adjust is not the most appropriate word.
You could put "Calculate height Mark" .
6) and 7) The check is clear to me. But putting two checks, one to calculate the HAE Mark and the other to calculate the HAE APC, I suppose would work. They should be exclusive. If I mark one the other is untilted. I think it would work. In any case, the function of the checked or unchecked check is clear to me.
8) It seems to me that if we correct what I wrote a message back about the Paste and Load buttons it would be ready soon.
9) I have tried the check and nothing was deleted. But we are in the process of programming by Nathan.
10) On this I did not get to try. Will investige.

Many greetings. We are close to not having to modify the base configuration anymore, which is, by the way, the one I use the most.
Be careful with the translations, I come from Spanish.
Angel

amlago, 2023-03-04:
> We are on the right way. I like it and it's what I was looking for. But we must adjust a couple of things
>
>     1. When I load coordinates with the Paste current Llh button, the Adjust Mark check [...]

The problem is that when using the Paste and Load button, the h-height is always loaded into the HAE Mark box, regardless of the state of the check (Adjust Mark).
We must make the height data h be loaded where appropriate as indicated by the Adjust Mark check.

tonycanike, 2023-03-05:
Thank you for your detailed responses!  They are appreciated.   I agree with most of what you said, and I have a couple questions.

I agree with (1)  - if copying or loading the coordinates from the ZED, the h from the ZED needs to go into the HAE APC box and the Mark HAE should be calculated.  (As if the "Calculate Mark HAE" is checked automatically.)  It might be confusing if someone unchecks the box and the firmware automatically checks it, but a tool-tip on the button could explain what's going to happen.

I agree with (3) - it is likely that someone would load coordinates from file or copy/load from the ZED, and then realize they need to enter the antenna height and/or phase center offset, and then recalculate.

The distinction between 2a and 2b is subtle and likely to be confused.  I'd like to discuss this more please.   If someone loads from file with the box incorrectly set, they could easily cause errors in their survey.

Do we really need 2a?   I believe we've agreed that the coordinate CSV file usually contains the mark HAE. This makes the most sense for surveyors setting up a tripod or pole over a mark.   If someone is operating a CORS station, the "mark" is the mounting point of the antenna, and the antenna height is zero.   A CORS operator would not want to store the HAE APC because they might need to change the antenna type and the phase center offset might be different.  If they really really really wanted the csv file to contain the APC HAE, the user could just enter zeros in for the antenna height and phase center offset.

If we get rid of 2a, _we don't need the check box at all!_  Less confusion.
1.  **Copy/Load/Paste from ZED:**   the firmware loads the ZED's HAE into the   HAE APC and calculates the HAE Mark.  The user can then save the HAE Mark to the file.  If they are operating a CORS station, they enter zero for the antenna heights and enter the phase center offset.  If they really want to save the HAE APC to the file, just put in zeros for the antenna height and phase center offset.
2.  **Load from CSV file**:  put the HAE from the CSV into the HAE Mark and **calculates the HAE APC.**  If they are operating a CORS station, they put zero in for the antenna height. If the CSV file has HAE APC in it, just put zeros in for the antenna height and phase center offset.

I still say (1) and (2) above are best on separate screens or sections of the UI.  They are very different use cases, and the calculations are different.

---

I never liked the button label wording  "Paste current LLh"  -  "paste" implies that the coordinates have been copied by the user from somewhere else and are in some copy/paste/clipboard/buffer.   I think it would be clearer if the button was labeled "Load LLh from ZED" or "Copy LLh from ZED".   This makes it clearer where the coordinates are coming from.   I like "Load...." because we can either load from ZED or load from file, "Copy..." is good too.

amlago, 2023-03-05:
Summary

Taking into account your opinions, which I respect a lot, I will try to make a summary to see if you agree and that Nathan can finish the programming.

1) **I need the check Adjust mark. For manual entries it is very good and avoids manual height calculations.**
I was already working with the Adjust mark check over the weekend and it was very useful.

2) **When I load coordinates with the Paste current Llh button, the Adjust Mark check should be checked automatically and the h should be loaded into the HAE APC box and calculate the HAE Mark**

3) **When we load coordinates from the list of coordinates saved in the file, the Adjust mark check should be unchecked automatically and the h should be loaded into the HAE Mark box and calculate the HAE APC.
That's because the coordinate file always contains the HAE Mark.**

I think this simplifies what I had written in previous posts and takes Tony's comments in part.

Greetings

P.D.    I am at UTC -3.
Tony, what UTC are you in? pure curiosity.

tonycanike, 2023-03-05:
Hi, 
If you need the check box, I'll believe you.    Ok.

How about:
a.  We use radio buttons instead of the check box. The radio buttons state affirmatively what's going to happen for either option so it is very clear to users.

             O Calculate HAE Mark         O Calculate HAE APC 

b.  Exactly one and only one radio button is selected at any time. 
c.   Do we need a "Calculate Nothing" radio button?

                 O Calculate HAE Mark         O Calculate HAE APC         O Calculate Nothing

d.   The "Paste LLh Coordinates" button is renamed "Load LLh from ZED"
e.   When the coordinates are loaded from the ZED, the "Calculate HAE Mark" radio button is set, the ZED's HAE is loaded into the HAE APC box, and the HAE Mark is calculated.
f.  When the coordinates come from the file, the "Calculate HAE APC" radio button is set, the file's HAE is loaded into  the HAE Mark box, and the HAE APC is calculated.
g.  There's a [Recalculate] button so people can correct their antenna height and phase center offset and recalculate.
                  O Calc HAE Mark      O Calc HAE APC 
                    [Recalculate]

h.  No antenna heights, no phase center offsets, no radio button settings are stored in the coordinate file.
i.  if "Calc HAE Mark" is selected, the HAE Mark field is not editable.  if "Calc HAE APC" is selected, the HAE APC field is not editable.  Other fields are editable.
j.   if the firmware can determine the antenna offset (eg Facet or Facet-L) the antenna offset is automatically pre-filled but it is still editable.  

k.  Can the UI elements be reordered so the are presented top-to-bottom in the order the user needs to manipulate them?  Not sure this is easy.   #330 
Perhaps:
  i. User enters antenna height  (do this first because the user just measured it and the number is floating around in their head, about to be forgotten.)
  ii. User enters phase center offset  (prefilled if Facet or Facet-L)
  iii. User chooses source of coordinates (CSV file, ZED, manual entry)
  iv.  UI adjusts automagically!!    The calculate radio buttons appropriately selected but can be changed.
  v.  User hits the "Load and Calculate" button.  Data is loaded from selected source and calculations performed.
  vi.  User reviews displayed results.  LL and the h's top-to-bottom to intuitively match real-world order:   HAE APC, phase center offset, antenna height, HAE Mark.)
  vii.  If necessary, user adjusts their entries and uses the "Load and Calculate" button again.
  viii.  User saves and exits, RTK device restarts.

Cheers, 

Tony.

PS UTC-5

amlago, 2023-03-05:
You have raised interesting things and I think some of them will be for the future and I don't even know if they can be deprogrammed on a web page.
But in general we can come to agree.

1) We replace the Adjust mark check with two radio buttons, one for Calculate HAE Mark and another for calculating HAE APC and only one can be selected.

2) If Calculate HAE Mark radio button is selected, the h is loaded in the HAE APC box
3) If Calculate HAE APC radio button is selected, h is loaded in the HAE Mark box
4) The "Paste LLh Coordinates" button is renamed "Load LLh from ZED" .
5) When the coordinates are loaded from the ZED (button "Load LLh from ZED"), the "Calculate HAE Mark" radio button is set, the ZED's HAE is loaded into the HAE APC box, and the HAE Mark is calculated.
6) When the coordinates come from the file, the "Calculate HAE APC" radio button is set, the file's HAE is loaded into the HAE Mark box, and the HAE APC is calculated.
7) There's a [Recalculate] button so people can correct their antenna height and phase center offset and recalculate.
8) if "Calc HAE Mark" is selected, the HAE Mark field is not editable. if "Calc HAE APC" is selected, the HAE APC field is not editable. Other fields are editable.
9) I like that the antenna heights and ARP are saved to the file, even if I change them later. That they continue to be saved would be my opinion.

If this summary is programmed, I am very satisfied.

Greetings
Angel

tonycanike, 2023-03-05:
Angel, I think we can agree on that!

It is useful to have the antenna height and phase center ARP offset that was used in the field saved somewhere so we can check the data later back in the office.  I'm OK with putting it in the coordinate file, though it's not my first choice.

Thank you for this good conversation!

Regards,

Tony.

amlago, 2023-03-05:
Excellent. It is a pleasure for me to exchange ideas.
Always at your service to talk about any topic and you always learn.
Keep in touch.

Greetings

Angel

amlago, 2023-03-05:
@nseidle 

 For Nathan
I copy the link of the result, of a very interesting conversation and exchange of ideas with Tony.
So you don't get lost among so many messages.

https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/417#issuecomment-1455195308

Greetings
Angel

nseidle, 2023-03-07:
Thanks to you both! Closing this out. #418 has the summary.

closed 2023-03-07 by nseidle