sparkfun/SparkFun_RTK_Firmware #418: Small Clarifying Changes to Base Geodetic Configuration (AP/Wifi)

tonycanike, 2023-03-07:
### Subject of the issue
In a long-running commentary on #417, @amlago and I debated, discussed, and agreed on small changes that would significantly clarify and improve the AP/Wifi Geodetic Base configurtion:

I'm opening a separate issue to document and request those changes.   #417 is already labelled as being in the RC and I didn't want the subsequent discussing and agreement in #417 to get missed.

### Your workbench
3.1 & 3.2 RC

### Requested Changes in https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/417#issuecomment-1455195308

But in general we can come to agree.

1.     We replace the Adjust mark check with two radio buttons, one for Calculate HAE Mark and another for calculating HAE APC and only one can be selected.
2.     If Calculate HAE Mark radio button is selected, the h is loaded in the HAE APC box
3. If Calculate HAE APC radio button is selected, h is loaded in the HAE Mark box
4. The "Paste LLh Coordinates" button is renamed "Load LLh from ZED" .
5. When the coordinates are loaded from the ZED (button "Load LLh from ZED"), the "Calculate HAE Mark" radio button is set, the ZED's HAE is loaded into the HAE APC box, and the HAE Mark is calculated.
6.  When the coordinates come from the file, the "Calculate HAE APC" radio button is set, the file's HAE is loaded into the HAE Mark box, and the HAE APC is calculated.
7.  There's a [Recalculate] button so people can correct their antenna height and phase center offset and recalculate.
8. if "Calc HAE Mark" is selected, the HAE Mark field is not editable. if "Calc HAE APC" is selected, the HAE APC field is not editable. Other fields are editable.
9. I like that the antenna heights and ARP are saved to the file, even if I change them later. That they continue to be saved would be my opinion.
10. If it were simple, the ARP could be filled in automatically, depending on whether Facet or Facet-L is used.

If this summary is programmed, I am very satisfied.

Greetings
Angel

nseidle, 2023-03-07:
@amlago @tonycanike - You two are amazing. Thank you for collaborating, debating, and arriving at a good solution. Also, thank you for the clear work orders :)

> We replace the Adjust mark check with two radio buttons, one for Calculate HAE Mark and another for calculating HAE APC and only one can be selected.

Done

> If Calculate HAE Mark radio button is selected, the h is loaded in the HAE APC box. If Calculate HAE APC radio button is selected, h is loaded in the HAE Mark box

Loaded when? When the user clicks 'Load LLh from ZED'? Or Load from the commonly used table? I will assume both.

> The "Paste LLh Coordinates" button is renamed "Load LLh from ZED" .

Done.

> When the coordinates are loaded from the ZED (button "Load LLh from ZED"), the "Calculate HAE Mark" radio button is set, the ZED's HAE is loaded into the HAE APC box, and the HAE Mark is calculated. When the coordinates come from the file, the "Calculate HAE APC" radio button is set, the file's HAE is loaded into the HAE Mark box, and the HAE APC is calculated.

Done.

> There's a [Recalculate] button so people can correct their antenna height and phase center offset and recalculate.

Is this needed? The disabled field (HAE APC or HAE Mark, depending on radio button state) is recalculated with every change.

[image]

> if "Calc HAE Mark" is selected, the HAE Mark field is not editable. if "Calc HAE APC" is selected, the HAE APC field is not editable. Other fields are editable.

Done.

> I like that the antenna heights and ARP are saved to the file, even if I change them later. That they continue to be saved would be my opinion.

Done.

> If it were simple, the ARP could be filled in automatically, depending on whether Facet or Facet-L is used.

Done.

This will be in the RC shortly. Please let me know what else needs to change.

amlago, 2023-03-08:
Hi @nseidle 
Thank you for your very positive comments. It encourages further discussion on firmware improvements.
Always on command.

Regarding the changes to the base configuration, as soon as I can test them I will inform you.

Greetings
Angel

tonycanike, 2023-03-08:
Excellent!  That was fast work @nseidle!

I concur with your clarifications/decisions and I don't believe the recalculate button is necessary.

amlago, 2023-03-08:
I just tried the new firmware.
I think that point 5) and 6) still do not automatically activate the corresponding  button radio.

"5-When the coordinates are loaded from the ZED (button "Load LLh from ZED"), the "Calculate HAE Mark" radio button is set, the ZED's HAE is loaded into the HAE APC box, and the HAE Mark is calculated."

"6-When the coordinates come from the file, the "Calculate HAE APC" radio button is set, the file's HAE is loaded into the HAE Mark box, and the HAE APC is calculated."

amlago, 2023-03-08:
Clarifications on points 5 and 6.
We don't want two possible button radio options for the Load LLh from ZED button and also for the Load button when using file coordinates. The button radios to be marked are opposite and unique.

If we use the Load LLh from Zed button, the Calculate HAE Mark-(Point 5) radio button should be checked automatically

b) If we use the Load button for file coordinates, the Calculate HAE APC radio button should be checked automatically (Point 6)

Greetings

amlago, 2023-03-08:
I think if we have two radio button options when loading coordinates, it may be misleading for less experienced people.
At the beginning I liked the idea of having the two radio buttons enabled for the option to load coordinates with the Load from file button, but in conversations with Tony I came to the conclusion that confusion could be generated.

nseidle, 2023-03-08:
> 5-When the coordinates are loaded from the ZED (button "Load LLh from ZED"), the "Calculate HAE Mark" radio button is set, the ZED's HAE is loaded into the HAE APC box, and the HAE Mark is calculated.

> 6-When the coordinates come from the file, the "Calculate HAE APC" radio button is set, the file's HAE is loaded into the HAE Mark box, and the HAE APC is calculated.

Ok, I think I've got this functionality working. Please check my work.

amlago, 2023-03-09:
@nseidle 
Nathan.
It worked. Excellent.

@tonycanike 
Tony. Try it when you can. It will make my work in the field much easier.

Greetings

tonycanike, 2023-03-09:
I look forward to trying all this amazing work out!  Thank you!  I'm heading into the field next week and will try it then.

XPav, 2023-03-13:
I'm looking at the changes in 3.2, and I'm a little unclear exactly which heights we're talking about.   Is there a diagram or ascii art or something that I can refer too to make sure I'm going the right directions and putting in the right numbers?

tonycanike, 2023-03-13:
^
 |  up

--+--  HAE of the antenna phase center (APC)

--+--  HAE of the antenna reference point (ARP)

--+--    HAE of the Mark (eg survey marker in the ground)

  |  down
  V

We generally want to know the height above the ellipsoid (HAE) of the survey marker in the ground. 

GNSS receivers generally calculate and report the height of the antenna phase center (APC).  We can not directly measure the APC in the field. 

As we care about the HAE of the Mark and not the HAE of the APC, we need to carefully calculate the HAE of the Mark from the HAE of the APC when taking a GNSS reading. 

Conversely, if (for RTK) we are setting up a base over a mark, we usually already know the HAE of the Mark and need to calculate the HAE of its APC and tell the base what it is. 

GNSS antennas have an antenna reference point (ARP) that we can measure - we can hold a measuring tape up to it in the field. This is a physical location on the exterior of the antenna.  It's usually on the very bottom where the antenna mounts onto a surveyor tripod or pole. 

GNSS antennas have a published offset from the ARP to the APC. It's usually in the range of a few centimeters. The APC is usually above the ARP so this is usually a positive number. I call this the "ARP to APC offset" formally but there are many shorthand names for it. 

Surveyors usually mount their GNSS antennas precisely above the survey marker on a pole or tripod. Surveyors then measure the vertical distance from the marker to the ARP.  Surveyor generally formally call this the "HI" - height of instrument.  Informal names include "Rod Height" or "antenna height". This distance is usually between 1.5 and 2m in general practice. 

HAE Mark + HI + ARP to APC offset = HAE APC

XPav, 2023-03-13:
Perfect, thanks for the diagram and detailed explanation about what's what.

nseidle, 2023-05-05:
This was implemented in [v3.3 of the RTK firmware](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v.3.3). Please let us know if you see any problems.

closed 2023-05-05 by nseidle