sparkfun/SparkFun_RTK_Firmware #422: Rate of RTCM corrections out of the RADIO port.

rrodas64, 2023-03-08:
Hello
I am testing RTK with LORA radios and due to its low air speed rate I would like to be able to select which messages are transmitted by radio as well as their delivery rate.

I quote what is in the manual:
''By default, this is one per second. During this phase, the ZED-F9P is outputting the RTCM corrections out of the RADIO port. Attaching an external serial radio to this port will allow the Base to send corrections to any Rover.''

Due to this I have interpreted that the messages are delivered every second, in order to facilitate the use of LORA radios there should be the feature of being able to change that transmission rate.

nseidle, 2023-03-09:
This is fine idea but there are LoRa radios capable of RTCM at 1Hz. What radios are you using? Can you not increase the airspeed to facilitate 1Hz transmissions at 530 bytes?

We could allow lower RTCM output but that may lead to problems with the Rover. We've done all our testing at 1Hz but I've seen RTK Fix last with no RTCM for ~15s (then RTK Float begins).

rrodas64, 2023-03-09:
i'm testing the EBYTE E32-900T30D radio. (https://www.ebyte.com/en/product-view-news.html?id=1525). Could you please inform where I can find information about 530 bytes is enought to transmit messages for the 4 sathelite constalations?
Since signal range is inversely proportional to Airspeed, I would like to be able to adjust the frequency of RTCM messages to achieve the highest possible range.
In order to achieve this, other manufacturers allow you to configure the equipment for example 1005(0.1HZ), 1074(0.5HZ), 1084(0.5HZ), 1094(0.5HZ), 1124 (0.5HZ), 1230(0.1HZ)

nseidle, 2023-03-09:
Ah, yes we can do that. You'll need to modify the message rates as well. 

Below is WIP. Will hit the release candidate soon.

[image]

Currently the message rates for RTCM are set to 1 (and 10 for 1230) but we'll fix that shortly. You can set message rates under the GNSS menu:

[image]

nseidle, 2023-03-10:
The fix rate was the wrong approach. We ended up adding settings for all RTCM rates. You can now individually set the RTCM messages when the device enter Base mode. The device will still follow the GNSS message rates when in Rover mode.

[image]

I've set two 'defaults'. 1Hz for 1005/74/84/94/124 + 10Hz for 1230, and 'low bandwidth' that is 1Hz for 1074/84/94/124 + 10Hz for 1005/1230.

[image]

These can also be configured through the serial menu.

Please give the latest RC a try and let us know.

rrodas64, 2023-03-10:
Excelent!!!. I'll test it.

tonycanike, 2023-03-10:
"The device will still follow the GNSS message rates when in Rover mode."

Is this meant to imply that the device will _not_ follow the GNSS message rates when in Base mode?

Depending on my plans for post-processing the base logs, I might only need the NMEA and RXM messages every 10 or even 30 seconds (eg OPUS) but generally I would like the RTCM more frequently.

nseidle, 2023-03-10:
> Nice, that was fast work!

Thanks! @rrodas64 found a feature that *I* needed as well, so it gets implemented quick :)

> Is this meant to imply that the device will not follow the GNSS message rates when in Base mode?

In Rover mode the device will follow the message rates set under the GNSS menu. In base mode the device will follow all rover GNSS messages except the Base RTCM message rates. 

> Depending on my plans for post-processing the base logs, I might only need the NMEA and RXM messages every 10 or even 30 seconds (eg OPUS) but generally I would like the RTCM more frequently.

This is a good use case. I believe this will work as is with one caveat. In base mode the fix rate is 1Hz (we could make this a setting but I really don't want to add more to the WiFi Config page). So if you need 10s logging but 1Hz RTCM, you would set the GNSS messages you want to log to 10, and leave the RTCM messages to default (1 for most). That way RTCM is generated at 1 Hz, send out RADIO port at 1Hz, logged at 1Hz, but your NMEA and RXM is logged at 0.1Hz. Clear as mud?

tonycanike, 2023-03-10:
Sounds good! Thanks, that should work perfectly.  

I do sometimes confuse what goes in the config box next to each msg type (msg/sec, sec/msg, msg/fix or fix/msg).  It's documented so no worries, it's just me. :)

tonycanike, 2023-03-15:
FYI, Mark Silver did a blog post on RTCM & data rates recently.

https://blog.ashgps.com/2023/03/14/to-fec-or-not-to-fec/

nseidle, 2023-05-05:
This was implemented in [v3.3 of the RTK firmware](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v.3.3). Please let us know if you see any problems.

closed 2023-05-05 by nseidle

nseidle, 2023-05-05:
[Docs added](https://docs.sparkfun.com/SparkFun_RTK_Firmware/configure_base/#rtcm-message-rates).