sparkfun/SparkFun_RTK_Firmware #424: Power on RTK Express overwrites U-Blox configuration

j-w-bullfrog, 2023-03-10:
### Power on RTK Express overwrites U-Blox configuration

[image]

### Your workbench
3.1 express + and express in base. 

### Steps to reproduce
Power cycle, the layer 7 default overwrites the flash, ram, & bbr. Default is not settable in the u-box s/w. The image above is after I connnect with u-blox and flash a config file

### IMU can calibrate
The imu in mower mode should be able to calibrate going less than a 1 mph. Automobile mode requires higher speed than a mower can run.

### Actual behavior
Sensor fusion never calibrates

tonycanike, 2023-03-10:
Yes, that's the documented behavior, for better or worse. 

https://docs.sparkfun.com/SparkFun_RTK_Firmware/configure_ucenter/

     "However, because the ESP32 does considerable configuration of the ZED-F9P at power on it is not recommended to modify the settings of the ZED-F9P using u-center. Nothing will break but your changes will likely be overwritten at the next power cycle."

nseidle, 2023-03-10:
What @tonycanike  said. The RTK firmware very much overwrites the configuration of the u-blox module. That's by design.

Which part of the u-blox configuration are you trying to set? Perhaps we can build that into RTK Firmware.

j-w-bullfrog, 2023-03-10:
Specifically CFG-NAVSPG-DYNMODEL  11 (mower) however, that requires that you manually set, the imu orientation.

# Config changes format version 1.0
# created by u-center version 22.07 at 15:37:00 on Friday, 10 Mar 2023
[del]
[set]
Flash CFG-NAVSPG-DYNMODEL  11                   # write value 11 - MOWER           to item id 0x20110021 in layer 2
Flash CFG-SFIMU-IMU_MNTALG_YAW 0x0                  # write value 0  0x0               to item id 0x4006002d in layer 2
Flash CFG-SFIMU-IMU_MNTALG_PITCH 0x0                  # write value 0  0x0               to item id 0x3006002e in layer
Flash CFG-SFIMU-IMU_MNTALG_ROLL 0x0                  # write value 0  0x0               to item id 0x3006002f in layer 
Flash CFG-SFODO-DIS_AUTODIRPINPOL 1                    # write value 1                    to item id 0x10070005 in laye
Flash CFG-SFODO-COMBINE_TICKS 0                    # write value 0                    to item id 0x10070001 in layer 2
Flash CFG-RATE-NAV_PRIO    0xa                  # write value 10  0xa              to item id 0x20210004 in layer 2
Flash CFG-NAV2-OUT_ENABLED 1                    # write value 1                    to item id 0x10170001 in layer 2
Flash CFG-SFIMU-AUTO_MNTALG_ENA 0                    # write value 0                    to item id 0x10060027 in layer 
Flash CFG-SFODO-USE_SPEED  1                    # write value 1                    to item id 0x10070003 in layer 2
  RAM CFG-NAVSPG-DYNMODEL  11                   # write value 11 - MOWER           to item id 0x20110021 in layer 0
  BBR CFG-NAVSPG-DYNMODEL  11                   # write value 11 - MOWER           to item id 0x20110021 in layer 1

nseidle, 2023-03-10:
Sounds like you need #423 plus the ability to specify the IMU orientation?

j-w-bullfrog, 2023-03-10:
Yep, since I'm the author of both of them. BTW J.W.Bullfrog and btreichel
are both me. my github account is just that old. Of course I'm old
enough to know that Jerimiah Wazza Bullfrog! You had to be on gmail REAL
early to get that one.

https://www.linkedin.com/in/bentreichel

¸ ...      ___________
·´º o`·, /__/ _/\_ ____/\
```)¨(´´´ | | | | |  | | || |l±±±±|
 °´ ¸,.-·~·~·-.,¸ `°·-. :º° ·~·~·-..,¸
As for me and my house, we will serve the Lord! (Joshua 24:15)

On Fri, Mar 10, 2023 at 5:31 PM Nathan Seidle ***@***.***>
wrote:

> Sounds like you need #423
> <https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/423> plus the
> ability to specify the IMU orientation?
>
> —
> Reply to this email directly, view it on GitHub
> <https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/424#issuecomment-1464639936>,
> or unsubscribe
> <https://github.com/notifications/unsubscribe-auth/AH2UE23VLB477AHVCYPIC5TW3O2ULANCNFSM6AAAAAAVXAJ3HI>
> .
> You are receiving this because you authored the thread.Message ID:
> ***@***.***>
>

j-w-bullfrog, 2023-03-11:
BTW, I'm not touching the express I'm using for a mobile correction sender; via 915 mhz radio pair, I'm talking about the express + (aka zed-f9R). Thats the one that has the sensor fusion / dead reckoning in it.  Which is why I need the https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/423 to fix this. These are dynamic models added by u-blox in sept 2022. I've used the overwrite on boot myself several times so I understand why.

nseidle, 2023-03-12:
CFG-NAVSPG-DYNMODEL - This setting is covered in #423
CFG-SFIMU-AUTO_MNTALG_ENA - This setting is already available in the Sensor Fusion menu

New settings that need to be added:

    CFG-SFIMU-IMU_MNTALG_YAW
    CFG-SFIMU-IMU_MNTALG_PITCH
    CFG-SFIMU-IMU_MNTALG_ROLL
    CFG-SFODO-DIS_AUTODIRPINPOL
    CFG-SFODO-COMBINE_TICKS
    CFG-RATE-NAV_PRIO
    CFG-NAV2-OUT_ENABLED
    CFG-SFODO-USE_SPEED

j-w-bullfrog, 2023-03-12:
Excellent, the 5 bottom settings are if you are not (or are) going to use
wheel tick sensors, but speed from the gnss instead. Yes, it does impact
accuracy (slightly), especially if you are running dead reckoning with out
a gnss signal (aka tunnel). But then, I don't think that tunnels need to be
mowed.;)
\
https://www.linkedin.com/in/bentreichel

¸ ...      ___________
·´º o`·, /__/ _/\_ ____/\
```)¨(´´´ | | | | |  | | || |l±±±±|
 °´ ¸,.-·~·~·-.,¸ `°·-. :º° ·~·~·-..,¸
As for me and my house, we will serve the Lord! (Joshua 24:15)

On Sun, Mar 12, 2023 at 2:32 PM Nathan Seidle ***@***.***>
wrote:

> CFG-NAVSPG-DYNMODEL - This setting is covered in #423
> <https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/423>
> CFG-SFIMU-AUTO_MNTALG_ENA - This setting is already available in the
> Sensor Fusion menu
>
> New settings that need to be added:
>
> CFG-SFIMU-IMU_MNTALG_YAW
> CFG-SFIMU-IMU_MNTALG_PITCH
> CFG-SFIMU-IMU_MNTALG_ROLL
> CFG-SFODO-DIS_AUTODIRPINPOL
> CFG-SFODO-COMBINE_TICKS
> CFG-RATE-NAV_PRIO
> CFG-NAV2-OUT_ENABLED
> CFG-SFODO-USE_SPEED
>
> —
> Reply to this email directly, view it on GitHub
> <https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/424#issuecomment-1465280652>,
> or unsubscribe
> <https://github.com/notifications/unsubscribe-auth/AH2UE25TV3GLPQRU2RUDVM3W3YXEBANCNFSM6AAAAAAVXAJ3HI>
> .
> You are receiving this because you authored the thread.Message ID:
> ***@***.***>
>

nseidle, 2023-03-12:
These 8 new settings have been implemented into the Sensor Fusion serial menu. It will be a bit before we get them into the WiFi Config page.

Again, this is untested (but should work!) until I can get to F9R hardware later this week.

j-w-bullfrog, 2023-03-12:
Cool, while I have a known stable version on the F9R, and the weather is
too cold and wet, I'll work on getting a static base and the rtk ntrip
caster going.

https://www.linkedin.com/in/bentreichel

¸ ...      ___________
·´º o`·, /__/ _/\_ ____/\
```)¨(´´´ | | | | |  | | || |l±±±±|
 °´ ¸,.-·~·~·-.,¸ `°·-. :º° ·~·~·-..,¸
As for me and my house, we will serve the Lord! (Joshua 24:15)

On Sun, Mar 12, 2023 at 4:57 PM Nathan Seidle ***@***.***>
wrote:

> These 8 new settings have been implemented into the Sensor Fusion serial
> menu. It will be a bit before we get them into the WiFi Config page.
>
> Again, this is untested (but should work!) until I can get to F9R hardware
> later this week.
>
> —
> Reply to this email directly, view it on GitHub
> <https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/424#issuecomment-1465310176>,
> or unsubscribe
> <https://github.com/notifications/unsubscribe-auth/AH2UE22FNJUEAKDNK2GHCH3W3ZIFRANCNFSM6AAAAAAVXAJ3HI>
> .
> You are receiving this because you authored the thread.Message ID:
> ***@***.***>
>

nseidle, 2023-04-14:
We've reworked how messages and commands are supported on different platforms (F9R, F9P) and different firmware version (v1.12, 1.13, 1.20, 1.30, 1.32, etc). It's all working well. I believe this issue has been taken care of. 

[image]

Please give the latest release candidate a try and let me know if you see any problems.

j-w-bullfrog, 2023-04-14:
Cool, I'll check it out. I was playing with it when I had time and it
seemed to be choking on some UBX messages.

https://www.linkedin.com/in/bentreichel

¸ ...      ___________
·´º o`·, /__/ _/\_ ____/\
```)¨(´´´ | | | | |  | | || |l±±±±|
 °´ ¸,.-·~·~·-.,¸ `°·-. :º° ·~·~·-..,¸
As for me and my house, we will serve the Lord! (Joshua 24:15)

On Fri, Apr 14, 2023 at 5:23 PM Nathan Seidle ***@***.***>
wrote:

> We've reworked how messages and commands are supported on different
> platforms (F9R, F9P) and different firmware version (v1.12, 1.13, 1.20,
> 1.30, 1.32, etc). It's all working well. I believe this issue has been
> taken care of.
>
> [image: image]
> <https://user-images.githubusercontent.com/117102/232164363-36f27ef7-5ff2-4d7a-b036-cadb54bd6583.png>
>
> Please give the latest release candidate a try and let me know if you see
> any problems.
>
> —
> Reply to this email directly, view it on GitHub
> <https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/424#issuecomment-1509354077>,
> or unsubscribe
> <https://github.com/notifications/unsubscribe-auth/AH2UE2ZZXXX7GRV73GTX7GTXBHE7VANCNFSM6AAAAAAVXAJ3HI>
> .
> You are receiving this because you authored the thread.Message ID:
> ***@***.***>
>

j-w-bullfrog, 2023-04-17:
I've tested many reboots w/o a failure. Cycled thru several iterations of different msg's (mostly all nema off). Did config via web interface, and serial config.  NAV PVAT is available in serial, but not web. Really don't care, just noting.

closed 2023-04-17 by j-w-bullfrog