sparkfun/SparkFun_RTK_Firmware #425: Add to the Gnss configuration the minimum value of C/N0 (db-Hz)

amlago, 2023-03-13:
Facet v3.2
Add to the Gnss configuration the value of C/N0 which is an indicator of the quality of the signal for the satellite to use (db-Hz)

1) I connected my Facet to u-center and I found that the minimum value of C/N0 to use satellites was 6 db-Hz.
I found it in the u-center at UBX-CFG-NAVX5
2) This is a very small value and forces the Zed-f9p to calculate position with satellites whose signals are degraded, due to crossing foliage of trees, walls, etc.
3) Try to raise the signal to 30 db-Hz and indeed there are many fewer satellites for the calculation.
4) This at first may seem bad, but it is actually good, when using the satellites with the best signal to determine the position.
I think it can help make the precision displayed on the screen more realistic. The latter is an assumption.

Request: I would like to add to the Gnss configuration the entry of the minimum value of C/N0 in db-Hz above which the satellites would be used. Recommended values would be 35 db-Hz

Greetings

nseidle, 2023-03-13:
If you start looking at all the settings in u-center you will make work for me for many years :)

I think this is a fine setting to add. However, I would consider this setting an advanced setting.

I need help finding a way to not overwhelm beginning users. Perhaps we add a checkbox called 'Advanced Settings'. When it is checked it would show all settings. What do you think?

amlago, 2023-03-13:
Hello.
I understand what your saying to me. I send you my apologies for filling you with extra work. (Applause for Nathan.)
I am very grateful for the very good care you give us.
Now:

1) I think enabling all settings may be more than what is needed.

2) Yes, I would like to put together in a separate box what we added a few days ago about elevation mask (minimum elevation angle) and what I requested about C/N0 (db-Hz).
Between the two variables I'm talking about, the satellites to use are filtered.
They would be filtered by elevation angle and by signal quality.

Greetings

nseidle, 2023-04-14:
I'm working on this. The F9R has a default of 20dBHz, the F9P has a default of 6 dBHz:

[image]

[image]

So we will make the setting and the defaults platform dependent.

nseidle, 2023-04-14:
[image]

In the RC. Please give it a try.

amlago, 2023-04-14:
Hello.
 I come from measuring in the field. I'll update tomorrow and try it.
Filtering by C/N0 can be useful, for example, when measuring in a city with tall buildings, where signal bounces generate erroneous positions.
As always very good job.
Thank you

amlago, 2023-04-15:
Hello, I could not test the modifications because with v3.3 of 04/14 it does not let me enter the Wi-Fi configuration.
It starts connecting to the Wi-Fi network and reboots.
So several times. I went back to v3.2 and entered the wifi config without problems.

nseidle, 2023-05-05:
This was implemented in [v3.3 of the RTK firmware](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v.3.3). Please let us know if you see any problems.

closed 2023-05-05 by nseidle

nseidle, 2023-05-05:
[Docs added](https://docs.sparkfun.com/SparkFun_RTK_Firmware/configure_gnss/#min-sv-elevation-and-cn0).