sparkfun/SparkFun_RTK_Firmware #429: Facet in Base will not restart casting if WiFI network disappears and reappears

XPav, 2023-03-30:
Using Release firmware 3.2.   

Standard Base setup -- a Facet connected to WiFi with casting to a local NTRIP Server (SNIP).  I turn it on, the Facet starts Casting, SNIP sees them, all good.

If I remove the WiFI network, then put it back, leaving the Facet on, the serial terminal shows that the Facet reconnected to Wi-Fi, but the Facet never goes back into "Casting" mode.

On the good side, if I remove just the NTRIP Server and then put it back (using SNIP, I just closed it and reopened it), after a few minutes, the Facet does start Casting again to SNIP.  

I would expect that the Facet would attempt Casting again after reconnecting to a Wi-Fi network.

nseidle, 2023-04-18:
Your expectations are correct: by design, NTRIP and WiFi should attempt auto-reconnect. I'm seeing that it's working (v3.2):

    11:59:37.024 -> WiFi IP address: 192.168.222.137 RSSI: -26
    11:59:45.888 -> Batt (119%): Voltage: 4.16V Discharging: -11.65%/hr Green
    11:59:45.934 -> Log file size: 87887 - Generation rate: 3.8kB/s
    11:59:45.980 -> WiFi link lost
    11:59:45.980 -> WIFI_CONNECTING
    11:59:45.980 -> Connection to NTRIP Caster was lost
    11:59:45.980 -> NTRIP_SERVER_ON
    11:59:46.026 -> Connecting WiFi... WiFi failed to connect: error #5. Next WiFi attempt in 15 seconds.
    11:59:50.891 -> Batt (119%): Voltage: 4.16V Discharging: -15.81%/hr Green
    11:59:50.985 -> Log file size: 98703 - Generation rate: 2.2kB/s
    11:59:55.941 -> Batt (119%): Voltage: 4.16V Discharging: -15.81%/hr Green
    11:59:55.988 -> Log file size: 108100 - Generation rate: 1.9kB/s
    12:00:00.911 -> Batt (119%): Voltage: 4.16V Discharging: -19.97%/hr Green
    12:00:00.958 -> NTRIP_SERVER_WIFI_STARTED
    12:00:00.958 -> Log file size: 117439 - Generation rate: 1.9kB/s
    12:00:01.006 -> Connecting WiFi... WiFi failed to connect: error #5. Next WiFi attempt in 30 seconds.
    12:00:05.951 -> Batt (119%): Voltage: 4.19V Discharging: -14.98%/hr Green
    12:00:05.997 -> Log file size: 126802 - Generation rate: 1.9kB/s
    12:00:10.990 -> Log file size: 137370 - Generation rate: 2.1kB/s
    12:00:10.990 -> Batt (119%): Voltage: 4.19V Discharging: -14.98%/hr Green
    12:00:15.988 -> Log file size: 146907 - Generation rate: 1.9kB/s
    12:00:15.988 -> Batt (119%): Voltage: 4.19V Discharging: -19.14%/hr Green
    12:00:20.987 -> Log file size: 156516 - Generation rate: 1.9kB/s
    12:00:21.033 -> Batt (119%): Voltage: 4.19V Discharging: -19.14%/hr Green
    12:00:25.985 -> Log file size: 165881 - Generation rate: 1.9kB/s
    12:00:26.032 -> Batt (119%): Voltage: 4.19V Discharging: -19.14%/hr Green
    12:00:31.034 -> Connecting WiFi... WiFi failed to connect: error #5. Next WiFi attempt in 45 seconds.
    12:00:32.904 -> Batt (119%): Voltage: 4.19V Discharging: -23.50%/hr Green
    12:00:32.950 -> Log file size: 179295 - Generation rate: 2.7kB/s
    12:00:37.907 -> Batt (119%): Voltage: 4.19V Discharging: -23.50%/hr Green
    12:00:37.954 -> Log file size: 188630 - Generation rate: 1.9kB/s
    12:00:42.920 -> Batt (119%): Voltage: 4.19V Discharging: -27.66%/hr Green
    12:00:42.966 -> Log file size: 199193 - Generation rate: 2.1kB/s
    12:00:47.922 -> Batt (119%): Voltage: 4.19V Discharging: -27.66%/hr Green
    12:00:47.968 -> Log file size: 208600 - Generation rate: 1.9kB/s
    12:00:52.919 -> Batt (119%): Voltage: 4.13V Discharging: -24.96%/hr Green
    12:00:52.965 -> Log file size: 217966 - Generation rate: 1.9kB/s
    12:00:57.960 -> Batt (119%): Voltage: 4.13V Discharging: -24.96%/hr Green
    12:00:58.007 -> Log file size: 227536 - Generation rate: 1.9kB/s
    12:01:02.970 -> Batt (119%): Voltage: 4.13V Discharging: -24.96%/hr Green
    12:01:03.063 -> Log file size: 238531 - Generation rate: 2.2kB/s
    12:01:07.975 -> Batt (118%): Voltage: 4.13V Discharging: -33.49%/hr Green
    12:01:08.069 -> Log file size: 248138 - Generation rate: 1.9kB/s
    12:01:12.972 -> Batt (118%): Voltage: 4.13V Discharging: -33.49%/hr Green
    12:01:13.066 -> Log file size: 257965 - Generation rate: 2.0kB/s
    12:01:16.010 -> Connecting WiFi... 
    12:01:22.587 -> WIFI_CONNECTED
    12:01:22.587 -> WiFi IP address: 192.168.222.137 RSSI: -31
    12:01:22.587 -> NTRIP_SERVER_WIFI_CONNECTED
    12:01:22.587 -> Batt (118%): Voltage: 4.13V Discharging: -37.65%/hr Green
    12:01:22.635 -> Log file size: 277149 - Generation rate: 3.8kB/s
    12:01:22.635 -> NTRIP_SERVER_WAIT_GNSS_DATA
    12:01:23.007 -> NTRIP_SERVER_CONNECTING
    12:01:23.054 -> NTRIP Server connecting to rtk2go.com:2101
    12:01:23.148 -> NTRIP Server connected
    12:01:23.148 -> NTRIP_SERVER_AUTHORIZATION
    12:01:23.986 -> NTRIP Server connected to rtk2go.com:2101 bldr_dwntwn2
    12:01:23.986 -> NTRIP_SERVER_CASTING
    12:01:28.566 -> Batt (118%): Voltage: 4.13V Discharging: -37.65%/hr Green
    12:01:28.612 -> Log file size: 288192 - Generation rate: 2.2kB/s
    12:01:34.780 -> Batt (118%): Voltage: 4.13V Discharging: -41.81%/hr Green

I ran this test by connecting to an AP, then unplugging it. The AP I have takes a long time (~120s) to begin running again so the ESP tried 5 wifi reconnects before success.

nseidle, 2023-04-24:
We think we've found some areas where the NTRIP Server and Client could shut down. We've got some other fixed to put in place first but we're working on this.

nseidle, 2023-10-27:
After lots of hammering, we believe we have this pretty well nailed down starting in v3.9 with more resiliency in v3.10.

nseidle, 2023-10-27:
An oldy, but a goody. This should be fixed in [v3.10](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v3.10).

closed 2023-10-27 by nseidle