sparkfun/SparkFun_RTK_Firmware #431: Issues with Survey Master 2.9 with facet v3.3

amlago, 2023-03-31:
Issues with Survey Master 2.9 with facet v3.3

I work with survey master 2.9 and Facet v3.2.
Everything works very well, it connects quickly and I see the coordinates of the point and the precision given by the Facet on the cell phone screen.

With version 3.3 the Facet connects to Survey master, but the coordinates of the point do not appear and the precision remains as a constant number on the Survey master screen and does not match what was reported by the Facet.
I don't know what changed from version 3.2 to version 3.3

nseidle, 2023-04-24:
I can only get v2.8.2 from [here](https://m.apkpure.com/survey-master/com.sinognss.sm.free). Looks like I could get v2.9 from [here](https://www.comnavtech.com/companyfile/4/) but the v2.8.2 app is telling me I can update to v3.0.0.

[image]

Above: Looks like they got rid of NMEA Device in v3.0.0. I'll try to load v2.9.

Ok, v2.9 operates the same as v2.8. Did you follow [these directions](https://docs.sparkfun.com/SparkFun_RTK_Firmware/gis_software/#survey-master)? I had to remember to turn on four additional NMEA messages to get it to properly report the device location: GGA, GLL, GRS, GSA, GST, GSV, RMC, VTG, ZDA

[image]

As you can see, I was able to get RTK Fix

amlago, 2023-04-25:
Hello.
I try with version 3.3 of 04/24 and update.
Thank you

amlago, 2023-04-25:
With version 3.3 04/24 Survey Master works fine.
It does not show the base coordinates correctly but that problem already existed and it is a problem of the Survey master and not of the Facet.
Thank you

tonycanike, 2023-04-26:
@amlago  do you have a theory about how Survey Master (connected to the Rover) would get the base coordinates to display?  I'm interested in getting SurvPC to know the base coordinates when it's attached to the Rover.   If SurvPC knows the base coordinates a number of very useful features become available - such as correcting survey-in base coordinates with more accurate coordinates back in the office and all the rover gathered coordinates are corrected.

I speculate - it's a wild guess - that perhaps if the rover forwarded the RTCM base position messages (eg 1005) from the base to the data collector then the data collector (SurvPC, Survey Master, etc.) would know the base position.

I wonder what the best way to test this idea would be.

- create a file of $NMEA messages from a rover interleaved with RTCM 1005 from a base and play it back  out to our data collectors? 

- can a Rover ZED be configured to forward certain RTCM it receives from the base on to the data collector?  Is it as 'simple' as configuring the rover ZED to output RTCM 1005 on UART1?  If it's getting RTCM from the radio connected to UART2, for example, would it forward them to UART1?

amlago, 2023-04-26:
Hi, Tony.
I have had this problem that the apps do not recognize the base coordinates in Survey master and in SWMaps.
I do not understand why it is because in the message 1005 are the coordinates.
About SWMaps I wrote to the creator some time ago and in an update it was solved. It now displays the base coordinates correctly.
With respect to Survey master I have not had a solution and I think we will not because in the last update they eliminated the possibility of connecting by nmea device.
Regarding your application, I suppose it is the same problem with respect to 1005.
The apps correctly use the 1005 message to correct the coordinates of the points taken in the field, but they do not show you the base coordinates used.
It seems so simple......
I don't know how to correct it.
I am open to discuss the issue.
Greetings

tonycanike, 2023-04-28:
Hi Angel, how is SWMaps getting the RTCM 1005 messages?

Is this your configuration?

`SparkFun Base --> External Radio --> space  --> External Radio --> SparkFun Rover --> SWMaps`

That's my configuration, except I use SurvPC instead of SWMaps.  I would think the Rover needs to send RTCM 1005 to the SWMaps or SurvPC.

Tony.

tonycanike, 2023-04-28:
So we both have this scenario in common

(RTCM source)  ---> Facet Rover --->  SW Maps or SurvPC

I think we need to ask @nseidle if the Facet can forward RTCM on to our data collector software.