sparkfun/SparkFun_RTK_Firmware #433: settingsFile not found in LittleFS

C4fishfood, 2023-04-08:
I'm using the firmware on a ESP32 Thing plus. I have installed the v3.2 binary using both the GUI, and compiled source from with Arduino app. I need to go into the settings (via serial terminal) switch the bluetooth radio to BLE each time I power the ESP uo in order for my iPhone to discover the GPS in SW maps. When I go to the system config after starting the the ESP the BLE setting remains there as it was before shutdown, but it isn't discoverable by my phone until I scroll through the other bluetooth settings and come back to BLE. 

Also, when I look at the debug when starting the serial connection to the ESP, I see the following (I suspect the littleFS directory needs a file? uploading the settings.h file to the littleFS did not make the error go away or fix the BLE issue):

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
Display not detected
LittleFS Started
SparkFun RTK Surveyor v99.99-Apr  8 2023
settingsFile not found in LittleFS
settingsFile not found in LittleFS
settingsFile not found in LittleFS
settingsFile not found in LittleFS
settingsFile not found in LittleFS
settingsFile not found in LittleFS
settingsFile not found in LittleFS
Profile 'Profile1' loaded
Fuel gauge not detected.
State: Base - Not Started

C4fishfood, 2023-04-08:
I have been able to make the littleFS settings file error go away by going through the menu interface and resetting each profile to the factory default settings. However, the problem with BLE pairing remains- the firmware retains BLE as the Bluetooth setting, but the my phone can't pair to the esp unless I go into the menu each time and cycle through the Bluetooth modes from BLE->classic->BLE

nseidle, 2023-04-14:
> settingsFile not found in LittleFS

This is not uncommon and indicates there is no profile found for that profile slot.

> my phone can't pair to the esp unless I go into the menu each time and cycle through the Bluetooth modes from BLE->classic->BLE

I've tried to replicate this but I can't, it works for me:

Steps I've tried:

* Pair over SPP
* Run SW Maps. See NMEA.
* Forget Bluetooth MAC 5A5A from phone
* Change RTK into BLE mode via Settings menu
* Discover from my phone 5A5A
* Pair to 5A5A
* Open SW Maps, connect via Instrument Model 'Generic NMEA (Bluetooth LE)'
* See NMEA.

I've also used Serial Bluetooth Terminal to verify NMEA flowing over both SPP and BLE.

C4fishfood, 2023-04-14:
Thank you for the reply Nathan. I suspect my problem may have a hardware aspect to it, as I'm using the Thing plus and a F9P breakout rather than one of Sparkfun RTK custom boards with the ESP embedded. Also, Im using an iPhone (which can't support the SPP to my knowledge), so that limits my ability to test the bluetooth classic part.

I have compiled the firmware with most of the functionality removed (commented out pre-processor for compile of ESPnow, L-band, SD_MMC). I can now get the ESP32 to automatically pair with BLE to my iPhone on powerup/restart (using the SWMap app on the phone).

However there is something that is triggering a CPU panic restart every 30 seconds or so. This is the error:

BT client Connected
State: Base-Temp - Settle, 2023-04-14 03:22:10.129
Waiting for Horz Accuracy < 1.00 meters: 1.08
Waiting for Horz Accuracy < 1.00 meters: 1.08
Waiting for Horz Accuracy < 1.00 meters: 1.08
Waiting for Horz Accuracy < 1.00 meters: 1.08
Waiting for Horz Accuracy < 1.00 meters: 1.08
Waiting for Horz Accuracy < 1.00 meters: 1.08
State: Base - Not Started, 2023-04-14 03:22:13.165
*WIFI_OFF
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x00000000  PS      : 0x00060330  A0      : 0x800d3dc0  A1      : 0x3ffd8e40  
A2      : 0x3ffda41c  A3      : 0x3ffed810  A4      : 0x00000000  A5      : 0x3ffc9df4  
A6      : 0x3ffc9e74  A7      : 0x3ffeb46c  A8      : 0x800d3daa  A9      : 0x3ffbf85c  
A10     : 0x0000001b  A11     : 0x00000000  A12     : 0x00000013  A13     : 0x00060023  
A14     : 0x00060021  A15     : 0x00000001  SAR     : 0x00000018  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x40091995  LEND    : 0x400919b7  LCOUNT  : 0xffffffff  

Backtrace:0xfffffffd:0x3ffd8e400x400d3dbd:0x3ffd8e60 0x40109295:0x3ffd8e80 0x40107661:0x3ffd8ee0 0x40153dd2:0x3ffd8f00 0x40154766:0x3ffd8f20 0x4017b5fd:0x3ffd8f70 0x4017d4e7:0x3ffd8f90 

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
Display not detected
ZED-F9P firmware: HPG 1.30
LittleFS Started
SparkFun RTK Surveyor v1.69-Apr 14 2023
Reset reason: ESP_RST_PANIC
settingsFile not found in LittleFS
settingsFile not found in LittleFS
settingsFile not found in LittleFS
settingsFile not found in LittleFS
settingsFile not found in LittleFS
settingsFile not found in LittleFS
Profile 'Profile2' loaded
Fuel gauge not detected.
GNSS configuration complete
System time set to: Friday, April 14 2023 03:22:19
State: Base - Not Started, 2023-04-14 03:22:20.707
*WIFI_OFF
Bluetooth broadcasting as: Surveyor Base-9BBA
State: Base-Temp - Settle, 2023-04-14 03:22:22.054

nseidle, 2023-04-14:
The RC is a wild west, sorry! Can you decode the backtrace?

C4fishfood, 2023-04-14:
no worries- I'm starting to look into the backtrace to see where the crash is occurring

C4fishfood, 2023-04-16:
I have decoded the back traces. There seems to be two mechanisms that lead to the ESP crashing, both related to bluetooth, and possible the BLE side of it. The I will continue to look into it, but Im outside my skill very so any assistance would be appreciated. 

The 1st type of crash occurs maybe every 30-60 seconds and is possibly related to a semaphore not being released properly?
Decoding stack results
0x40083e15: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 402
0x400958e9: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c line 121
0x4009b7c9: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c line 85
0x400968e1: xQueueSemaphoreTake at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1545
0x4017d429: l2c_fcr_proc_pdu at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/stack/l2cap/l2c_fcr.c line 1230
0x4014ffd0: ip6_output_if_src at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv6/ip6.c line 1264
0x4014f6a5: ip6_select_source_address at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv6/ip6.c line 347
0x401832f2: sdp_server_handle_client_req at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/stack/sdp/sdp_server.c line 388
0x40150033: ip6_output_if at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv6/ip6.c line 1162
0x4017b625: l2cble_init_direct_conn at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/stack/l2cap/l2c_ble.c line 891
0x4017d50f: l2c_fcr_proc_pdu at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/stack/l2cap/l2c_fcr.c line 696

C4fishfood, 2023-04-16:
the second crash mechanism occurs every time I try to pair my iPhone with BLE via SWMap:
0x40114fd5: BLEServer::handleGATTServerEvent(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*) at C:\Users\josep\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\BLE\src\BLEServer.cpp line 209
0x4010fb7d: BLEDevice::gattServerEventHandler(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*) at C:\Users\josep\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\BLE\src\BLEDevice.cpp line 123
0x4015ed6a: btc_gatts_cb_to_app at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/btc/profile/std/gatt/btc_gatts.c line 38
0x4015f6fe: btc_gatts_cb_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/btc/profile/std/gatt/btc_gatts.c line 761
0x40186595: btc_thread_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/btc/core/btc_task.c line 193
0x4018847f: osi_thread_run at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/osi/thread.c line 68

nseidle, 2023-04-24:
Are you able to load our binaries onto your hardware? Do you experience a similar issue? I am trying to remove your compilation/toolchain from the algebra.

C4fishfood, 2023-04-24:
Yes, I can load your binary, and I have made some progress in troubleshooting it, but still haven't gotten it to work just yet. 

At the time of my 1st post, I assumed the ESP could communicate to the GPS via UART serial alone, but its seems that connecting it via I2C as well as fixed the problem for the BLE not automatically re-connecting on start-up (when using the Sparkfun binary). But the GNSS connection window on the SW Maps app showed invalid fix... I used the Bluefruit Connect app on my phone to connect to the ESP via BLE, and it doesnt show any serial traffic. I was suspecting the the binary may have been compiled using different UART pins than are broken out on the ESP Thing plus, which is why I started move to try to compile it in the Ardunio IDE and ran into that toolchain problem/ESP crash.

I have connected my computer to an FT232 breakout to the UART pins on the ESP, and I can send serial data over BLE to my iPhone with the Sparkfun Binary loaded. So that wireless connection is not the issue. I can also see the GPS status when I do a hard wired terminal connection to the ESP, so I know the I2C communication works. I can also see serial traffic coming off the GPS UART with the FT232...

nseidle, 2023-04-25:
Ah! Ok, good work. I assume ... yep, looking at your output your board is identifying itself as a Surveyor because there are no ID resistors. Please have a look at the Surveyor (schematic)[https://cdn.sparkfun.com/assets/c/1/b/d/8/SparkFun_RTK_Surveyor-v13.pdf]. You'll see multiple connections between the ZED and the ESP including I2C (predominantly for config but also for PVT) and UART1 to UART2 (for message output being broadcast over BT). 

I'll help as best I can but supporting a DIY combo build is going to be hard and unpredictable. We tried to write the firmware to fail gracefully (ie, if the Fuel Gauge IC doesn't respond, mark it offline and contine) but since we've only tested against our hardware, all bets are off.

C4fishfood, 2023-04-25:
Im getting closer! It seems like this may have something to do with the baud rate and timing for how the ESP attempts to configure the GPS. The default rate of the GPS is 38400 but it seems that the RTK default rate maybe 0? I have been intermittently been able to get a data stream to work from the GPS UART, to the ESP, and over BLE to the iPhone. But it doesnt always work. Im able to set the port data rate in the ESP when the gps is not connected, and when I then hook it up im able to get a good data stream. But the port data rate is fixed at 0, and will not take a new setting, if I have the GPS hooked up at startup

nseidle, 2023-04-25:
The default rate for UART1 and UART2 on the ZED is indeed 38400. We increase this (using I2C) to 460,800 to handle the higher data when more messages are turned with higher fix rates (10Hz or more). I'm a bit confused because this should all be handled during configUbloxModule() using the internal settings and defaults (ie 460800 for UART1).

You should consider using u-center to view the ZED's settings and message output.

C4fishfood, 2023-04-25:
Ok, I got it working well enough for my purposes. I don't full understand what's going on, but that's because your level of programming is over my head! There are a few other firmware options to handle NTRIP on an ESP, but I really wanted to get the Sparkfun firmware working vs the others because you have done such good work on it. Thank you for that, and also thank you for assisting me here! If it helps at all, my hack to get things working was to only hook the UART1 TX on the gps to the UART2 RX on the ESP (GPIO16), and to have the UART2 TX (GPIO17) on the ESP hooked only to UART2 RX on the GPS. This allows me to received the GPS solution on my phone, and to push the NTRIP correction stream from SW Maps to the GPS. I can now get an RTK fix on iOS. Sweet! Im not really sure why, but having the UART1 on the GPS be bi-directional to the ESP would block the BLE radio from broadcasting or allowing a previously paired device from reconnecting.

nseidle, 2023-04-25:
Yay! Congrats. And thanks for walking through what it took to get working. Other folks will benefit from that.

I'm going to close for now but let me know if anything else pops up.

closed 2023-04-25 by nseidle