sparkfun/SparkFun_RTK_Firmware #434: Failed to enter wifi config mode. Facet restarts

amlago, 2023-04-17:
Facet v 3.3 04/14
Hello.

In the update v3.3 04/14 I have not been able to enter the Wifi configuration.
I put the wifi config option and the sign connecting to the wifi network appears.
After a few seconds it does not connect and the Facet restarts.
I tried placing the wifi config in AP mode using the bluetooth serial configuration and it doesn't work either. It reboots.

amlago, 2023-04-18:
I put the factory values and I was able to enter the wifi config.
But after a few tries the same thing happened to me. It says connecting to wifi and reboots.
I went back to version 3.2 and everything works fine.

nseidle, 2023-04-18:
Thanks for reporting this! 

I've been seeing similar problems. Are you attached to a computer when the device reboots or over WiFi? Can you copy/paste the output of the serial terminal around when the device resets?

I believe the issue is caused by SD write tasks during the WiFi connection attempt. We've added additional watch dog servicing to avoid a watch dog reset (WDT) [here](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/release_candidate/Firmware/RTK_Surveyor/Tasks.ino#L463) in v3.3 4/17. This has cleared up the issue for me. Please give it a try.

tonycanike, 2023-04-23:
FYI I have this problem with 3.3 March ?? version.  I'm going back to v3.2.   Had to abort a job in the middle of the day and go back to the office to downgrade the firmware. Every time I tried to put it in WiFi Config from the front panel, it rebooted and went back to rover mode.

amlago, 2023-04-24:
I already reported this problem with the restart when entering the wifi config in previous versions and the fastest solution was to restore default values.

With version 3.2 I have worked without problems.

The restart problem with the wifi config when it appears I had it immediately when updating.

nseidle, 2023-04-24:
I do not see this problem on any of my units using today's release candidate (April 24). I also do not see this problem in v3.2 firmware so that is a big concerning.

If possible, can you capture the serial output during a reset? It looks something like this:

    Bluetooth broadcasting as: Express Rover-9CEE
    [ 79074][I][BluetoothSerial.cpp:251] esp_spp_cb(): ESP_SPP_INIT_EVT
    [ 79077][I][BluetoothSerial.cpp:258] esp_spp_cb(): ESP_SPP_INIT_EVT: slave: start
    [ 79078][D][NVM.ino:97] recordSystemSettingsToFileSD(): Removing from SD: /SFE_Express_Settings_0.txt
    [ 79089][I][BluetoothSerial.cpp:359] esp_spp_cb(): ESP_SPP_START_EVT
    [ 79155][D][NVM.ino:139] recordSystemSettingsToFileSD(): Settings recorded to SD: /SFE_Express_Settings_0.txt
    [ 79176][D][NVM.ino:169] recordSystemSettingsToFileLFS(): Removing LittleFS: /SFE_Express_Settings_0.txt
    [ 79303][D][NVM.ino:181] recordSystemSettingsToFileLFS(): Settings recorded to LittleFS: /SFE_Express_Settings_0.txt
    *NTRIP_CLIENT_OFF
    State: Rover - No Fix, 2023-04-24 20:44:43.493
    Batt (97%): Voltage: 4.21V Charging: 11.44%/hr Green
    [ 80155][D][Begin.ino:1141] deleteSDSizeCheckTask(): sdSizeCheck Task deleted
    Log file name: /SFE_Express_230424_204443.ubx
    State: Rover - Fix, 2023-04-24 20:44:43.596
    Rover Accuracy (m): 0.3542
    Rover Accuracy (m): 0.3523
    Rover Accuracy (m): 0.3507
    Batt (97%): Voltage: 4.21V Charging: 11.44%/hr Green
    [ 86019][D][States.ino:1152] requestChangeState(): Requested System State: 13
    State: Display Setup, 2023-04-24 20:44:49.646
    State: WiFi Config Not Started, 2023-04-24 20:44:53.168
    [ 90037][D][Bluetooth.ino:227] bluetoothStop(): Bluetooth turned off
    *NTRIP_CLIENT_OFF
    *NTRIP_SERVER_OFF
    [ 90397][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 0 - WIFI_READY
    [ 90399][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 2 - STA_START
    [ 90400][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: sparkfun-guest
    Connecting WiFi... [ 92844][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 1 - SCAN_DONE
    [ 92844][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
    [ 92845][I][WiFiMulti.cpp:119] run(): [WIFI] 17 networks found
    [ 92850][D][WiFiMulti.cpp:151] run():        0: [4][E4:90:7E:F6:2F:C8] Tempy (-49) *
    [ 92858][D][WiFiMulti.cpp:151] run():        1: [1][74:AC:B9:E7:7E:4C] sparkfun-iot (-53) *
    [ 92866][D][WiFiMulti.cpp:151] run():        2: [1][7E:AC:B9:E7:7E:4C] sparkfun-internal (-53) *
    [ 92874][D][WiFiMulti.cpp:151] run():        3: [1][82:AC:B9:E7:7E:4C] test-guest (-53)
    [ 92882][D][WiFiMulti.cpp:149] run():  --->   4: [1][7A:AC:B9:E7:7E:4C] sparkfun-guest (-54) *
    [ 92891][D][WiFiMulti.cpp:151] run():        5: [1][EE:63:DA:25:70:7C] test-guest (-75)
    [ 92899][D][WiFiMulti.cpp:151] run():        6: [1][EA:63:DA:25:70:7C] sparkfun-internal (-76) *
    [ 92907][D][WiFiMulti.cpp:149] run():  --->   7: [1][E6:63:DA:25:70:7C] sparkfun-guest (-76) *
    [ 92915][D][WiFiMulti.cpp:151] run():        8: [1][E0:63:DA:25:70:7C] sparkfun-iot (-77) *
    [ 92923][D][WiFiMulti.cpp:151] run():        9: [6][E0:63:DA:28:0A:84] sparkfun-iot (-78) *
    [ 92932][D][WiFiMulti.cpp:151] run():        10: [6][EA:63:DA:28:0A:84] sparkfun-internal (-78) *
    [ 92940][D][WiFiMulti.cpp:151] run():        11: [6][EE:63:DA:28:0A:84] test-guest (-78)
    [ 92948][D][WiFiMulti.cpp:149] run():  --->   12: [6][E6:63:DA:28:0A:84] sparkfun-guest (-78) *
    [ 92956][D][WiFiMulti.cpp:151] run():        13: [1][60:D0:2C:13:D6:88] NI-Wireless (-87) *
    [ 92965][D][WiFiMulti.cpp:151] run():        14: [1][60:D0:2C:53:D6:88] NI-Guest (-87) *
    [ 92972][D][WiFiMulti.cpp:151] run():        15: [1][60:D0:2C:93:D6:88] NI_IOT (-87) *
    [ 92980][D][WiFiMulti.cpp:151] run():        16: [1][60:D0:2C:93:DE:98] NI_IOT (-90) *
    [ 92988][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 7A:AC:B9:E7:7E:4C SSID: sparkfun-guest Channel: 1 (-54)
    [ 93063][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
    [ 93124][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
    [ 93125][D][WiFiGeneric.cpp:880] _eventCallback(): STA IP: 10.8.252.91, MASK: 255.255.252.0, GW: 10.8.252.1
    [ 93130][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
    [ 93135][D][WiFiMulti.cpp:175] run(): [WIFI] SSID: sparkfun-guest
    [ 93141][D][WiFiMulti.cpp:176] run(): [WIFI] IP: 10.8.252.91
    [ 93147][D][WiFiMulti.cpp:177] run(): [WIFI] MAC: 7A:AC:B9:E7:7E:4C
    [ 93152][D][WiFiMulti.cpp:178] run(): [WIFI] Channel: 1
        
        
        Network Configuration:
        ----------------------
                 SSID: sparkfun-guest
          WiFi Status: 3
        WiFi Strength: -55 dBm
                  MAC: 40:F5:20:57:9C:EC
                   IP: 10.8.252.91
               Subnet: 255.255.252.0
              Gateway: 10.8.252.1
                DNS 1: 10.16.7.12
                DNS 2: 10.16.7.13
                DNS 3: 0.0.0.0
        
    [ 93221][D][menuBase.ino:653] getFileLineSD(): File /StationCoordinates-ECEF_0.csv not found
    [ 93226][E][vfs_api.cpp:102] open(): /littlefs/StationCoordinates-ECEF_0.csv does not exist, no permits for creation
    [ 93229][D][menuBase.ino:580] getFileLineLFS(): File /StationCoordinates-ECEF_0.csv not found
    [ 93258][D][menuBase.ino:653] getFileLineSD(): File /StationCoordinates-Geodetic_0.csv not found
[... cut ...] false,coordinateInputType,0,minCNO,6,wifiNetwork0SSID,sparkfun-guest,wifiNetwork0Password,sparkfun6333,wifiNetwork1SSID,,wifiNetwork1Password,,wifiNetwork2SSID,,wifiNetwork2Password,,wifiNetwork3SSID,,wifiNetwork3Password,,wifiConfigOverAP,0,wifiTcpPort,2947,enableRCFirmware,false,
    [ 93534][D][Form.ino:225] startWebServer(): Web Server Started
    State: WiFi Config, 2023-04-24 20:44:56.935
    Batt (97%): Voltage: 4.21V Charging: 11.44%/hr Green

I will keep trying to replicate the problem.

amlago, 2023-04-24:
With version 3.3 of 04/24 everything seems to work very well.
I've been in and out of wifi config with no problem.
Eureka

closed 2023-05-05 by nseidle