sparkfun/SparkFun_RTK_Firmware #440: Write today's date to log file in $GNTXT message

tonycanike, 2023-04-23:
### Subject of the issue

Sometimes multiple days of log data get written to a file called "SFE" and I need to figure out which day's data are in the SFE file.

The NMEA messages give the time, but not the date.

It's really really hard to cut up a file using Notepad++ or some other text editor if we don't know where each day's data are

It would also be useful to know the logging actually worked, and having the $GNTXT message with today's date would be helpful.

Yes, ideally everything would work and this wouldn't be necessary.  But it does get frustrating being a beta-tester so help us out a little!

Thanks.

Facet running 3.3 Dev from March-something.  I had to downgrade to 3.2 release due to wifi config issues so don't remember for sure.

nseidle, 2023-04-24:
> Sometimes multiple days of log data get written to a file called "SFE"

There is a log on your SD card called "SFE.ubx"? That's a rather large issue that we would need to fix. Or are you seeing a mixing of data between log files, aka issue #401?

> Yes, ideally everything would work and this wouldn't be necessary. But it does get frustrating being a beta-tester so help us out a little!

Ah. That's very fair. I'll get after it.

nseidle, 2023-04-24:
[image]

Date format is: YYMMDD,HHMMSS

tonycanike, 2023-04-25:
Thank you!  Awesome!

nseidle, 2023-05-05:
This was implemented in [v3.3 of the RTK firmware](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v.3.3). Please let us know if you see any problems.

closed 2023-05-05 by nseidle