sparkfun/SparkFun_RTK_Firmware #441: Reset to Factory Defaults - Warn SD card will be formatted.

tonycanike, 2023-04-23:
3.3-dev and 3.2

When resetting to factory defaults there is a "are you sure" confirmation, but there's no warning (reminder) that the SD card will be reformatted.   Sad :( when I forget that part.

nseidle, 2023-04-24:
Sorry for the confusion. A factory reset does the following:

* Erase/format the *internal* file system (this is LittleFS, and is used if no SD is detected)
* Sets the ZED-F9x to factory defaults (baud rate, I2C address, update rate, etc)
* If an SD card is detected, remove the setting file, and ECEF/Geo coordsinate files associated with the current profile.

It does not format or touch previously logged data on the SD card. 

I'll update the serial print message to "Formatting internal file system..." to clarify.

tonycanike, 2023-04-25:
Thanks, appreciate the quick response!

commit b529173c (2023-04-24): Clarify formatting of internal FS when doing factory reset.

Fix #441
closed 2023-05-05 by nseidle