sparkfun/SparkFun_RTK_Firmware #442: CR-LF needed before "$GNTXT,01,01,01,Reference station position seems incorrect" messages

tonycanike, 2023-04-24:
### Subject of the issue
It seems that most of the $GNTXT NMEA-like messages inserted by the RTK Firmware have a CR-LF before them.  So the $ is always in the first column of a new record.

However, the  "$GNTXT,01,01,01,Reference station position seems incorrect"  messages do not and so do not.  They generally appear in random columns after a record of binary data.

Should there be a CR-LF before the  "$GNTXT,01,01,01,Reference station position seems incorrect"  messages?

nseidle, 2023-04-24:
> It seems that most of the $GNTXT NMEA-like messages inserted by the RTK Firmware have a CR-LF before them.

This is incorrect. We do not print anything before these statements. However, at the end of every NMEA sentence is a followed by an \r\n. So when you see nicely aligned stuff, it's a bunch of NMEA. When it goes awry, it's likely because a NMEA sentence follows a UBX (binary) message.

[image]

> However, the "$GNTXT,01,01,01,Reference station position seems incorrect" messages do not and so do not.

I'm not familiar with this message. Are you talking about the custom NMEA messages we insert into the logs in general? They will follow the same rules and formatting. Arg, it doesn't look like we've documented the custom sentences anywhere succintly but the general format is this:

    snprintf(nmeaTxt, sizeof(nmeaTxt), "$GNTXT,%02d,%02d,%02d,%s*", totalNumberOfSentences, sentenceNumber, textID, textMessage);

Which looks like [this](https://docs.sparkfun.com/SparkFun_RTK_Firmware/system_status_report/) in practice.

Do you really need all NMEA messages to start on their own line? What parser are you using that needs this?

tonycanike, 2023-04-25:
That all makes sense, thanks. 

The parser is my eyes looking at the file "SFE" (no .extension) while searching in a text editor trying to figure out what's what to tease it apart.   It's just easier to look at if all the NMEA-like $statements start on a new record, but not necessary. 

I presume the warnings about the reference station position occurs because I turned on the base and it started up fixed with yesterday's position before I configured it for today.  I was trying to find where the warnings stopped and delete that part of the file.  There were a couple static sessions in the one SFE file so it took a little doing. 

The "start new log file" wasn't working (never showed the new file name on the wifi page) and then the wifi configuration stopped working.   The newly named files were on the SD card but zero len.  The data ended up in the file "SFE".

nseidle, 2023-04-25:
> Facet running v3.3-Mar 11 2023

Please give April 24 RC a try. We've fixed quite a bit over the last two weeks. If you see a file named **SFE** (with no extension) again, please let me know.

tonycanike, 2023-04-25:
The inability to use WiFi configuration was a real show-stopper for me and I've backed down to v3.2 as @amlego mentioned he did.

tonycanike, 2023-04-26:
Will do.  Thanks for all the help!

closed 2023-04-26 by tonycanike