sparkfun/SparkFun_RTK_Firmware #445: Config fails with Bike dynamic model

Stefal, 2023-04-25:
### Subject of the issue
If I create a user profile with "Bike" dynamic model, the Rtk Express can't load this preset. The lcd displays "Rover failed" then reboot.

### Your workbench
* What version of RTK firmware are you running? 3.2
* F9P Firmware : 1.32
### Steps to reproduce
Create a new user profile, and set it to 3Bike dynamic model"

### Actual behavior
The RTK Express reboot in loop
Messages on the serial link:
```
Rover config failed
Batt (36%): Voltage: 3.96V Charging: 44.93%/hr Yellow
Rover config failed
Rover config failed
Rover config failed
Rover config failed
Rover config failed
```

nseidle, 2023-04-25:
The Bike Dynamic model is not available on the F9P platform.

[image]

I'll get the Bike model removed from the menu.

nseidle, 2023-05-05:
This should be fixed in [v3.3 of the RTK firmware](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v.3.3). Please let us know if you still see problems.

closed 2023-05-05 by nseidle