sparkfun/SparkFun_RTK_Firmware #447: ROVER: Decode RCTM 1005,1006,1007 and record in $GNTXT messages to rover log file

tonycanike, 2023-04-28:
### Subject of the issue
The rover decodes the  RTCM 1005 to plain text and record in $GNTXT messages in the rover log file written.

Split out from #405 

I'm not actually sure we need this, or even want it.  I leave it open to discussion.

PaulZC, 2023-05-07:
Hi Tony (@tonycanike ),

I've upgraded the u-blox library (v3.0.12). It is now able to parse the correction data being 'pushed' to the u-blox module. It is able to parse 1005 and 1006, allowing the ARP ECEF and height to be extracted. Now we have this, it should be a piece of cake to copy the ARP into a new Rover GNTXT message.

**Question:** how essential is 1007? If I'm reading it right, it contains the name (description) of the antenna - but not its location?

Thanks - and best wishes,
Paul

tonycanike, 2023-05-08:
Paul, awesome news!  1007 isn't critical to me at all. Thanks for all the hard work - these things really help!

Tony.

PaulZC, 2023-05-22:
Just a quick update that this feature is working well in the "Reference Station Ethernet" version of the release candidate. It won't be long until it is merged into release-candidate proper.

The u-blox library integrity checker can now also print the content of any GNTXT messages it finds:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/blob/main/Utils/UBX_Integrity_Checker.py

tonycanike, 2023-05-22:
Great news, thanks!

PaulZC, 2023-05-30:
This feature in the latest release_candidate binary.

Open the Logging Menu, and select option 5 ("5) Log Antenna Reference Position from RTCM 1005/1006: ") to enable. Option 6 sets the logging interval.

The ARP is logged in a custom GNTXT,01,01,10 message as ECEF-X, ECEF-Y, ECEF-Z, Antenna Height. The Antenna Height will be zero if the data was extracted from 1005.

The latest version of the integrity checker will print the GNTXT messages if you ask it to. See https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/476#issuecomment-1567346348.

Enjoy!
Paul

nseidle, 2023-06-06:
This has been implemented in v3.4. Docs are [here](https://docs.sparkfun.com/SparkFun_RTK_Firmware/configure_data_logging/#serial-interface).

closed 2023-06-06 by nseidle