sparkfun/SparkFun_RTK_Firmware #449: The NEO-D9C module is not supported in v3.2-Mar 8 version 2023.

pepefrog1234, 2023-04-29:
### Subject of the issue
I tried to use the UART 2 port for the NEO-D9C to input differential data, but the firmware changes the UART 2 input data type to RTCM 3 at startup, leading to a failure to use differential positioning (CLAS).

### Your workbench
[image]
As shown above, the satellite signal first passes through NEO-D9C to acquire the L6 signal from QZSS and then transmits the signal to NEO-F9P through the feeder.

### Steps to reproduce
The firmware will overwrite the UART 2 settings when it is activated.

### Expected behavior
I would like to add a setting option to adjust the input format of UART 2.
Because NEO-D9C only supports output in UBX format.

### Actual behavior
The firmware will overwrite the UART 2 settings when it is activated.

nseidle, 2023-05-01:
We don't yet have a product that incorporates the NED-D9C. If you'd like to contribute a PR to support it, we'd be willing to consider it.

PaulZC, 2023-05-30:
I've added the protocol support in this commit: https://github.com/sparkfun/SparkFun_RTK_Firmware/commit/3bfbd0cb8d9d54d678a210d712eccc44bb57544b

TLDR: This _would_ also allow correction data from an external NEO-D9S to be input over UART2 too. But you would also need to set the encryption keys manually, or automatically, so the ZED could decode the data. And, right now, the key entry and update requires ```online.lband``` to be true.  We would need to significantly restructure the code to support manual key entry without having the NEO-D9S present on the I2C bus. I'm proposing we don't consider that as an option / future upgrade, and close this issue once the protocol support has been merged.

PaulZC, 2025-09-09:
Closing this. Protocol support was added some time ago...