sparkfun/SparkFun_RTK_Firmware #450: failure to boot with 3.2

gdt, 2023-05-01:
I have an RTK Express that used to be running 2.4.  I updated to 3.2 from the release artifact.

I first tried OTA, before I realized 2.4 was too old.

I then used esptool (on a mac) with a script translated from windows:

```
#!/bin/sh

AUX=$HOME/SOFTWARE/GEO/SparkFun_RTK_Firmware_Binaries/bin

esptool.py --chip esp32 \
  --port /$HOME/tty.sparkfun --baud 115200 \
  --before default_reset --after hard_reset \
  write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect \
  0x1000 ${AUX}/RTK_Surveyor.ino.bootloader.bin \
  0x8000 ${AUX}/RTK_Surveyor_Partitions_16MB.bin \
  0xe000 ${AUX}/boot_app0.bin \
  0x10000 $1
```
and that succeeded but on rest there was no display.

Running kermit and watching at 115200, I see
```
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8
ets Jul 29 2019 12:21:46
```
so it seemst this isn't booting ok.

I woudl be happy to run an official esptool script, but there appears to be only windows, and I think I translated it ok.  I used it (mostly as is, with path changes) before and that was ok.

gdt, 2023-05-01:
v3.1 behaved the same way.  I did a full erase_flash and re-flashed with the same results.

nseidle, 2023-05-01:
This likely has to do with boot partitions. As part of our [bootloader tool](https://github.com/sparkfun/SparkFun_RTK_Firmware_Uploader), we check the flash size of the ESP32 and load the correct partition file to match. Can you run the tool via python? What is the flash size of your ESP32?

nseidle, 2023-05-01:
I agree, it's should be and is likely 16MB (I don't think we sold any Expresses with 4MB but it's best to let the tool verify that). 

The output from our tool looks great. Is it still resetting? You may want to do a esptool erase_flash command.

nseidle, 2023-05-01:
> I am assuming the boot/partition files have been stable for a long time.

Yes.

I successfully loaded my RTK Express with v3.2 using the esptool.py and the following command. All files came from the [resource sub directory](https://github.com/sparkfun/SparkFun_RTK_Firmware_Uploader/tree/main/RTK_Firmware_Uploader/resource) and were locally located in the esptool.py directory):

esptool.py --chip esp32 --port COM28 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 RTK_Surveyor.ino.bootloader.bin 0x8000 RTK_Surveyor_Partitions_16MB.bin 0xe000 boot_app0.bin 0x10000 RTK_Surveyor_Firmware_v3_2.bin

[image]

gdt, 2023-05-02:
I ran that (with port changed, and using esptool.py that I have installed), and it behaved the same way as I reported previously.   Looking, it's different.   After "Auto-detected..." I see
```
Warning: Image file at 0x1000 is protected with a hash checksum, so not changing the flash size setting. Use the --flash_size=keep option instead of --flash_size=16MB in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
```
whereas your output shows (typing from image, don't worry if I got it not quite right):
```
Flash params set to 0x024f
```

So esptool seems to think the bootloader bin has a hash checksum and something should be different.   I ran again with  ../esptool.py and got the `Flash params set to 0x024f` and after writing/reset I was able to boot.   So I'll close this and open another issue about the flash process failing with more recent esptool.  Thanks for the help!

closed 2023-05-02 by gdt

gdt, 2023-05-02:
And not part of this bug, with v3.2, I got a float fix (with MassDOT RTK network) indoors, and QField connects over bluetooth and I see position and a sensible error estimate in QField.   So I think there is only one thing wrong, and all else is good - which is great, and thank you for all the coding/shepherding.

nseidle, 2023-05-02:
RTK Float indoors is always good to see!