sparkfun/SparkFun_RTK_Firmware #453: RTK Express providing hostname when getting address from hotspot

dwightbrew, 2023-05-04:
When I am in the field, the RTK Express connects to my Android phone’s WiFi hotspot. My phone is also where I run GField and SW Maps, I configure my RTK Express to obtain RTK data over the internet using my phone’s hotspot. However, if I want to connect from my phone to my RTK Express using TCPIP, I have an issue, because my phone’s hotspot assigns a random IP address to my RTK Express, and the RTK Express (apparently) does not provide a hostname to my phone’s hotspot when obtaining an IP address. I know it is possible to provide a hostname when obtaining an IP address from my phone’s hotspot, because when I connect other Windows computers to my phone’s hotspot, I can ping these other computers by name.

It would be useful if a hostname could be provided when the RTK Express is obtaining a DHCP IP address so I could configure this hostname in GField and SW Maps.

nseidle, 2023-05-05:
@dwightbrew - You need to know the IP address of the RTK device, correct? If we periodically displayed the device's IP address on the OLED, would that solve your issue?

gdt, 2023-05-06:
I'm not @dwightbrew  but there are two aspects.  Yes, one is knowing the address.   The other is having a setup that causes the phone program (or some other program on some other computer) to connect without having to fiddle with it.   Sending mdns might enable that -- someone needs to resaarch that.

(For me, being able to use Bluetooth SPP with QField is great as I don't need to fiddle with the addresses.  Vespucci with ardusimple used to be great when Android's subnet was stable, and now it's a minor hassle.)

dwightbrew, 2023-05-07:
@nseidle - I don't think periodically displaying the IP address is something I would use. Although I haven't tried this, I think I could get the address by put it in config mode to get the address and then back to normal mode. As @gdt pointed out, this request is more about not having to fiddle with the settings each time.

Not being able to easily get to my RTK Express is not a showstopper for me, it just limits some of the things I would like to try. For example, I use a few different NTRIP casters. Currently I have each NTRIP caster as a separate profile. I have thought about using the NTRIP (Lefebure) app to switch between different casters. Although Lefebure can supply NTRIP data using TCP/IP, I don't know how to do this even if I know the RTK Express address. I would use Bluetooth on Lefebure to provide the NTRIP data and then TCP/IP for QField.

Rather than periodically displaying the IP address, I would prefer to wait for a solution to access the RTK Express by hostname, even it this is a lower priority that may take some time coming.

nseidle, 2023-05-13:
We can enable mDNS at the expense of some RAM (we're already running out, but that's a different issue).

[image]

Is this what you're looking for? We can make the device connect via 'rtk.local'.

gdt, 2023-05-13:
I suspect that is what @dwightbrew is looking for.  I totally get the RAM issue, though, but if it only uses RAM when enabled and people only enable what they need, that might all be best.

nseidle, 2023-05-13:
Sounds good. MDNS can be enabled from the WiFi menu.

[image]

dwightbrew, 2023-05-14:
This looks very promising. I downloaded the RTK_Surveyor_Firmware_RC-May_13_2023 release, but I am having trouble getting the WiFi working. The config menu says that WiFi is not compiled (image attached).

[image]

nseidle, 2023-05-17:
Whoops! Sorry about that. I committed with the modules turned off. RC is now updated correctly.

nseidle, 2023-06-06:
This has been implemented in firmware release v3.4.

closed 2023-06-06 by nseidle