sparkfun/SparkFun_RTK_Firmware #458: Bluetooth serial config: Possible error when entering base coordinates

amlago, 2023-05-10:
Bluetooth serial config: Possible error when entering base coordinates

Yesterday when entering coordinates of the base station using the bluetooth serial configuration, the entry of the value failed.
I was working in DDD mm ss format with no decimal values in the seconds.

If you added an ss.0 extension to the seconds if the value was accepted.

In other words, you had to enter a decimal extension to the seconds so that the value was accepted even if this value was zero.
The value with whole seconds should be accepted if they are the values I have and not have to add the decimal extension in a mandatory way.

Any coordinate value with decimals of a second is accepted correctly.
I hope it has been understood. As always apologies for my translator.
Greetings

nseidle, 2023-05-13:
Understood! Thanks for reporting. I didn't realize 'DD MM SS' would be used. 3 new types added.

    Given '-1051105' should be -105.1847222: -105.184722222 Type: DDMMSS
    Given '-105 11 05' should be -105.1847222: -105.184722222 Type: DD MM SS
    Given '-105-11-05' should be -105.1847222: -105.184722222 Type: DD-MM-SS

I am sort of surprised this format would be used at all; the precision is pretty low. Are you just using it for quick setups?

I'll get this into the RC shortly.

amlago, 2023-05-13:
Hi Nathan. 
 I was testing with integer values, no decimals for convenience, and I thought it was weird that it wouldn't let me enter the values.  Of course they were laboratory tests.  It is very rare that working in the field does not require high precision.  I myself asked you a long time ago to add the coordinates with 8 figures in the wifi config.  So clarified.  
Thank you.

commit 9158de29 (2023-05-13): Add decimal-less coordinate types and input

Fix #458