sparkfun/SparkFun_RTK_Firmware #461: recieve L-band corrections and broadcast via IP + receive via IP and apply PointPerfect corrections on 2nd unit

edgecase14, 2023-05-12:
### Subject of the issue
This would be a new feature to support the following use case:  Frequently arborists wish to inventory trees with high accuracty location
This presents a problem when using PointPerfect corrections, when neither L-band or cellular data for IP based corrections is available.
One solution would be to place an L-band antenna where reception is possible, and then relay the SPARTN messages via WiFI or 900MHz telemetry radio to a ZED-F9P based receiver operating under heavy tree canopy.

### Your workbench
* What version of RTK firmware are you running? v3.2
* What radios are you using: Bluetooth, WiFi and/or 900MHz telemetry radio
*  What app are you using to connect over Bluetooth? QFIELD
* Are you transmitting NTRIP back to the device? This would be similar to NTRIP
* Are there any additional details that may help us help you?
Both TOP106 type l-band antenna in an area with a clear view of sky, or a parabolic dish or helix antenna under tree canopy are being considered.  The L-band receiver would be mounted on a tripod and aimed at the Inmarsat, while backpack worn ESP32/ZED-F9P based "rovers" (typically 1-2 units) are used .

### Steps to reproduce
Field testing has shown that L-band signal will drop out much sooner than WAAS or Autonomous GNSS service, under tree canopy, especially when canopy is south of the rover.

### Expected behavior
Ideally L-band corrections should continue to function as long as GNSS signals are useable.

### Actual behavior
Rover switches to WAAS (DGPS) and continues to function well, long after L-band has dropped out.

nseidle, 2023-05-12:
> Ideally L-band corrections should continue to function as long as GNSS signals are useable.

Thanks for reporting and proposing a solution. I get the impression from your title that you want PointPerfect corrections over WiFi, but then the content of the issue says you want PP over radio. Which is it?

I think allowing an L-Band Facet (I know you're on custom hardware) to access its PP corrections using WiFi is a fine idea. The use case would be a user with cellular coverage, but no or poor L-Band reception, could continue RTK if GNSS was good enough.

> One solution would be to place an L-band antenna where reception is possible, and then relay the SPARTN messages via WiFI or 900MHz telemetry radio to a ZED-F9P based receiver operating under heavy tree canopy.

Please note this is already possible using a base is [assisted mode](https://docs.sparkfun.com/SparkFun_RTK_Firmware/configure_base/#assisted-base). Basically, an L-Band unit surveys in but with really accurate position. That base then starts transmitting RTCM. If you hook a radio to the RADIO port, it will transmit to a rover Facet and you'll have, effectively, a L-Band based relay.

nseidle, 2023-06-19:
Closing due to inactivity. Please see issue #497 for a discussion of PointPerfect over WiFi (or other) on non L-Band devices.