sparkfun/SparkFun_RTK_Firmware #462: Potential logic error on the battery level determination

gigapod, 2023-05-12:
Found a potential logic error  here:

https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/50b4c3fed9f143b37ee5634b167c095dd01924c7/Firmware/RTK_Surveyor/System.ino#L333

Code execution never reaches the final  **else** statement - actual code:

```c++
if (battLevel < 10)
      snprintf(tempStr, sizeof(tempStr), "Red");
    else if (battLevel < 50)
      snprintf(tempStr, sizeof(tempStr), "Yellow");
    else if (battLevel >= 50)
      snprintf(tempStr, sizeof(tempStr), "Green");
    else
      snprintf(tempStr, sizeof(tempStr), "No batt");
```
I think it should be something like this:

```c++
if (battLevel < 10)
      snprintf(tempStr, sizeof(tempStr), "Red");
    else if (battLevel < 50)
      snprintf(tempStr, sizeof(tempStr), "Yellow");
    else if (battLevel < 105)
      snprintf(tempStr, sizeof(tempStr), "Green");
    else
      snprintf(tempStr, sizeof(tempStr), "No batt");
```

I found on the DL board that if the value returned from getSOC() is > ~105, then a battery isn't connected, and the system is being powered by USB. But this could be subjective.

-K

nseidle, 2023-05-12:
Agreed. I can't remember my original plan but it was likely a belt-and-suspender catchall.

I too have seen the fuel gauge return >100% if no battery is attached. This is a fine solution. I may throw an upper limit of 110% with an error above that (belt and suspenders again).

gigapod, 2023-05-12:
Yeah - I'm using 110. in DL for this situation.  I've seen the gauge return up to 103 in my testing with battery connected ... then backs down to 100 as it's algorithm dampens out is rate curve/gathers more data ...

commit 9e6abd9c (2023-05-13): Fix battery level logic

Fix #462