sparkfun/SparkFun_RTK_Firmware #468: Feature Request

j-w-bullfrog, 2023-05-18:
### U-Blox debug 

U-blox support wants a file for support issues that writes the config file and their 'debug' data set in the .ubx files. Would it be possible to add a debug option to the sd card recording? This would eliminate having to have my laptop on the test cart, or it being with bluetooth range. 

### Your workbench
ZED-9fr in sparkfun express + with a sparkfun express working as a rtk casting station.

nseidle, 2023-05-23:
The debug feature within u-center enables a bunch of messages. We could replicate these messages. Step 1 is ID'ing the messages that are enabled. @j-w-bullfrog - Can you ask your u-blox contact if they know what messages need to be enabled? Or is it documented somewhere? Otherwise, we have to dig into the commands that are issued from u-center which can be a bit time consuming.

j-w-bullfrog, 2023-05-23:
Yeah, * can find out the setup.*

https://www.linkedin.com/in/bentreichel

¸ ...      ___________
·´º o`·, /__/ _/\_ ____/\
```)¨(´´´ | | | | |  | | || |l±±±±|
 °´ ¸,.-·~·~·-.,¸ `°·-. :º° ·~·~·-..,¸
As for me and my house, we will serve the Lord! (Joshua 24:15)

On Tue, May 23, 2023 at 11:17 AM Nathan Seidle ***@***.***>
wrote:

> The debug feature within u-center enables a bunch of messages. We could
> replicate these messages. Step 1 is ID'ing the messages that are enabled.
> @j-w-bullfrog <https://github.com/j-w-bullfrog> - Can you ask your u-blox
> contact if they know what messages need to be enabled? Or is it documented
> somewhere? Otherwise, we have to dig into the commands that are issued from
> u-center which can be a bit time consuming.
>
> —
> Reply to this email directly, view it on GitHub
> <https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/468#issuecomment-1559759199>,
> or unsubscribe
> <https://github.com/notifications/unsubscribe-auth/AH2UE23MTPUGJGRILOKB72LXHTPITANCNFSM6AAAAAAYG36TRI>
> .
> You are receiving this because you were mentioned.Message ID:
> ***@***.***>
>

j-w-bullfrog, 2023-05-23:
Here is 1 second of the packet stream from u-center. If I get a list from support I'll post that. Also they write the config file into the log file at the head. I have jpg's of the mag view after that.

17:14:47  R ->  100 UBX-NAV-PVT          iTOW=234905000, fix=3, fixok=1, validTime=1, validDate=1, llh=40.813389/-89.624399/195.7
17:14:47  R ->   36 UBX-NAV-HPPOSECEF    
17:14:47  R ->   44 UBX-NAV-HPPOSLLH     
17:14:47  R ->   28 UBX-NAV-VELECEF      
17:14:47  R ->   44 UBX-NAV-VELNED       
17:14:47  R ->  790 UBX-NAV-ORB          
17:14:47  R ->  484 UBX-NAV-SAT          
17:14:47  R -> 1120 UBX-NAV-SIG          
17:14:47  R ->   28 UBX-NAV-CLOCK        
17:14:47  R ->   72 UBX-NAV-RELPOSNED    
17:14:47  R ->   52 UBX-ESF-STATUS       
17:14:47  R ->   72 UBX-ESF              
17:14:47  R -> 1184 UBX-0C-31            
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->  172 UBX-0C-62            
17:14:47  R ->  220 UBX-0C-34            
17:14:47  R -> 1248 UBX-0C-43            
17:14:47  R -> 2252 UBX-SEC              
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   76 UBX-SEC              
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R -> 2172 UBX-08-10            
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R -> 2172 UBX-08-10            
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   76 UBX-SEC              
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->  332 UBX-MON              
17:14:47  R ->  114 UBX-0C-4B            
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
[... 76 lines cut ...]
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->  148 UBX-03-09            
17:14:47  R ->  156 UBX-08-10            
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R -> 4076 UBX-SEC              
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
17:14:47  R ->   32 UBX-ESF-MEAS         
17:14:47  R ->   36 UBX-ESF-MEAS         
[image]

And screenshots of the msg screen, 
[image]
[image]

j-w-bullfrog, 2023-05-23:
Also, here is a deub file. The comm rate needs to be 921600. Guess, not. Unsupported type. Send me an email add if you want it.