sparkfun/SparkFun_RTK_Firmware #470: Facet v3.47 16/05/23Failed to upload files to SD using WIFI config

amlago, 2023-05-21:
Facet v3.4 05/16/23

Failed to upload files to SD using WIFI config.
When uploading files to the SD with the UPLOAD button, the file name appears on the screen, but when I exit the changes and see the file list again, the new file does not appear.
It was tested from a cell phone and from a notebook with the same result.
In previous versions it loaded coordinate files without any problem. I would like to review the task of loading files.
Clarification: downloads work great

PaulZC, 2023-05-22:
Just a note that this issue might already be fixed...?

Using the not-yet-published release candidate:

Using Ethernet on the Reference Station - the last two files were both uploads, one small, one large:

[image]

Likewise, using WiFi on Facet:

[image]

amlago, 2023-05-22:
Hello. Did you go back in to confirm that the files are on the SD?
When I try to upload the files they appear on the screen, but when I enter again they are gone.

PaulZC, 2023-05-22:
Hi,

Yes, they're definitely there:

[image]

[image]

But I am using the "Reference Station Ethernet" branch. I think Sparky will be releasing another pre-release in a few days. We will check again to make sure this issue has truly gone.

Thanks for reporting,
Paul

amlago, 2023-05-22:
Hello.
  Did you go back in to confirm that the files are on the SD?
When I tried to upload the files they appear on the screen, but when I entered again they were gone.
Update
Today, I just uploaded txt and ubx files with no problem.
A csv file of coordinates gave me trouble uploading. It could be that the file was corrupted.
I tried again with the csv file and it worked. I don't know what's wrong.
Thank you

amlago, 2023-05-22:
I did another test to upload a file and it failed again. I tried charging from my cell phone.
Seems like a random glitch. I will keep testing and reporting.