sparkfun/SparkFun_RTK_Firmware #472: Feature Request - NMEA to cloud aggregator

MarkAnthonyPayne, 2023-05-22:
Similar to the way the NTRIP client can pull RTCM and send NMEA position. Can there be an option to output an NMEA position to remote servers to allow a RTK device be used as a vessel/asset tracker?

nseidle, 2023-05-25:
The device can be set up at a TCP Server and Client via WiFi reporting its NMEA data (location). It's a hop and a jump to running a service on a server that harvested the incoming packets and stashes them anywhere you need. Is that sufficient for your purposes?

MarkAnthonyPayne, 2023-06-05:
I just doubled checked this. Installed Vespucci as a test and configured with correct IP/Port and ensured that Facet was configured as TCP Client (pulled SD card and manually checked the config file). Also, connected to CH340 over serial to ensure correct IP reported for Facet. Vespucci reports connection refused, and nothing seen on Wireshark (unless I ping the Facet).

nseidle, 2023-06-12:
This has been cross posted here: https://forum.sparkfun.com/viewtopic.php?f=116&t=59682

I just double checked, [MDNS](https://docs.sparkfun.com/SparkFun_RTK_Firmware/configure_with_wifi/), and TCP client server all work with Vespucci if you use a cellular hotspot.

I think you've got what you need, let us know if you don't.

MarkAnthonyPayne, 2023-06-13:
Was in my list to check.... Still even using a WiFi hotspot I cannot connect using 'NMEA from TCP client' when Facet is configured as TCP Client. When Facet is configured as TCP Server I seem to connect (well I get no errors) using 'NMEA from TCP server'. I'm still not convinced TCP client works given I could not see any network activity using wireshark.