sparkfun/SparkFun_RTK_Firmware #473: facet v3.3 Failed to save base coordinates (latitude) to file

amlago, 2023-05-22:
facet v3.3
Failed to save base coordinates to file
Today working in the field I tried to save coordinates of the base station with wifi config.
Pressing the Add button saved the station, but with failures.
The latitude is not saved and the word undefined is put in its place.
With version 3.2 the base coordinates are correctly saved in the file

amlago, 2023-05-25:
Facet v3.4 from 05/25
With the latest update, the latitude is not saved in the base coordinates file either.
It is replaced by the word undefined.

nseidle, 2023-05-26:
Yep! Sorry, should be fixed now.

[image]

While I was in there, I added support for coordinates with no decimals. The DD MM SS coordinates display a bit oddly in the commonly used coordinate box, but they are functional.

nseidle, 2023-06-06:
This has been implemented in firmware release v3.4.

closed 2023-06-06 by nseidle