sparkfun/SparkFun_RTK_Firmware #476: Enable UBX-SEC-UNIQID for logging

PaulZC, 2023-05-26:
Based on this forum discussion:

https://forum.sparkfun.com/viewtopic.php?p=241438#p241438

PaulZC, 2023-05-26:
Can't see a configuration key for it though. May have to poll it directly? And/or add it to the u-blox library?

nseidle, 2023-05-26:
Ya, it's not really a message or a configurable value, so I suspect it will have to be a helper function in the lib similar to the MON-VER firmware report.

PaulZC, 2023-05-27:
OK. This is now in the u-blox library:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/releases/tag/v3.0.14

PaulZC, 2023-05-27:
Shall I do a GNTXT message for it?

nseidle, 2023-05-27:
Yes please. Also, consider display it on the system menu.

PaulZC, 2023-05-29:
OK. This is done. The bulk of the changes are in here: https://github.com/sparkfun/SparkFun_RTK_Firmware/commit/52fca372b9ce1628dc690164e3fc74487ed9e325

The unique chip ID will pop up on the system menu, the WiFi configuration page, and also gets added to the log file as GNTXT: 01,01,11 is the GNSS unique chip ID; 01,01,07 is the last four digits of the ESP32 MAC Address. Those two together will uniquely identify the RTK unit.

```
UBX Integrity Checker

Could this file contain any NMEA messages? (Y/n):
Could this file contain any RTCM messages? (Y/n):
Do you want to repair the file? (y/N):
Do you want to print any GNTXT messages found? (y/N): y

Processing e:SFE_Reference_Station_230529_114041.ubx

NMEA $ found at byte 0

$GNTXT,01,01,00,ESP_RST_POWERON*1C

$GNTXT,01,01,03,v99.99-May 29 2023*7A

$GNTXT,01,01,04,HPG 1.32*34

$GNTXT,01,01,11,E74032F95D*28

$GNTXT,01,01,07,6CB2*53

$GNTXT,01,01,09,230529,114041*7A

$GNTXT,01,01,08,0,0,0,*45

Processed 74714 bytes
File size was 74714
Longest valid UBX message was 0 bytes
Longest valid NMEA message was 78 characters
Longest valid RTCM message was 0 bytes
Message types and totals were:
Message type: GNTXT      Total: 7
Message type: GNGST      Total: 158
Message type: GNRMC      Total: 158
Message type: GNGGA      Total: 158
Message type: GNGSA      Total: 790
Message type: GPGSV      Total: 78
Message type: GLGSV      Total: 41
Message type: GAGSV      Total: 39
Message type: GBGSV      Total: 39
Message type: GQGSV      Total: 39

Bye!
```

nseidle, 2023-06-06:
This has been implemented in firmware release v3.4.

closed 2023-06-06 by nseidle