sparkfun/SparkFun_RTK_Firmware #481: SiK Telemetry Radio V3 does not seem to work in Dev 3.3 0r Dev 3.4

j-w-bullfrog, 2023-06-02:
### SiK Telemetry Radio V3 does not seem to work in Dev 3.3 0r Dev 3.4
The last time I used the paired telemetry radios I was on the official release firmware. I then started using rtk2go for rtk corrections with the 3.2 to 3.4 dev's. No problems. I wanted to test outside my wifi range so I wanted to switch back to the radios.  They both have green light solid, and red light flashing like its handling data, but the rover acts like it has not rtk available to it.

### Your workbench
RTK express surveyed into fixed coordinates (aka 24 hr raw and send file to Canada). RTK express plus working as the rover.

### Steps to reproduce
switch from ntrip to telemetry radio. They worked great before I went with rtk2go.
.
### Expected behavior
express plus acknowledging its getting and using rtk data. 

### Actual behavior
acts like no correction data available.

j-w-bullfrog, 2023-06-07:
Note to self, don't work on complicated stuff when om pain meds awaiting a root canal. You do dumb shit stuff like plugging the radios into the data port instead of the radio port on the other side.

closed 2023-06-07 by j-w-bullfrog

nseidle, 2023-06-07:
I was just about to setup a test. Sorry to hear about the root canal but I'm glad you got it sorted!

j-w-bullfrog, 2023-06-07:
BTW, is there a way to send data back from the rover to the base over the paired radios?

nseidle, 2023-06-07:
What data do you mean? The radios will take any serial. Are you talking about transmitting logs or coordinates back to the base? It may be possible to send small amounts of positional data back to the base but that's a pretty unique use case.

j-w-bullfrog, 2023-06-07:
Well, my biz partner has become enamored of the idea of remote diagnostics of the rover/mower so I'm figuring we need to put a cellular chip in the system to accomplish that.