sparkfun/SparkFun_RTK_Firmware #482: Altitude decreases over time when connected to NTRIP

JonathanLeeDeKock, 2023-06-03:
### Subject of the issue
Altitude decreases over time when connected to NTRIP

### Your workbench
RTK Surveyor, Firmware 3.2, U-Blox Firmware HPG 1.32, Connected to MACORS NTRIP IMAX via WiFi.

### Steps to reproduce
On an iPhone in SW MAPS or via u-center connected by USB to the u-blox or via the serial connection to the esp32 with the debug message for location active you can see the altitude decrease by about 1 meter per second throughout the time connected.

### Expected behavior
Altitude should remain relatively consistent.

### Actual behavior
The Altitude reported goes down by about 1 meter per second while connected to the NTRIP client.  I'm not sure if this is something in the RTK firmware for how it is configuring the u-blox, or if this is a problem directly in the u-blox computations.

Without the connection to the NTRIP system the altitude is a fairly consistent 60 meters above sea level.  Once connected the altitude begins dropping and drops at a rate of about 1 meter per second.

Screenshots show a few seconds apart of output in u-center.  I am fairly certain that I am not nearly a kilometer below the sea level and going down further.  :-)

[image]
[image]

JonathanLeeDeKock, 2023-06-07:
> You win the award for most eye-brow raising issue of the week! Do you have
> a 2nd device capable of acting as a base? I want to remove MACORS from the
> equation.
>
> I don't *think* this has anything to do with the RTK Firmware but I'm
> eager to find out.

Hi Nathan,

  Thanks for looking into this.

  I'm afraid I don't have a second device, nor do I know anyone local that
has one.

  I have been trying various experiments to see if I can determine what
configurations cause the problem and what don't.  Unfortunately, what I've
found is that it seems to be intermittent.  Most of the time I've noticed
it I was using SW Maps, and it happened when I told them to average the
results to create a point, before that they'd have a reasonable altitude
value, and then as it averaged it would go down by about 1 meter per second
while it was averaging.  So, I initially thought it must be a bug in their
averaging system.  But, then I was just playing with it on my computer
using u-center and putty and noticed that it too was dropping at a rate of
about 1 meter per second.  Once it got down to nearly a kilometer below sea
level that was just too humorous, and I had to take a screen shot to send
to you guys!  :-)

  I'll keep futzing around with it on an ad-hoc basis and see if I can
determine what triggers the intermittent problem.  But, ya, I agree it
seems unlikely to be in the RTK firmware.  I can't imagine how it could be
MACORS either, but if you know of a different free NTRIP caster near me,
I'd be happy to add it to the mix of things to try.  It may also be that
the issue I saw in SWMAPS was different than the issue I saw on the
computer, and they just both had the strange symptom of reducing the
altitude by about 1 meter per second.  But, I have the feeling it is most
likely something in the u-blox firmware - but what it is and how to trigger
it, I currently have no idea...

Jonathan

gdt, 2023-06-07:
Also, I'd characterize this as "bad altitude" more than "decreasing".  Are the horizontal coordinates correct?  Overall this feels like marginal reception and the F9P algorithms dealing badly.

gdt, 2023-06-07:
My reference mark is X + 1.9m, ITRF2014 HAE, per OPUS, 30h occupation with borrowed old Trimble dual-frequency GPS and geodetic antenna.   I used QFIeld, and recorded into a geopackage in EPSG:6319, the values are x + 3.2m +/- 0.1m.

I guess it's a QField question if it is backing out the ortho height correction applied by the F9P, into HAE.  Certainly it should, as receiver geoid models are terrrible.   This difference is too high; I need to recheck my APC to mark correction that I did have in QFIeld, etc.  But this is not what you are seeing - chasing the last few meters is entirely different.  On the check shots, the horizontal errors (assuming OPUS is right, not entirely fair) were about 2.5 cm and 4.5cm.  This is 20-30s average with a handheld prism pole trying to keep it leveled.  (I consider this horizontal result to be excellent and as good as can be expected.)

JonathanLeeDeKock, 2023-06-07:
Hi Greg,

  Thanks for looking into this, too!

  Normally I have an SFETOP106 antenna on it.  And most of the time it
works fine and doesn’t have this issue.

  In this case, yes, I definitely had a bad signal.  I use a cheap antenna
from Amazon that I set in my windowsill (indoors with fairly horrible
reception).  I do this a day in advance of using the system so the GPS can
gather the 12.5 minutes of data it needs for full knowledge of the
satellite’s orbits so when I actually begin using it the next day it is
much closer to a warm/hot start than a cold start.

  I’ve had many cases where it worked great in the field.  In fact I have a
local geodetic marker that I was able to get a position on using it.  It’s
horizontal position was +/- 13 cm in NGS/NOAA - my GPS using MACORS took
four 4+ hour readings and consistently positioned it +/- 1 cm at an average
of 12.6 cm to the north-east of NOAA’s official position.  Similarly, it
pegged the height, that they only had accurate to within a few meters, to a
very consistent value at 54.5 meters above sea level +/- 3 cm.  So it
routinely gets a very consistent quite reasonable answer.

  That’s why it was so surprising when I saw it getting worse and worse
while I was waiting for it to gather the orbital data and ultimately
claiming I was a kilometer below sea level.  It was just too crazy - I felt
I had to report it to somebody as whatever algorithm it is that’s messing
this up so badly has some kind of bug in it - it may be a bug only
triggered by poor reception - but still there’s a bug somewhere for it to
end up a kilometer off on altitude while being well within a meter of
accuracy, and correct, in the horizontal dimensions.  But, like I surmised
before, that’s probably something down in the firmware of the ZED-F9P, I
just don’t know of any way to report it to them.

Jonathan

gdt, 2023-06-07:
Was your antenna l1 only?   I have used the SFE TOP106 indoors (top floor wooden house) and gotten FIX, and reasonable FLOAT values.

Sounds like a F9P firmware bug to me, to enter FLOAT when QC should have said not good enough.

JonathanLeeDeKock, 2023-06-09:
Yes it is L1 only.
https://www.amazon.com/Bingfu-Waterproof-Navigation-Connector-Tracking/dp/B07R7RC96G

It was doing it again to me this morning.  I tried powering it off and on a
few times, kept getting worse.  So then I swapped in my SFETOP106 antenna,
even though it was indoors.  It immediately turned around and started
moving toward the positive direction.

I should mention that I have it set in Stationary mode, rather than
Portable, in case that matters.

It seems to be doing some kind of averaging as it is not immediately going
to the correct altitude.  But with the SFETOP106 attached it is moving
towards a positive value at the rate of about 1 meter every 10
seconds, although the rate of change seems variable.  As I write this I'm
still 35 meters underground, it thinks.  It will take a while to get me
back above grade at this rate.

Note that the horizontal location is spot-on with both antennas.  It knows
what windowsill in my house I have the unit sitting in.  So even indoors
with a cheap antenna it is getting that correct.

Jonathan

nseidle, 2023-06-09:
> I would say that bad behavior with poor signals (indoors) with an L1-only antenna is basically out of scope for this product. (I am just a user, not speaking for SF!!)

Yes that.

I think this mystery has been solved. For anyone reading, high-precision GNSS requires an L1/L2 antenna (at least) with a clear view of the sky. You can experiment in doors and near a window and, with the amazing power of the ZED, you'll be pleasantly surprised at how good at is at getting a general GPS fix. But expect +/- tens of feet inaccuracy, or in this case, 1km.

tonycanike, 2023-06-09:
Since you're using an L1-only antenna, have you tried configuring the F9P to only use L1 signals?  It looks like u-center can do this with the UBX-CFG-GNSS message.  I'm not able to test this right now on my system, so it's just an idea.  I'd guess the L1-only antenna is doing something to the L2 signals, perhaps changing the phase, that the F9P just can't account for.

Or just not worry about it, as it appears that your use case for your L1-only antenna is to simply update the almanac in the F9P and not get a coordinate solution for your windowsill.  

You might try the portable instead of stationary model too.   RTK solution calculations involve combining measurements over time, perhaps only for a few second when it first fixes, and generally via a Kalman filter, so at some level one could say it's always "averaging" measurements in some fashion.  Even when they are fixed you can see the solution coordinates moving around as the RTK algorithm incorporates new data into the solution. (I use the word average very loosely.)  I suspect the stationary model configures the Kalman filter so the changes in the output solution coordinates are very slow.  I don't think this will solve the obviously wrong elevation, just help the F9P converge to a more accurate solution faster after you swap antennas without re-initializing the RTK engine.

Sometimes float solutions are pretty close, sometimes they aren't.  And I've learned that bad fixed solutions are more common than I initially expected when I first started doing RTK work, and that all bets are off for float.   Though being almost a 1km below the ellipsoid/geiod/msl is more than I've seen, so thanks for sharing this!

I'd be interested to hear what happens to the elevation if you stop trying RTK (no NTRIP feed) and just get a DGPS or Autonomous solution.  That test should still load the almanac into the F9P for your work.  (I really don't know about the last bit; in my work I don't need to save the few seconds on turn-on.  I don't worry much about getting cold vs. warm starts. It takes me a few minutes to setup my tripod, pole, and data collector so that's not important to my use.)

---you can stop reading here, the below is long and not totally relevant---

When I get fix or float solutions that I suspect to be inaccurate, generally under challenging solutions, I "dump" the receiver to force it back into autonomous mode, hopefully reinitialize the RTK engine (could be said as being analogous to starting a new average) and I then generally get a fix that's usually accurate.  I might do this 3-4 times to be sure I know which solution(s) was the bad one(s).  Sometimes I do this 10 times and never get two fix solutions within 0.25' of each other - very heavy tree canopy being a common cause.  I also do this routinely to check my solution, and I often come back to important points a few hours later or the next day as another check on the solution.  By "dump" I mean physically obstruct the antenna; but it can be done electrically.  Some receivers can be commanded to "reinitialize" the RTK engine.

It sounds like you effectively tried this via power cycling. 

As an aside, here's a GNSS dealer who built a robot to dump survey GNSS receivers to test their RTK engines. It's pretty cool and his blog has a wealth of information.  I found the video fascinating.  Video, website, and blog links:
https://www.youtube.com/watch?v=JJbHPj9oZsU
https://igage.com/
https://blog.ashgps.com/

It's well-known behavior among surveyors that fix and float solutions under challenging conditions can be flat out wrong.  The amount of computation going on in the device is incredible.  It's more of a statistical model and less of an analytical solution.  Trimbles get bad fixes, Sokkias get bad fixes, Carlsons get bad fixes, (older) Javad's , U-Bloxes too, you name it.   Just talk to a bunch of surveyors over beers or join a surveyor group on social media and you'll hear the endless stories of bad fixes, especially under tree canopy.    There are surveyors who will setup on a point, get a solution with, eg., GPS+GLONASS, then reconfigure to use Galileo+Beidou to get a second solution, and then compare the solutions.

The newer most expensive survey-grade equipment automatically simultaneously runs multiple RTK engines (with different configuration parameters) and compares the solutions to toss out bad fixes.  Carlson calls this a "Fix+" solution.  Kinda like the space shuttle having three separate computers programmed by three separate programming teams and comparing the answers.   This is what you get for $30K vs. $0.75K.  I find it very telling when the Trimbles of the world can't prevent bad fixes, but instead calculate fixes multiple different ways & compare solutions to catch bad fixes.

I will say the first time I got a bad fix or crazy solutions I thought "what the heck this isn't supposed to happen, there must be a bug!"   But I've since learned that these things aren't magic, they are just tools,  and they are pretty amazing tools at that.