sparkfun/SparkFun_RTK_Firmware #483: RTK Surveyor Crash after Profile Switch

JonathanLeeDeKock, 2023-06-03:
### Subject of the issue
RTK Surveyor Crash after Profile Switch

### Your workbench
RTK Surveyor, Firmware 3.2

### Steps to reproduce
Connect to ESP32 via USB on any COM channel, switch from any profile to any other profile.  After reboot it starts normally, then immediately crashes with this message:

[image]

After the second reboot, It then seems to be fine and returns to normal behavior.

The message references an SDSize - I do have a 32 GB SD installed in the Surveyor.

### Expected behavior
Should just reboot once and be fine, not crash.

### Actual behavior
See reproduction steps.  It crashes mid-way.

nseidle, 2023-06-07:
Sorry for the hassle. I believe we fixed this issue at [this commit](https://github.com/sparkfun/SparkFun_RTK_Firmware/commit/3d7fb8d00dd9aa88b128b684ca135c66dcf00709) and is included in [v3.4](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v3.4). Please try upgrading the firmware and let us know if that fixes the issue.

JonathanLeeDeKock, 2023-06-07:
No worries.  I've upgraded to 3.4.  I'll test it out and see what happens.

As a side note, after I upgrade, each time it reboots I get a bunch of
messages like this:

Warning: Settings size is 4928 but current firmware expects 1992.
Attempting to use settings from file.
Warning: Settings size is 4928 but current firmware expects 1992.
Attempting to use settings from file.
Warning: Settings size is 4928 but current firmware expects 1992.
Attempting to use settings from file.
Warning: Settings size is 4928 but current firmware expects 1992.
Attempting to use settings from file.
Warning: Settings size is 4928 but current firmware expects 1992.
Attempting to use settings from file.
Warning: Settings size is 4928 but current firmware expects 1992.
Attempting to use settings from file.
Warning: Settings size is 4928 but current firmware expects 1992.
Attempting to use settings from file.
Warning: Settings size is 4928 but current firmware expects 1992.
Attempting to use settings from file.
Warning: Settings size is 4928 but current firmware expects 1992.
Attempting to use settings from file.
Warning: Settings size is 4928 but current firmware expects 1992.
Attempting to use settings from file.
Warning: Settings size is 4928 but current firmware expects 1992.
Attempting to use settings from file.
Warning: Settings size is 4928 but current firmware expects 1992.
Attempting to use settings from file.
Warning: Settings size is 4928 but current firmware expects 1992.
Attempting to use settings from file.
Warning: Settings size is 4928 but current firmware expects 1992.
Attempting to use settings from file.
Warning: Settings size is 4928 but current firmware expects 1992.
Attempting to use settings from file.
Warning: Settings size is 4928 but current firmware expects 1992.
Attempting to use settings from file.

I was able to correct this by going to each profile, making a random
change, and then putting it back to the desired value.  It seemed that
would cause it to rewrite the settings file on the SD card and it would get
better.  Is there a more efficient way to get it to resave all the settings
in the new format expected by the new firmware?

PS:  So far, yes, I've been changing profiles a few times while I wrote
this response and it hasn't crashed yet after a change. So it seems to be
resolved as it did it on every profile change before...

Jonathan

nseidle, 2023-06-12:
Got it. We'll suppress that message so that it doesn't appear in production firmware.

nseidle, 2023-06-21:
I believe the primary issue has been fixed and the debug messages are now suppressed in v3.5.

closed 2023-06-21 by nseidle