sparkfun/SparkFun_RTK_Firmware #498: Compilation Error - Bluetooth

ImRaptor, 2023-06-13:
### Subject of the issue
Fails to compile if COMPILE_BT is not defined.  Error message is 'bluetoothSerial' was not declared in this scope.
Function pinBluetoothTask does not have the #ifdef/#endif block defined for COMPILE_BT so it is trying to load things that have been excluded.
Wrapping the function content with the appropriate #ifdef COMPILE_BT block corrects the issue.

### Steps to reproduce
Comment out COMPILE_BT in RTK_Surveyor.ino

nseidle, 2023-06-19:
Welcome! Thanks for reporting! 

I forgot some compile guards. [This is fixed](https://github.com/sparkfun/SparkFun_RTK_Firmware/commit/22db172282a9e3bac16eb60f28847144a7916452) in RC.

nseidle, 2023-06-21:
This has been implemented in v3.5.

closed 2023-06-21 by nseidle