sparkfun/SparkFun_RTK_Firmware #611: Locked out of Config if Wifi Not Present

mwls-sean, 2023-08-20:
### Subject of the issue
After adding in a Wifi SSID and setting the mode to `WiFi`, if that SSID isn't available, there is no way to get into Config to change things. 

### Your workbench
* What version of RTK firmware are you running? 3.6
* What radios are you using: Bluetooth, WiFi, and/or ESP-Now? Just trying to use WifI in Wifi mode (vs AP).
* Are there any additional details that may help us help you?

### Steps to reproduce
1. Configure the Facet to join a wifi SSID  and set it to WifI mode.
2. Let it join to ensure it actually works. 
3. Turn off Facet 
4. Turn off the router or similar so that SSID isn't available any longer
5. Turn on the Facet
6. In the Base (may Rover too), it will flash the Wifi icon
7. Going to Config, it will try and connect to the SSID configured, but fails, and exits Config mode

### Expected behavior
Ideally it reverts back to AP mode if it can't connect to an existing SSID while choosing the Config menu option. 

### Actual behavior
Going to Config, it will try and connect to the SSID configured, but fails, and exits Config mode. Only option is to hook up the USB and configure via Serial mode or bring the Wifi back.

nseidle, 2023-09-14:
Sorry for not responding sooner! I can confirm this on an RTK Facet L-Band that is trying to get its keys. It has to do to our recent changes to the network layer. We'll get it sorted as quickly as possible.

nseidle, 2023-09-19:
This is fixed in [v3.8](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases). Please let us know if you see any other problems.

closed 2023-09-19 by nseidle