sparkfun/SparkFun_RTK_Firmware #632: Sparkfun Facet RTK bluetooth issue with profiles?

skyviewaerials, 2023-08-30:
### Subject of the issue
Using Profiles on the Sparkfun Facet RTK rover/base the first profile is the only one to pass data to the bluetooth device.

### Your workbench
V 3.7
connections are via Bluetooth and WiFi, using SWMaps on IOS and Android
the app connects to the device but data never populates the fields.
There is an Ntrip connection that is functioning and does pass data to the phone

### Steps to reproduce
this is the condition on start up, I tried to start the device on a particular profile but unless it is the first on the list of profiles the phone
[image]
[image]
[image]
[image]
[image]
 will not get data from the device

### Expected behavior
Bluetooth connection should be available to all profiles

### Actual behavior
Bluetooth function only works with the first saved profile.

nseidle, 2023-09-07:
Cross posted here: https://forum.sparkfun.com/viewtopic.php?f=116&t=60128&p=243492

I cannot replicate this issue in v3.7 firmware. I can switch between profiles successfully and then connect to the device using [Serial Bluetooth Terminal](https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=en_US&gl=US&pli=1).

[image]

[image]

I'm able to connect and fill data over SW Maps on Android: BT Classic requires the 'SparkFun RTK' instrument model, BLE requires the 'Generic NMEA (Bluetooth LE) Instrument model.

skyviewaerials, 2023-09-07:
I installed the Serial Bluetooth Terminal app on my Android tab and I can switch between bluetooth protocols but it is still the same problem, only Profile 1 passes data, SBT connects but doesn't capture data. I tried to connect Tera Term via USB bluetooth while in Rover mode but this is the response I get.
[image]

skyviewaerials, 2023-09-10:
So does this indicate a hardware issue?

tonycanike, 2023-09-10:
You might want to try a clean slate - reformat your SD card AND reset the whole Facet to factory defaults via the serial menu.     Reconfigure the Facet and save the profile anew (don't save your old profiles, recreate them.)

I have had odd behaviors which clear up when I start from a clean slate.   Note that I am not testing your situation, this is a general suggestion.  

I believe in my instance the odd behaviors I experienced was due to testing not-quite-perfect release candidates on my hardware.

nseidle, 2023-09-11:
I like @tonycanike's recommendations.

> So does this indicate a hardware issue?

I don't believe it is. The fact that you have the default profile working normally would indicate that the hardware is capable of all things (BT, WiFi, GNSS, etc). Initially I suspected the firmware may have had an issue but we haven't been able to replicate this.

skyviewaerials, 2023-09-11:
Thank you Nathan Seidle and tonycanike, I think I have a repeatable issue.
I believe the problem stems from a configuration item, I created profiles and swopped the configuration parameter AP to WIFI, that sets the Facet to reboot the ip address as 0.0.0.0, this profile will no longer pass bluetooth data although it shows a connection, for either IOS or Android connections. The fix is to delete the profile and set the profile back to factory specs and repopulate the data by hand.

nseidle, 2023-09-12:
Sorry, I still cannot replicate. We must be doing something differently.

Note: Once Config mode has been entered, Bluetooth is disabled, Perhaps you are expecting it to work while you are configuring the device?

Here's the steps I took:
* Power on a Facet, no SD card
* Reset Facet to factory defaults
* After boot, config WiFi (option 6), set Config to WiFi:
[image]
* Exited all menus so the options would save to the profile
* Power cycle device
* From Android, forgot device 'Facet Rover-23FA'.
* Paired with 'Facet Rover-23FA'.
* I connect and get expected NMEA output.

> swopped the configuration parameter AP to WIFI, that sets the Facet to reboot the ip address as 0.0.0.0

No, that does not set the Facet to reboot at IP 0.0.0.0. This setting, if set to 'WiFi', will make the device use a local WiFi network when in Config mode, rather than the default Access Point (AP) mode. 

If the user presses the power button enough times to select 'Config', the device will shut down Bluetooth and start the Web server. If the setting is set to AP, the device will broadcast itself as an available WiFi network that you can connect to and configure the device. If this setting is set to 'WiFi', the device will try to connect to one of the available WiFi networks and the user can connect to the device over the supplied WiFi network. In neither case is Bluetooth (SPP or BLE) available.

Please let me know if you are still experiencing problems. Please provide as many steps as you can so that I can replicate.

skyviewaerials, 2023-09-12:
Hello Nathan,I see where I forgot a step, after setting up a profile using a serial connection to Terra Term to install the wifi parameters, I switch to the web interface, after selecting the parameter for wifi instead of the default AP, in the new profile, the unit resets to the 0.0.0.0 at this point, I would shut down the Facet and restart it, but that causes the bluetooth issue. What I’ve done since was to connect using the serial connection and factory reset the problem profile and reentered the data, and bluetooth functions for both iOS and Android.Seems that I’m having another issue with caster connections, I’m seeing html data coming from the Facet during serial connections when unit is sending credentials to the caster, which is rejecting the connection?Alternatively I see the caster connection continually drop and reconnect, something I can not see unless I’m using the serial connection. I have been able to get a Fix and record data, also I’ve been able to set the unit as a base and cast RTK data to a drone.Any help is greatly appreciated.Thanks Mike Sent from my iPhone

skyviewaerials, 2023-09-15:
Hello Nathan Seidle and tonycanike, this is what I'm seeing for the Ntrip connection, I'm confused by the html?
[image]

tonycanike, 2023-09-15:
I personally use my data collector software to connect to the casters I use, so I'm not the best person to answer this.  However, it looks like your login credentials are incorrect.   Note that there is a password that is only casually disguised in your above screen shot, you might want to remove the screen shot (or change the password) so someone doesn't try to use it.

skyviewaerials, 2023-09-15:
Thank you tonycanike, I removed the image. I did see that I made a mistake in the mount point for that profile. I do see connections being rejected but the credentials are correct as I am trying them on both an iPhone and an Android tablet using ntrip checker.

skyviewaerials, 2023-09-16:
Ok, I tried resetting to factory defaults but I can only connect to a single caster service, the rtk2go.com that is embedded in the firmware, I can change mount points so i have some coverage.
I have 5 different caster credentials that will not authorize.NTRIP_CLIENT_CONNECTING
NTRIP Caster responded with bad news: HTTP/1.1 401 UNAUTHORIZED
Content-Type: text/html; charset=utf-8
Content-Length: 317

<!doctype html>
               <html lang=en>
                             <title>401 Unauthorized</title>
                                                            <h1>Unauthorized</h1>
                                                                                 <p>The server could not verify that you are authorized to access the URL requested. You either supplied the wrong credentials (e.g. a bad password), or your browser doesn&#39;t understand how to supply the credentials required.</p>
This is what is being returned.
It states browser?
 Any help is greatly appreciated,
Mike

skyviewaerials, 2023-09-16:
I did a little more poking around under the hood, the only issue I see is a Fail for mux?
[image]

I tried to reset the unit to factory spec, it didn't make any difference, all of the casters I am attempting to connect to are fully qualified state and government operations. I can connect to any using my iphone or android tablet.
Thanks for looking.
Mike

nseidle, 2023-10-27:
We're closing out old issues. I think this topic also morphed into problems connecting to mountpoints outside of rtk2go/emlid. Please open a new issue if you're still having difficulties.

PR #798 (2025-09-09): Add NetworkServer
This PR:
* Adds the `NetworkServer` class which supports both WiFi and Ethernet
* Modifies PvtServer so it uses `NetworkServer` instead of being hardwired to WiFi
    * Resolves #797 
    * Includes keep-alive DNS calls for WiFi when no clients are connected - prevents disconnects due to `ASSOC_EXPIRE`
* Periodically displays the PVT / UDP Server IP address and port
    * This can be disabled if desired - via the Network menu (option 8)
* `settingsToDefaults()` calls `checkMessageRates()` to ensure the base and rover rates are correctly initialized - before being passed (e.g. [... cut]
commit 1031bd14 (2025-09-09): Resolve #632