sparkfun/SparkFun_RTK_Firmware #654: Possible failure to access wifi config with rtk.local

amlago, 2023-09-17:
Possible failure to access wifi config with rtk.local

Facet v3.6
I just tried accessing wifi config for the first time by typing rtk.local in the explorer.
I could not enter using AP mode or Wi-Fi mode.
I don't know if I have something configured wrong or if it is a firmware error.
Greetings

nseidle, 2023-09-18:
Thanks for reporting! Please try v3.8. We have fixed many bugs including a new captive portal that should remove the need for entering 'rtk.local'. Please give it a try and let us know if you still have problems.

amlago, 2023-09-18:
With v3.8, when you connect the cell phone to the Facet Ap network, the Wi-Fi config page automatically opens.
I tried several times and it worked.
It promises to be useful.
Thank you

nseidle, 2023-09-19:
This is fixed in [v3.8](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases).

closed 2023-09-19 by nseidle