sparkfun/SparkFun_RTK_Firmware #665: NMEA Stream is not being sent over bluetooth

TheCre8r, 2023-09-22:
### Subject of the issue
NMEA Stream is not being sent over Bluetooth

### Your workbench
* Version 3.8
* Serial Bluetooth Terminal or any Bluetooth software
* Default configuration, fresh out of the box.

### Steps to reproduce
No Bluetooth communication. Tested a basic simple ESP32 BIN to confirm SerialBT was working, and it works fine, so issue with the RTK firmware.
Tested COM port via USB for U-blox - it works fine, Screen shows the correct GPS satellite connections so data is being sent to ESP32. 

### Expected behavior
Should send all of the NMEA stream to the terminal.

### Actual behavior
Receiving no NMEA stream in the terminal.

nseidle, 2023-09-25:
I can't replicate this. I've got:

* RTK Facet
* Factory reset
* Connected to SW Maps and Serial Terminal over BT
* Regular NMEA and updates are seen as expected

Did you select the 'SparkFun RTK' as the Instrument Model?

TheCre8r, 2023-09-26:
I downloaded SWMaps to confirm that I am not seeing anything there, either. I have also confirmed the 'SparkFun RTK' as the Instrument Model in SWMaps. I have also attached a screenshot from Serial Bluetooth Terminal for your convenience.

[image]
[image]
[image]
[image]

TheCre8r, 2023-09-26:
I started going back through old firmware, to see if any would work, and v1.4 works great!

tonycanike, 2023-09-26:
I upgraded my Facets a couple days ago to v3.8 and I had no problems getting the NMEA data over Bluetooth to my data collector (SurvPC 7.03 on a Win10 tablet).  It just worked flawlessly.  I even got RTK Fixed solutions from my office inside my house, but that's not relevant here.  

When I do upgrade (and downgrade) firmware, I almost always reset the Facets to factory defaults.  I've gotten stuck in the distant past with goofy problems when upgrading, and resetting to factory defaults almost always solved the issues.  Over the last year I've tested a lot of release candidate (RC) firmware versions, and have often downgraded and upgraded, and I've just gotten into the habit of frequently resetting the Facets to factory defaults.   And anytime I have strange errors, I am very careful to reset the Facet to default factory conditions. 

I see you said it was default out-of-the-box; I will say there were times I **thought** I had the Facet at default settings but actually I did not.  

I find it best to:
1.  Pull the SD card out of the Facet and use a separate PC to reformat the SD card, or at least delete all the settings (profile) files.
2. If upgrading, upgrade the firmware.  (or downgrading)
3. Use the serial terminal to reset the Facet to factory defaults.  As part of this it reformats the internal file system (LittleFS or something) to clear out the stored settings (profiles).
4. Power off the Facet
5. Reinsert the now-clean SD card.
6. Power on the Facet.

Long ago I thought I'd have a carefully-curated set of settings files that I used to setup for my different work scenarios.  I'm a former programmer, and the idea of managing (and perhaps even editing) the configurations via the settings files strongly appealed to me. 

However, this has proven to not be the best solution for me.  The configuration interfaces (serial, WiFI) have gotten robust and easier to use in the past 8-12 months, and I've gotten more familiar with using them.  The Facet stores settings files both internally and on the SD card, and I always forget which one wins, especially when I'm out in the field, the mosquitos are biting, and the cows are bumping my equipment.  I suspect the settings files change internally between firmware upgrades, and some of the RC versions I've tested over the years weren't happy with that, so I just clear out the old files with every firmware change. It's faster now for me to simply set the Facet's configuration each day.  So I reset to factory defaults with every firmware upgrade and I do not try to bring in old settings file.

TheCre8r, 2023-09-27:
That sounds different from how a firmware upgrade should go... I appreciate your input, but I did not get any value out of that. I am glad it is working for you though!

nseidle, 2023-09-27:
I've tried to replicate this multiple times this week and have failed.

> That sounds different from how a firmware upgrade should go...

Tony is basically telling you to do a [factory reset](https://docs.sparkfun.com/SparkFun_RTK_Firmware/menu_system/#factory-reset).

Is your Bluetooth set to SPP? BLE would cause this. A factory reset would fix it.

Does downgrading the firmware to v3.5 change anything? v1.4 is so old that doesn't help us much nail down any change that may have affected the system.

nseidle, 2023-10-27:
We're closing out old issues. Please open a new issue if you're still having difficulties.

closed 2023-10-27 by nseidle