sparkfun/SparkFun_RTK_Firmware #666: WiFi does not work in "config" the first time entering "config"

dwightbrew, 2023-09-23:
My setup is an RTK Express running firmware version 3.8.

In "non config" mode, the RTK Express pulls a DHCP address and properly joins the network. Things work as expected.

When I switch from "non config" mode into "config" mode, the RTK Express briefly flashes a valid DHCP address and then changes to an 0.0.0.0 address. 

However, if I immediately select "config" a second time, the RTK Express pulls a DHCP address and properly joins the network. Things then work as expected the second time entering "config".

skyviewaerials, 2023-09-25:
I had something similar happen to me, I was using the bluetooth serial connection and tried to switch to WIFI config in the systems menu, the unit removed any connection information ie no SSID and ip as 0.0.0.0, additionally the unit configured in web mode would copy my first profile and place the data in any additional profile I selected even though I did no save any changes.

nseidle, 2023-10-27:
I think I just saw this today. Do you have PVT Client or Server active? That's what's currently causing problems for me. With them disabled, RTK enters Config successfully and prints a valid IP.

skyviewaerials, 2023-10-28:
I checked my config files and I do not have TCP server or TCP client activated, I just flashed the v3.10 of the firmware, I will have to check the function again.

skyviewaerials, 2023-10-30:
additionally I am having problem with keeping the wifi connection going, I updated to v3.10 and my first 2 profiles will not connect to wifi, I changed the data in the wifi menu and saved the changes and there was no effect.
I switched back to v3.9 and reloaded my profiles, I have connections wifi and bluetooth but both keep dropping.

nseidle, 2023-10-31:
> I have connections wifi and bluetooth but both keep dropping.

Can you post any serial logs that show what's happening?

skyviewaerials, 2023-10-31:
Where are the serial logs located? I will try v3.10 again, I did a test with my base running v3.10, I know it was casting an Ntrip signal so it had to be connected to wifi, as I was using profile 1_IOS Local which connects to my hotspot.
I did have and issue when connecting to the base via serial bluetooth as each time I did the base started a new ubx file?
I was trying to collect about 2 hours of data for some PPP testing. I had multiple Ntrip client connected, I connected the rover with firmware v3.9 using the LoRa Serial radios and collected a few points with SW Maps with an RTK Fix.
I have some time this morning, I will try another test using v3.10 on the rover.

skyviewaerials, 2023-10-31:
OK after uploading v3.10 to the rover and testing, I noticed that the Ntrip client wasn't enabled for the first 2 profiles, after enabling both, wifi connected.
Is this expected, there will not be a wifi connection unless Ntrip client is enabled?

nseidle, 2023-11-01:
> Where are the serial logs located? 

These are normally captured using tera term or other terminal program while [connected over serial](https://docs.sparkfun.com/SparkFun_RTK_Firmware/configure_with_serial/).

That is correct. The device won't connect to WiFi unless WiFi is needed. For example if any of the following 'users' need connectivity, then the networking layer will turn on:

* NTRIP Client/Server
* PVT Client/Server
* UDP Server
* Auto-update

skyviewaerials, 2023-11-01:
Thank you for the explanation Nathan, I was using U-Center via serial to connect to the Facet. It seem the program was turning off the Ntrip client for the particular profile I was working with.

skyviewaerials, 2023-11-07:
This shows up, I have to restart the profile and then the unit function correctly.
[image]

dwightbrew, 2023-11-20:
I originally posted: "When I switch from "non config" mode into "config" mode, the RTK Express briefly flashes a valid DHCP address and then changes to an 0.0.0.0 address."

Setup: RTK Express running firmware 3.9 and 3.10 (and previously 3.8)

I originally reported this issue running firmware version 3.8. I didn't / don't seem to have the problem running firmware 3.9. However, I do when running firmware 3.10. 

Today I switched between 3.9 and 3.10 and back several times. Running 3.9 I never have this problem with the 0.0.0.0 address and running 3.10 I always have this problem

In version 3.10, reentering config mode a second time pulls a valid DHCP address, but there seem to be issues when configuring the device using the web interface. The second time entering "config" mode, the RTK Express pulls a DHCP address, properly joins the network., and serves up the web configuration pages. However, it is difficult to change settings as the RTK Express sometimes doesn't respond. Hitting refresh in the browser always reconnects, but the refreshed page doesn't always agree with the stored configuration.

skyviewaerials, 2023-11-20:
Hitting the refresh when using the web interface causes my profiles to duplicate into alternate profile positions, I try to avoid the web interface when making major changes.
I keep a set of profiles to swap when this issue causes problems, as the profile files are changed inadvertently.