sparkfun/SparkFun_RTK_Firmware #667: Web TCP Client and TCP Server Check Boxes Do Not Reflect Current State

dwightbrew, 2023-09-23:
Device: RTK Express
Firmware: 3.8

When I configure the RTK Express using the Web interface, in the WiFi Configuration menu, the TCP Client and TCP Server checkboxes do not reflect the actual state of these options in the configuration file. The check boxes always show up unchecked after checking one, saving the configuration, and then refreshing the web page.

nseidle, 2023-10-27:
I tried duplicating this today using d3.10 firmware. The check boxes are getting saved and correctly transmitted. Perhaps we fixed it in the interim firmware changes. I'm going to mark it as fixed for now. Please give the d3.10 binary a try (should be released shortly) and let me know if you're still seeing the issue.

nseidle, 2023-10-27:
This should be fixed in [v3.10](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v3.10).

closed 2023-10-27 by nseidle