sparkfun/SparkFun_RTK_Firmware #669: Doubt about the maximum number of Facet channels.

amlago, 2023-09-24:
Facet v3.8
Using the u-center I see what appears to be a maximum of 60 channels in use.
The zed-f9p is supposed to have just over 180 available channels.
Can someone explain to me what causes this difference?
I add screenshots of the u-center.
greetings

amlago, 2023-09-24:
Image
[image]

amlago, 2023-09-24:
[image]

nseidle, 2023-09-29:
Interesting question. I brought it up with u-blox: 

"The listed channels assignment in your ubx-cfg-gnss is used for M8 and before and are for L1 only band. For the F9/M10/F10 products the “Gen 9 configuration” should be used, and in that window you no longer have the ability/need to set tracking channels as the firmware will dynamically assign based on the current number of SV deployed, but with 184 channel, it will never be an issue."

It sounds to me like it's not a concern.

Thanks for asking!

closed 2023-09-29 by nseidle