sparkfun/SparkFun_RTK_Firmware #670: Feature request?

skyviewaerials, 2023-09-24:
### Subject of the issue

A number to indicate which profile is being used on the Facet RTK screen.

### Your workbench
v3.8, v1.32

Bluetooth, WiFi, SW Maps  transmitting NTRIP back to the device

### Steps to reproduce
The number for the profile could be placed where the recount number is for debugging, maybe a dash under the number to indicate it is a profile and not the reset number? It could flash between the profile number and the debug number when debugging?

### Expected behavior
When the unit starts the number displayed would be the profile in use.

### Actual behavior
The unit starts and there isn't any indication of the profile in use at the moment.

nseidle, 2023-09-25:
Thanks for asking! This is a reasonable request. 

> maybe a dash under the number

This is a good idea but requires custom fonts which we're not able to do at this time.

The number of characters is pretty limited to 6 (usually). Do you think '3_MyPr' is recognizable for 'MyProfile' in profile #3?

skyviewaerials, 2023-09-25:
That sounds like a good way to do it, since my profiles are for different city locations, ie: Fullerton is profiles #4 that would show as '4_Full'.

nseidle, 2023-10-27:
This has been fixed in [v3.10](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v3.10).

closed 2023-10-27 by nseidle

skyviewaerials, 2023-10-28:
That is great, I see there is an extra space before the number 5 profile.
Additionally can the profile number be shown on the display while the Facet is functioning?
I was thinking it could go in the same spot as the debug reset counter? The debug counter could flash or a dashed line under the number to differentiate it from the profile number?
Thanks for all the hard work on such an exciting product.

nseidle, 2023-11-01:
> I see there is an extra space before the number 5 profile.

What is your #5 profile name? The text is centered so you may see '5_ABC' centered (spaced before/after).

> Additionally can the profile number be shown on the display while the Facet is functioning?

Please create a new request.

> Thanks for all the hard work on such an exciting product.

Flattery will get you everywhere, including feature requests. Thank you for saying thank you!

skyviewaerials, 2023-11-01:
I checked my profiles and the issue is with 5_emlid, would this cause the extra space to show in front of the number?