sparkfun/SparkFun_RTK_Firmware #671: PointPerfect L-band keys missing at power on with Custom Thingstream device profile token

edgecase14, 2023-09-25:
### Subject of the issue
Using custom hardware based on SparkFun ESP32, firmware v3.5 and v3.7, you must go to PointPerfect setup, Disable PP, Enable PP, then exit, now L-band corrections work.

### Your workbench
* What version of RTK firmware are you running? 3.5 or 3.7
* What radios are you using: Bluetooth, WiFi, and/or ESP-Now? Bluetooth serial terminal.
* Are there any additional details that may help us help you?

### Steps to reproduce
Power on.  bluetooth serial terminal, enter +++ to get setup menus.
Choose s - it displays L-band Online, no keys.
eXit, go to PointPerfect setup, choose 1) disable, then 1) enable.
exit, go to setup (s)
it displays L-band Online, keys valid.  After a minute or two, RTK Fixed is achieved.

### Expected behavior
Keys should be available at bootup, without using bluetooth serial terminal.

### Actual behavior
No PP corrections will happen without manual intervention.

nseidle, 2023-09-29:
Hi edgecase14 - Sorry this isn't working for you! We're focused on the 'L-Band through SparkFun' path. If you would like to take a run at fixing it, I would welcome the PR.

edgecase14, 2023-10-03:
PR sounds great, that way I don't have to maintain a fork (yet).

PaulZC, 2024-06-13:
I believe we've fixed this. I'm able to get keys and corrections from power-on using a custom device token with the latest Release Candidate firmware. Closing...

Best wishes,
Paul

closed 2024-06-13 by PaulZC