sparkfun/SparkFun_RTK_Firmware #677: L-band corrections stop while in Bluetooth serial console echo mode

edgecase14, 2023-09-26:
### Subject of the issue
RTK Fix lost after having +++ Bluetooth serial console open for a minute.

### Your workbench
* What version of RTK firmware are you running? 3.7
* What radios are you using: Bluetooth.  What app are you using to connect over Bluetooth - Android Bluetooth serial terminal.  Are you transmitting NTRIP back to the device No

### Steps to reproduce
Unit is in RTK Fixed mode with PointPerfect L-band corrections.  Connect with Bluetooth serial terminal, enter console with +++
After a short time, HPA jumps, due to lost RTK Fixed status.  It does not return until "b" is chosen to exit console mode.  You can leave the bluetooth NMEA stream run, and it will pick up the fix again.  You can verify this by entering +++ console again, quickly checking status, then "b" to exit console again... if you don't linger, the RTK Fixed status stays.

### Expected behavior
I don't see any theoretical reason why the firmware can't keep relaying the SPARTN messages from the NEO-D9S to the ZED-F9P, even when the bluetooth console is open.

nseidle, 2023-09-28:
Opening the system menu blocks the loop() functions, including updating the GNSS with PMP messages (aka encrypted SPARTN).

We are out of RAM so we can't spin up another task to handle menu servicing. Right now, I don't see a way we can fix this without risking system instability. 

Normal RTK Fix should be maintained for nearly 30 seconds before the ZED drops to RTK Float due to lack of correction information. Is this blocking your normal use? Can you describe your use case further?

At this time, I don't think we can fix it, but I'll put this into the requirements for the next generation of products (that have more RAM).

edgecase14, 2023-10-03:
Ok thanks for explaining, perhaps the resolution can be just adding this to the documentation.  Is there a new platform on the horizon?  I noticed that the ESP32 S3 and other variants have the same or less RAM.

nseidle, 2023-10-03:
> perhaps the resolution can be just adding this to the documentation.

Great point. Did you know you can do PRs for docs as well? (wink wink nudge nudge) I'll leave this open until we get it documented.

Yes, we are moving to an ESP32 with more RAM. The S3 doesn't support Bluetooth 2.0 (needed for SPP) so we will not be moving to it.

nseidle, 2023-10-27:
This has been documented [here](https://docs.sparkfun.com/SparkFun_RTK_Firmware/configure_with_serial/#l-band-performance-during-serial-configuration).

closed 2023-10-27 by nseidle