sparkfun/SparkFun_RTK_Firmware #688: Power LED does not turn on (until 20 secs after power on)

velle, 2023-10-03:
Is this behavior 

**Hardware/firmware**

I have a Sparkfun RTK Surveyor, purchased in December 2021, running firmware version 2.5. A Qwiic display is connected. 

**Issue**

- Normal behavior (afaik): Power LED turns on, roughly 3 seconds after flipping power switch. Qwiic display alive 1 second after power on.  
- Occasional unexpected (faulty?) behavior: Power LED does not turn on until 17-20 seconds after power on. Meanwhile PPS led starts blinking at 4 seconds after power on. Qwiic display never shows anything. 

**A few days ago**

I had been using it for an hour, and suddenly bluetooth connection (to my Android phone running SWMaps) was gone (this often happens) so I powered off and on. The power LED did not come on, but the PPS LED was blinking, and only after around 20 seconds did the Power LED come on. The video describes the behavior. 

VIDEO

I tried maybe ten times, with and without power via USB, and the behavior was the same. The Qwiic display never showed any activity. 

When it finally was done booting and the power LED was on, then I was unable to make a bluetooth connection, however I only attempted this once. 

I went home, and after two days I powered on again, and then it started as expected with no issues. 

**First occurance, half a year ago**

At that time I did not have the Qwiic display at that time. 

It had been in storage for a few months and I was expecting it to work. When I turned it on there was (as far as I remember) no activity in any LEDs). I was indoor, but if I had been outdoors maybe the PPS LED would have blinked. I tried many times over the course of two days. 

The Power LED did turn blue when I connected it to power via USB. 

Suddenly. After experimenting for hours, it turned on as expected.

nseidle, 2023-10-03:
Thanks for reporting. Excellent videos, they really help us see what's going on. I also love the waterproofing! Nice work.

> Power LED does not turn on until 17-20 seconds after power on.

FYI - The ESP32 lights up all the LEDs (like a car dashboard) to confirm they are all functional after all the setup functions have completed, to indicate to the user that it's now ready to use. The ESP32 is obviously running the firmware, there's just a big delay, but only sometimes. 

I've seen a long boot process occasionally after an ESP32 has been programmed for the first time. This is normally caused by the initial formatting/initialization of the LittleFS file system. Once that LFS is in place, boot times come down to ~5 seconds or less. So I don't think it's that.

To really trouble shoot the issue, we would need you to connect to a [serial terminal](https://docs.sparkfun.com/SparkFun_RTK_Firmware/configure_with_serial/) to view the output from the 'ESP32 CONFIG' port. That would give us a lot more insight into what's causing the delay.

Thoughts:

* Does the problem change if a display is not connected?
* Is an SD card inserted? Does the problem change if an SD card is not present? If an SD card is present, does formatting change the problem?
* If you're open to it, please [upgrade to v3.9](https://docs.sparkfun.com/SparkFun_RTK_Firmware/firmware_update/#updating-firmware-using-the-uploader-gui). Does the problem change?

We're also happy to have you return the unit for analysis. I haven't seen a slow failure like this before.

velle, 2023-10-03:
Thanks for the friendly and helpful response :)

Yes, an SD card was inserted. I believe that I did try removing it when it happened 6 months ago, but I cant say with certainty. 

Currently the surveyor is booting like it should, but if it happens again, I will try all of your suggestions :)

And in any case, I will upgrade the firmware as soon as I have a little more time. 

**How unusual are my bluetooth / wifi issues, and could it be related?**

I have in general had some issues with unstable Bluetooth and WiFi connections. When set up to connect to an access point, it cant be trusted to actually connect. More often than not it took several retries (some times as much as 5 retries), waiting at least a minute for each retry, before finally getting it to connect. This happened in different settings, including on the country side where I would expect no interference. That's actually why I bought the Qwiic display, because I was hoping it could give me some hints on why it was not connecting. In the end I decided to not use wifi at all, and only use bluetooth. Bluetooth is also a bit unreliable, but only to the extent that now and then (no more than once per hour) I realize that it has disconnected, and I don't see a good reason that it should have disconnected. Is the unreliable WiFi a known/common issue? Is it likely that my issues with with wifi/bluetooth are related to the boot issue?

nseidle, 2023-10-03:
> Thanks for the friendly and helpful response :)

Of course! Sorry for the trouble your unit is causing.

> Is the unreliable WiFi a known/common issue?

In short, no, it's not a known/common issue. We have slowly but surely squashed most bugs out of the firmware, and many (not all) of the hardware bugs. That said, we still rarely get units that fall outside of the curve. A pseudo-failing or slow failing Bluetooth/WiFi is interesting. 

Let's start with a firmware upgrade. If you still see WiFi/BT issues please let us know and we'll get it back for analysis.

nseidle, 2023-10-27:
I'm going to close this issue due to lack of activity. Please feel free to reopen with new information.

closed 2023-10-27 by nseidle