sparkfun/SparkFun_RTK_Firmware #689: New Feature Request - S/W built in for 24 hr survey in for base rtk station.

j-w-bullfrog, 2023-10-05:
### Subject of the issue
S/W built in for 24 hr survey in for base rtk station.

### Your workbench
Currently using a surveyed in rtk express as a base station.

### Expected behavior
Would like to have the unit perform this function on request and also have a cellular data connection. We are planning on building robotic systems for ZRT mowers and don't trust the customers to get the station surveyed in correctly.

nseidle, 2023-10-05:
We've thought about this feature before and we haven't arrived at a good solution to automate the data conversion and submission steps. 

We have units with cellular on the roadmap. It's hard to say when they'll be ready.

Are you planning on installing a permanent base at each site? Doesn't security become a concern? Why not use L-Band instead?

j-w-bullfrog, 2023-10-06:
Uncertain.  We love the 14 mm repeatability of of the surveyed in RTK, But we also realize that PP via celular may be sufficient. We also want cell for diagnostics, updates, etc. Not really worried about security (maybe we should be) but we are looking at 2(?) + acres to golf courses so we are expecting more rural to suburban locations.

gdt, 2023-10-06:
It strikes me that this feature is beyond the code/memory budget of the ESP.   I would focus instead on an API to gather raw data for some period (4h-48h) and obtain that, and to be able to set the survey-in cooods, with off-device software to then interface with various static carrier phase services like OPUS.    If you really want accuracy and not just repeatability, I don't see anything other than full-on static carrier phase as being the right approach.

Product wise, it might be that one needs a tiny unix computer with modem etc. to host all of this.

As it is,this enhancement request is unclear about what it really means to do.

tonycanike, 2023-10-12:
I too don't understand exactly what this request is really asking for. 

Does the end application really need accurate absolute in-the-real-world coordinates?  Or would relative precise "turn the mower withing 0.014m of the same spot every time" suffice?

Will the end user be mounting their base on a permanent mount and so the base will in the same physical location every day of operation?

How will the user specify where the mower turns?  
--Will someone figure it out in CAD using accurate absolute real-world coordinates?  If so,what datum will be used?  Will you need a surveyor to provide an existing survey so the CAD engineer has good coordinates?
-- Will the user manually drive the mower to "train" it and the mower just needs to follow the same path?  If it's the latter, do you really need accurate coordinates or just repeatable relative precise coordinates?

My questions are getting at do you need accuracy or precision?   Those two things are different, and there are different approaches to efficiently achieve each one.

j-w-bullfrog, 2023-10-12:
Actually its need to be under about 100 mm of the path it previously followed.  14 mm is the bottom number that they will report for pdop. From U-Blox manual "Position dilution of precision"

DavidKelleySCSC, 2023-10-22:
I would like to understand this user need a bit better.  At the core, you want to grab a long periods of measurements (~24 hours) so that you can then post process them (translate the raw RTCM 3.x to RINEX, decimate them, and submit to the local version of OPUS etc.) to get a precise location in a known datum.  And presumably this is largely automated so you can just point to a base (directly or as found on NTRIP) and come back latter for the resolved answers.   

Got that part.  **Now the questions**.  As [tonycanike](https://github.com/tonycanike) asked, is it abs or relative position you want? Asked another way, Do you in fact want to know the datum (frame of reference) or is repeatability the primary goal here.  As these bases fixed or are they set into place for the mowing job and then moved elsewhere?  Do you need to be alerted if the based moved? And I presume these are short baselines, so things like tide loading do not apply. 

[In my world (as an NTRIP Caster developer) there are a large number of base station operators who do know where their base is placed so this need overlaps with that one (because then the end user (NTRIP Client device) does not know either). And we also have a number of high-grade deployments/operators who want to offer the ability is translate from one datum to another on the fly for such users.   We have solutions to this today but they required manual effort to use them an automating this is what grabbed my attention. Thanks for any feedback.]

j-w-bullfrog, 2023-10-22:
My request is based upon this page; https://learn.sparkfun.com/tutorials/how-to-build-a-diy-gnss-reference-station#gather-raw-gnss-data

It is my understanding that the accuracy of the correction factors is impacted by knowing the accuracy of the position of the antenna. That said, my co-developer is enamored with the relative positioning provided by a fixed antenna. That said, from a engineering viewpoint, PointPerfect is probably sufficient for our purposes. I have a taken data with 'survey" in mode, and we will be looking at it to see how it effects the ability to repeat a path within our required bounds. In the end its about options and trade-offs.

tonycanike, 2023-10-22:
To me the words "survey in" in the title of this request are confusing. 

The words "survey in" have a very specific meaning that seem to be very different from what the original requestor actually wants to do. 

u-blox, and hence SparkFun, uses
 the words "survey in" to refer to the process of gathering **autonomous solutions** (coordinates, perhaps +/- 1-2m) for a short period of time (perhaps 60 seconds), averaging those solutions to one set of coordinates, and setting the base location to that one set of coordinates. These coordinates are not terribly accurate but they will give excellent relative positioning for the RTK session, and could be used to give excellent relative coordinates for future sessions.  

Gathering **raw** observations for 2+ hours, post-processing those solutions to get a very accurate solution (eg using OPUS to get +/- 1cm coordinates), and setting the base to that coordinate solution is not what u-blox users usually mean by the words "survey in".  

The process described in 
https://learn.sparkfun.com/tutorials/how-to-build-a-diy-gnss-reference-station#gather-raw-gnss-data 
is not what u-blox users usually mean by the words "survey in".  

Here are two references to what "survey in" means in our u-blox/SparkFun world: 

1. The section "Setting Up a Temporary Base" in
https://learn.sparkfun.com/tutorials/setting-up-a-rover-base-rtk-system

2. https://docs.sparkfun.com/SparkFun_RTK_Firmware/correction_sources/#temporary-base

gdt, 2023-10-22:
> My request is based upon this page; https://learn.sparkfun.com/tutorials/how-to-build-a-diy-gnss-reference-station#gather-raw-gnss-data
>
> It is my understanding that the accuracy of the correction [...]

I'm really having a hard time understanding you.    If a station is *generating* RTCM3 reference data (strictly, these are not "corrections"), then if the assumed coordinates of the antenna phase center (APC) are not correct, then any positions computed from that reference data will be in error by that amount.   If that's what you meant, that makes total sense.   I don't follow "relative positioning provided by a fixed antenna".   If you mean "set up a base with wrong coords, and do RTK, and be happy if the difference between any pair of points is accurate, even if they are all wrong", then ok, but I think everything needs to be clearer.

gdt, 2023-10-22:
Everything @tonycanike  said makes senes to me.   I really don't see this "gather 24h RAW observations" being something that needs to be in the firmware.  Somebody needs to write some code to run on a unix computer to take raw obs, call opus, and send a "set the base" API call.   But given that we are talking about possible workflows of setup/rove/leave, then maybe we really want "do survey-in (original meaning), take RTK positions on rover, simultaneously log RAW, and then later OPUS and then compute adjusted RTK positons, and finally use them".  But ~none of that is going to make sense in the ESP.

j-w-bullfrog, 2023-10-23:
U_Blox supports a msg that gives relative distance and direction of the correction antenna in the unit receiving the corrections. (175 m & 210 degrees)

gdt, 2023-10-23:
That message lacks height, and it seems best to record positions and then apply a delta in XYZ later.

DavidKelleySCSC, 2023-10-23:
A few comment on definitions (all IMHO but common to the industry), soapbox on. 

> a msg that gives relative distance and direction of the correction antenna

This distance is called the **_baseline_**.    I suspect you would want the final delta (more commonly called the **_offset_**) to be LLH rather then ECEF XYZ.   The direction/heading (angle) given us usually measured from rover as a point of initial departure, the angle is of course different at the destination and along the route, but often this can be ignored. Similarly, the very slight angle between "up" and the local gravity and geoid model is also typically ignored for short distances. 

The term "**_raw_**" is used in various ways by various manufactures and published standards.  It most often means the data from the GNSS base, regardless of the message format used.  So it might mean RTCM 3,x messages, or it might mean the Ublox UBX format, or some other format.  This is made more confusing to new users by the common practice of referring to the base stations "raw observables" which, as noted, are what is actually sent to the rover device.  [RTCM ver 2 sent range offsets for each satellite as its corrections, RTCM ver 1 sent an actual XYZ offset, RTCM 3 sends the raw observable measurements and lets the rover devices work it out.]

**All** RTK positioning is always **_relative_** to the location of the base station antenna, right or wrong.  As I have posted in other places, what the Base Station says about its antenna phase center location is taken as the word of God, right or wrong.   And, as noted, any error directly translates to the rover device.  In theory (but perhaps not in practice) various frames of reference (which we can call coordinate systems here) refer back to the prime meridian and a model of the earth and some would say that is, or represents, an **absolute** position.  Others disagree and point out that various daily deflections in the earth itself make this nothing but an average guess.  A great deal of barely and hops has been consumed over this topic.  

Any RTK solution that is working (has achieved a fixed ambiguity solution it can hold) is going to very **_precise_**.  Whether it is **_accurate_** or not depends on the presumed quality of the base station location it is taken from.  Many people use the concept of **_repeatability_** here, because their own primary mission need to to able to return to the same area/region/field and get the same result when they occupy the same location the next year/season/session.  

The term "**survey-in**" came to the GNSS world from conventional survey techniques and simply means to generate some local truth you can then work with.  When they did your home boundaries they can to first surveyed-in from nearby established monuments before they could do the work.  The term is at least 100 years old, probably much older.   In the GNSS world this term started to be used about 30 years ago convey to the early user community what the base was doing as it developed a local average position it could then use for the the measurement set (campaign).  Nothing mysterious there, and folks still post-process the data for a final set of positions as a routine best practice.  Just about every vendor has a "push a button and wait'' feature of this type, although the internal design details vary.  Soapbox off. 

Jumping back the original post that started this,... I am still not getting a clear view of the problem to be solved (and with all respect to Dr Fermi not at a higher level either).  I think the best advise was been from gdt that you should just develop a suitable API to extract the needed raw data when you want/need it.    [And I think there is a cross cutting need for a general tool that can grab a set of obs data from such a base and determine/check its ECEF location and map it back to a known/presumed frame of reference.]

nseidle, 2023-10-27:
I have dreamed of the following:

* A device is deployed and configured to "become a permanent base". 
* Device gathers raw data for 24 hours (or a length of time the user sets)
* Device converts data as needed. In my head, I need to convert the RAWX/UBX to OBS (I currently use RTKLIB for this, but I suspect I can do it programmatically in real time). Perhaps I'm wrong.
* Device compresses the OBS.
* Device submits the OBS to CSRS (or equivalent) for analysis. I don't have a good plan for how to automate this.
* Device monitors for a response via email. Once the email is received, the device parses the ECEF coordinates and enters base mode using the coordinates.

I'd like to hear from @j-w-bullfrog if that's what they are looking for.

Let me be clear, if this is indeed roughly what the feature request is, it's not possible nor are we looking into implementing it at this time. But I think it's possible and interesting/challenging/fun to attempt. Might be a fun side project.

gdt, 2023-10-27:
@nseidle Broadly your list of goals makes sense, except also include OPUS not just PPP.   However as you say this is a real-computer class workload, not an ESP workload, and I don't think it makes sense to do in the sparkfun express.   It could be that the SFE has apis to gather and to set, and the real work happens on a computer or a phone..

Whether it's email or something else is a detail.   Ideally these services would have a nice M2M API.

tonycanike, 2023-10-27:
I agree with @gdt and @nseidle (if I understand them correctly) that this is not a reasonable feature for the RTK firmware.  

This would be better implemented in a companion app.

j-w-bullfrog, 2023-10-27:
Yep, that's basically it.  I followed the process to get the current base station we have working. the process  https://learn.sparkfun.com/tutorials/how-to-build-a-diy-gnss-reference-station#gather-raw-gnss-data

nseidle, 2023-10-27:
Thanks j-w-bullfrog. A fine idea, but we can't get it done with our current setup.

closed 2023-10-27 by nseidle