sparkfun/SparkFun_RTK_Firmware #690: When RTK2GO mount point is down, RTK Express unit says its still receiving data.

j-w-bullfrog, 2023-10-06:
### Subject of the issue
When RTK2GO mount point is down, RTK Express unit says its still receiving data.

### Your workbench

RTK express d.3.9 running as fixed base. RTK Express + running as rover getting rtk correction signals via client from mtpoint via wifi.

### Steps to reproduce
First discovered when trying to see how close the base unit could report correction data to the rover. Left running over night, but forgot to plug the base unit to power. Next morning rover says its still getting data. Restarted it and left it run for a while, then took it down to config. Unit still reported getting data. Both for fixed and survey in.

### Expected behavior
Stop showing its receiving correction data

### Actual behavior
Doesn't matter if mtpoint is up or down. Still says receiving data.

nseidle, 2023-10-06:
> RTK Express unit says its still receiving data.

What does this mean? Are you seeing an icon that leads you to believe it's receiving data? Or is there a serial message your seeing? Or is the RTK Express showing as having an RTK Fix within SW Maps?

j-w-bullfrog, 2023-10-06:
The icons on the small display screen show that I'm still receiving data via wifi. And the double crosshairs(?) are still saying its working. The values that its reporting are in tenths of a meter which makes it very strange..

nseidle, 2023-10-06:
To try to replicate your problem, this is what I did:
* RTK Surveyor connected to [bldr_SparkFun1](http://rtk2go.com:2101/) mountpoint on RTK2Go
* RTK Fix achieved
* I turned off the base station (bldr_SparkFun1)
* Within 10s RTK2Go de-listed the mountpoint 
* Within 15s the RTK Surveyor was disconnected from RTK2Go. Output below:
[image]

This is what I expect. I cannot replicate your issue.

I am pretty sure your device was getting corrections from somewhere. The down arrow that blinks momentarily over the WiFi symbol indicates data movement, which is entirely independent of the double cross hairs (which only occur what then module reports RTK Fix). I dunno where you were getting RTCM corrections, but it was coming in from somewhere. I'm going to close this issue since I don't think it's a problem but if you have any other tests you'd like me to run, please let me know.

closed 2023-10-06 by nseidle

j-w-bullfrog, 2023-10-07:
I will have a good look at this. Familiar with where to find it.

j-w-bullfrog, 2023-10-09:
Did the same test as you, got the same results using wifi and bluetooth. Had to reset my wifi profile since something must have corrupted itself. That said I was trying to compare my surveyed in fixed base to survey in mode to compare correction results. Back to that.