sparkfun/SparkFun_RTK_Firmware #692: Strange Wifi behaviour (fixed)

Stefal, 2023-10-10:
### Subject of the issue
A Facet was not broadcast the SSID in Wifi Config

### Your workbench
 - This Facet was using an old firmware (I don't remember the release number)
 - When I was in Wifi config mode, the lcd displayed the SSD "RTK Config" and the ip address
But !
 - There was no RTK Config network available

After several factory reset, and upgrade to firmware 3.6 and 3.7 ... still no wifi network available.

With the serial link I've switched the Facet from AP to Wifi and entered the network/password of my local wifi network. It worked !!
Then I switched back to AP, and now everything is Ok, I can see the RTK Config wifi network.

weird !

nseidle, 2023-10-27:
Weird indeed! I'm not sure what the issue could be. I've seen the ESP-NOW radios sometime interfere with the WiFi. Were you using ESP-NOW at all? 

I'm going to close for now. Thanks for reporting. Please let me know if it comes up again.

closed 2023-10-27 by nseidle

Stefal, 2023-10-27:
>  Were you using ESP-NOW at all?
Not at all.

nseidle, 2023-12-27:
Adding cookie crumbs: I believe this issue was related to #741 and may be fixed in v3.11 firmware going forward.