sparkfun/SparkFun_RTK_Firmware #693: FacetLB - setting to BLE appears to start an endless reboot loop

tonycanike, 2023-10-12:
### Subject of the issue
When I configure the FacetLB to use Bluetooth BLE mode (for SWMaps on iPhone), it appears to get stuck in an endless reboot loop.  

I can not connect to the FacetLB with SWMaps; I presume it doesn't see the FacetLB as the FacetLB is rebooting.

### Your workbench
Facet L-Band with V3.9

### Steps to reproduce
1. Turn on FacetLB
2. Use WiFi AP config or Serial Config to set Bluetooth mode to BLE.
3. Exit configuration.
4. Unit goes into endless reboot cycle.
5. Connect and start Serial Config
6. Use serial menu to turn off Bluetooth or set it to Classic mode.
7. Unit appears to operate properly.
8. Use serial menu to set Bluetooth to BLE.
9. Unit reenters reboot loop.
10. Use serial menu to turn off Bluetooth or set it to Classic mode.
11. Unit appears to operate properly.
12. repeat....

I tried doing a factory reset, configuring WiFi credentials for my WiFi network, letting it get it's PP keys, setting BLE, and saw the same behavior.

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8
Display started
LittleFS Started
ZED-F9P firmware: HPG 1.32
SparkFun RTK Facet L-Band Direct v3.9
Reset reason: ESP_RST_PANIC
Profile 'Profile1' loaded
Fuel gauge configuration complete
Batt (71%): Voltage: 4.04V Charging: 27.25%/hr Green
GNSS configuration complete
NEO-D9S firmware: PMP 1.04
System time set to: Thursday, October 12 2023 15:10:11
STATE_KEYS_STARTED --> STATE_KEYS_NEEDED, 2023-10-12 15:10:11.076
STATE_KEYS_NEEDED --> STATE_KEYS_DAYS_REMAINING, 2023-10-12 15:10:11.128
Batt (71%): Voltage: 4.04V Charging: 27.25%/hr Green
Days until PointPerfect keys expire: 58
STATE_KEYS_DAYS_REMAINING --> STATE_KEYS_LBAND_CONFIGURE, 2023-10-12 15:10:13.221
STATE_KEYS_LBAND_CONFIGURE --> STATE_ROVER_NOT_STARTED, 2023-10-12 15:10:14.926
Bluetooth Low-Energy broadcasting as: Facet L-Band Direct Rover-515E

Stack smashing protect failure!

abort() was called at PC 0x4014008f on core 0

Backtrace:0x40084ad9:0x3ffe1c900x40096255:0x3ffe1cb0 0x4009c2ad:0x3ffe1cd0 0x4014008f:0x3ffe1d50 0x4016ddf2:0x3ffe1d70 0x401b44e9:0x3ffe1dd0 0x40193773:0x3ffe1df0 0x40197c2d:0x3ffe1e10 0x4019235b:0x3ffe1e30

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8
Display started
LittleFS Started
ZED-F9P firmware: HPG 1.32
SparkFun RTK Facet L-Band Direct v3.9
Reset reason: ESP_RST_PANIC
Profile 'Profile1' loaded
Fuel gauge configuration complete
Batt (71%): Voltage: 4.04V Charging: 20.38%/hr Green
GNSS configuration complete
NEO-D9S firmware: PMP 1.04
System time set to: Thursday, October 12 2023 15:10:21
STATE_KEYS_STARTED --> STATE_KEYS_NEEDED, 2023-10-12 15:10:21.825
Batt (71%): Voltage: 4.04V Charging: 20.38%/hr Green
STATE_KEYS_NEEDED --> STATE_KEYS_DAYS_REMAINING, 2023-10-12 15:10:21.880
Days until PointPerfect keys expire: 58
STATE_KEYS_DAYS_REMAINING --> STATE_KEYS_LBAND_CONFIGURE, 2023-10-12 15:10:23.971
STATE_KEYS_LBAND_CONFIGURE --> STATE_ROVER_NOT_STARTED, 2023-10-12 15:10:25.792
Bluetooth Low-Energy broadcasting as: Facet L-Band Direct Rover-515E

Stack smashing protect failure!

abort() was called at PC 0x4014008f on core 0

Backtrace:0x40084ad9:0x3ffe1c900x40096255:0x3ffe1cb0 0x4009c2ad:0x3ffe1cd0 0x4014008f:0x3ffe1d50 0x4016ddf2:0x3ffe1d70 0x401b44e9:0x3ffe1dd0 0x40193773:0x3ffe1df0 0x40197c2d:0x3ffe1e10 0x4019235b:0x3ffe1e30

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8
Display started
LittleFS Started
ZED-F9P firmware: HPG 1.32
SparkFun RTK Facet L-Band Direct v3.9
Reset reason: ESP_RST_PANIC
Profile 'Profile1' loaded
Fuel gauge configuration complete
Batt (71%): Voltage: 4.04V Charging: 24.75%/hr Green
GNSS configuration complete
NEO-D9S firmware: PMP 1.04
System time set to: Thursday, October 12 2023 15:10:34
STATE_KEYS_STARTED --> STATE_KEYS_NEEDED, 2023-10-12 15:10:34.076
Batt (71%): Voltage: 4.04V Charging: 24.75%/hr Green

SparkFun RTK Facet L-Band Direct v3.9
** Bluetooth Low-Energy broadcasting as:  **
Menu: Main
1) Configure GNSS Receiver
2) Configure GNSS Messages
3) Configure Base
4) Configure Ports
5) Configure Logging
6) Configure WiFi
7) Configure Network
p) Configure User Profiles
r) Configure Radios
P) Configure PointPerfect
s) Configure System
f) Firmware upgrade
x) Exit

nseidle, 2023-10-26:
Replicated. Fixed. We'll get an RC out shortly.

nseidle, 2023-10-27:
This has been fixed in [v3.10](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v3.10).

closed 2023-10-27 by nseidle