sparkfun/SparkFun_RTK_Firmware #702: Fatal and nonfatal errors compiling firmware in Arduino IDE

demurs73582, 2023-10-15:
I am trying to compile v3.9 firmware binaries from the source following the directions here:\

https://docs.sparkfun.com/SparkFun_RTK_Firmware/firmware_update/#compiling-source

After some normal messages, I get the warnings listed below. Then some normal messages. After the message:

"Using library SparkFun_WebServer_ESP32_W5500 at version 1.5.5 in folder: D:\ARDUINO-ESP\ARDUINO-IDE\Sketchbook\libraries\SparkFun_WebServer_ESP32_W5500"

the IDE reports the error "ping timeout" and then stops.

Any suggestions?

Thanks.

Warnings:

In file included from D:\GNSS\SparkFun_RTK-Firmware\SparkFun_RTK_Firmware-main\Firmware\RTK_Surveyor\RTK_Surveyor.ino:154:
d:\ARDUINO-ESP\ARDUINO-IDE\Sketchbook\libraries\SdFat\src/SdFat.h:448:2: warning: #warning File not defined because __has_include(FS.h) [-Wcpp]
 #warning File not defined because __has_include(FS.h)
  ^~~~~~~
In file included from D:\GNSS\SparkFun_RTK-Firmware\SparkFun_RTK_Firmware-main\Firmware\RTK_Surveyor\RTK_Surveyor.ino:157:
D:\GNSS\SparkFun_RTK-Firmware\SparkFun_RTK_Firmware-main\Firmware\RTK_Surveyor\FileSdFatMMC.h: In member function 'virtual void FileSdFatMMC::flush()':
D:\GNSS\SparkFun_RTK-Firmware\SparkFun_RTK_Firmware-main\Firmware\RTK_Surveyor\FileSdFatMMC.h:161:22: error: request for member 'flush' is ambiguous
             _sdFile->flush();
                      ^~~~~
In file included from d:\ARDUINO-ESP\ARDUINO-IDE\Sketchbook\libraries\SdFat\src/FsLib/FsLib.h:31,
                 from d:\ARDUINO-ESP\ARDUINO-IDE\Sketchbook\libraries\SdFat\src/SdFat.h:33,
                 from D:\GNSS\SparkFun_RTK-Firmware\SparkFun_RTK_Firmware-main\Firmware\RTK_Surveyor\RTK_Surveyor.ino:154:
d:\ARDUINO-ESP\ARDUINO-IDE\Sketchbook\libraries\SdFat\src/FsLib/FsFile.h:207:8: note: candidates are: 'void FsBaseFile::flush()'
   void flush() { sync(); }
        ^~~~~
In file included from C:\Users\T470\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/Stream.h:26,
                 from C:\Users\T470\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/Arduino.h:177,
                 from C:\Users\T470\AppData\Local\Temp\arduino\sketches\13B7CC93C94AAC5ACE0275B2CF6E9A1E\sketch\RTK_Surveyor.ino.cpp:1:
C:\Users\T470\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/Print.h:112:18: note:                 'virtual void Print::flush()'
     virtual void flush() { /* Empty implementation for backward compatibility */ }
                  ^~~~~
In file included from D:\GNSS\SparkFun_RTK-Firmware\SparkFun_RTK_Firmware-main\Firmware\RTK_Surveyor\RTK_Surveyor.ino:397:
D:\GNSS\SparkFun_RTK-Firmware\SparkFun_RTK_Firmware-main\Firmware\RTK_Surveyor\bluetoothSelect.h: In member function 'virtual void BTLESerial::onConnect(BLEServer*)':
D:\GNSS\SparkFun_RTK-Firmware\SparkFun_RTK_Firmware-main\Firmware\RTK_Surveyor\bluetoothSelect.h:155:57: error: expression cannot be used as a function
         connectionCallback(ESP_SPP_SRV_OPEN_EVT, nullptr);
                                                         ^
D:\GNSS\SparkFun_RTK-Firmware\SparkFun_RTK_Firmware-main\Firmware\RTK_Surveyor\bluetoothSelect.h: In member function 'virtual void BTLESerial::onDisconnect(BLEServer*)':
D:\GNSS\SparkFun_RTK-Firmware\SparkFun_RTK_Firmware-main\Firmware\RTK_Surveyor\bluetoothSelect.h:161:54: error: expression cannot be used as a function
         connectionCallback(ESP_SPP_CLOSE_EVT, nullptr);
                                                      ^
D:\GNSS\SparkFun_RTK-Firmware\SparkFun_RTK_Firmware-main\Firmware\RTK_Surveyor\Bluetooth.ino: In function 'void bluetoothStart()':
D:\GNSS\SparkFun_RTK-Firmware\SparkFun_RTK_Firmware-main\Firmware\RTK_Surveyor\Bluetooth.ino:216:61: error: no matching function for call to 'BTSerialInterface::register_callback(void (&)(esp_spp_cb_event_t, esp_spp_cb_param_t*))'
         bluetoothSerial->register_callback(bluetoothCallback); // Controls BT Status LED on Surveyor
                                                             ^
In file included from D:\GNSS\SparkFun_RTK-Firmware\SparkFun_RTK_Firmware-main\Firmware\RTK_Surveyor\RTK_Surveyor.ino:397:
D:\GNSS\SparkFun_RTK-Firmware\SparkFun_RTK_Firmware-main\Firmware\RTK_Surveyor\bluetoothSelect.h:16:23: note: candidate: 'virtual esp_err_t BTSerialInterface::register_callback(void (**)(esp_spp_cb_event_t, esp_spp_cb_param_t*))'
     virtual esp_err_t register_callback(esp_spp_cb_t *callback) = 0;
                       ^~~~~~~~~~~~~~~~~
D:\GNSS\SparkFun_RTK-Firmware\SparkFun_RTK_Firmware-main\Firmware\RTK_Surveyor\bluetoothSelect.h:16:23: note:   no known conversion for argument 1 from 'void(esp_spp_cb_event_t, esp_spp_cb_param_t*)' to 'void (**)(esp_spp_cb_event_t, esp_spp_cb_param_t*)'
Multiple libraries were found for "Ethernet.h"
  Used: D:\ARDUINO-ESP\ARDUINO-IDE\Sketchbook\libraries\Ethernet
  Not used: C:\Users\T470\AppData\Local\Arduino15\libraries\Ethernet
Multiple libraries were found for "SparkFun_Qwiic_OLED.h"
  Used: D:\ARDUINO-ESP\ARDUINO-IDE\Sketchbook\libraries\SparkFun_Qwiic_OLED_Graphics_Library
  Not used: D:\ARDUINO-ESP\ARDUINO-IDE\Sketchbook\libraries\SparkFun_Qwiic_OLED_Arduino_Library

nseidle, 2023-10-15:
> error: request for member 'flush' is ambiguous _sdFile->flush();

This error is what I see when I use the ESP32 core that is above v2.0.2. Long story short, revert your ESP32 Arduino version to v2.0.2.

nseidle, 2023-10-15:
Thanks for attempting to compile the source! You are joining a very ~~elite~~ small club.

The docs *should* be accurate, but have a look at the [workflow yaml](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/main/.github/workflows/compile-rtk-firmware.yml) for the day-to-day compilation we use.

demurs73582, 2023-10-15:
Thanks, nseidle. Nothing like the voice of experience for advice!

After reverting to ESP core v2.0.2, the v3.9 source now finishes compilation, reporting only one warning:

In file included from D:\GNSS\SparkFun_RTK-Firmware\SparkFun_RTK_Firmware-main\Firmware\RTK_Surveyor\RTK_Surveyor.ino:154:
d:\ARDUINO-ESP\ARDUINO-IDE\Sketchbook\libraries\SdFat\src/SdFat.h:448:2: warning: #warning File not defined because __has_include(FS.h) [-Wcpp]
 #warning File not defined because __has_include(FS.h)
  ^~~~~~~
 
Near as I can tell with just a quick review, that warning is probably expected and the result of intentional choices in included files. Before actually loading the sketch to my Facet, though, I would appreciate a confirmation of that.

Thanks again, nseidle.

nseidle, 2023-10-16:
Yep - that is currently the only warning. Congrats! Welcome to the club (we really should have stickers or something). 

If you feel so inclined, we welcome PRs.

Don't worry about loading incomplete or bad code onto a unit. It's always recoverable; you can always load previous firmwares regardless of the state you get a unit into.

For your education (and anyone reading) we are currently limited to v2.0.2 of the ESP32 core because of RAM concerns. The newer versions absorb another ~12k of RAM (presumably in the WiFi and BT libraries). This pushes this current codebase to the limit in terms of stability. We stay at v2.0.2 because we don't (yet) need the newer core features.

closed 2023-10-16 by nseidle