sparkfun/SparkFun_RTK_Firmware #712: Updating from 3.3 to 3.9 via Config web page fails

gdt, 2023-10-19:
### Subject of the issue
I'm trying to update an RTK Express from 3.3 to 3.9 via the builtin web page.

### Your workbench
* unit is running 3.3, working fine.   When I use it I use wifi to a hotspot for RTCM and BTSPP to a phone for QField.  All is ok.

### Steps to reproduce
* Put unit in config mode.  observe that it shows up as "RTK Config".  Join that ssid (iMac, also ethernet)
* click around and verify that it basically seems to work fine
* download the 3.9 bin from github releases
* click 'upload' and select that file
* while doing this, computer and unit are on opposite sides of the house, but it all seemed ok before I started uploading.  Will repeat with them side by side.

### Expected behavior
* update should succeed

### Actual behavior
* observe that 'waiting' never goes away.   Reloading the page then times out.

### Notes
* I've had this paged out for a while.
* This is not a help request; I know I can update via esptool, via uSD, and via the direct download.  It's a report that something I expected to work did not.
* I realize 3.3 is wicked old as we say in Boston and if you don't want to deal with that, that's ok

nseidle, 2023-10-27:
I'm getting more and more suspicious of the WiFi based config. I think the AsyncWebServer library is amazing, but also not entirely bug free (I can say the same about this firmware). The file upload/download interface is not as mature as the system configuration update (there's a few related issues).

[image]

Please give that a try and let me know.

gdt, 2023-10-27:
I have not been using the update by checking method, because I'm using the wifi config where the ESP is the AP, and thus there's no route.    Is it the case that when you click check, it disconnects, finds one off the configured networks to use as a client, does the fetch, and comes back to the config setting?    That could make sense, but it violates the principle of least astonishment and I have never come to expect it.   Maybe I didn't read well.

gdt, 2023-11-02:
Now I see that it stays as the config AP and also becomes a STA on the configured WiFi.  But after synack to github that looks fine, it is sending a FIN.   However I get it that we are basically debugging 3.3 at this point and that if I were you I wouldn't want to do that!

nseidle, 2023-11-02:
Too many TLAs my friend. 

> However I get it that we are basically debugging 3.3 at this point and that if I were you I wouldn't want to do that!

I do not understand.

nseidle, 2023-11-02:
Ah, never mind. I see now. Yes, we are commenting about v3.3 firmware behavior. 

I am assuming from your comment, that the update succeeded? Are we ok to close this out?

gdt, 2023-11-02:
I used esptool and a cable to update to 3.9.    So yes, I am no longer interested in figuring out why the 3.3 firmware cannot receive new firmware :-)    There are likely still related issues but I'll open new tickets.   Thanks for the help.

One thing that is perhaps in the docs that I didn't read well enough, but is to say more loudly that the unit can, when in config mode, also access the configured wifi as a client.   This is surprising to those who expect a wifi chip to be in one mode at a time, and it explains a lot.

closed 2023-11-02 by gdt